Benjamin-Dobell/Heimdall #367: QT5 dependencies

cederom, 2016-09-29:
Hey! In most cases commandline version of the Heimdall will do the work. Forcing QT5 dependencies makes it really hard to compile as QT5 fails to build (tried macOS 10.12 MacPorts and Brew). As the installer fails too (maybe also because of QT) there is no way to run Heimdall on my system. This is wrong :-( QT should be an option not the default as it introduces heavy and unreliable dependencies..

Benjamin-Dobell, 2016-09-29:
You're running a beta OS, what did you expect?

closed 2016-09-29 by Benjamin-Dobell

Benjamin-Dobell, 2016-09-29:
Also, Cmake option `DISABLE_FRONTEND`

cederom, 2016-09-29:
macOS 10.12 is now a release ;-)
http://www.apple.com/macos/sierra/

Benjamin-Dobell, 2016-09-29:
All Apple software is beta 😉 

Don't get me wrong, OS X has been my primary OS for years. But it's all downhill. USB stack rewrite in 10.10-10.11 (now closed source) is particularly terrible (I'm sick of my USB 3 hubs playing up). It's not and never has been Apple that makes it a good OS, it's the community driven UNIX tools that make it superior to Windows. Apple do to make beautiful hardware, it's just all too often crippled by amateur buggy software and the company's non-existent software support channels (again hardware support/replacements is great).

But yes, Sierra is brand new, I don't control Qt or Homebrew; unfortunately compatibility issues are commonplace with OS upgrades. If you have problems with either of those projects please open issues (after searching for existing issues) where appropriate in the respective projects bug trackers.