Benjamin-Dobell/Heimdall #369: Mac OS X build need repackaging

db17, 2016-10-03:
Since El Capitan (Mac OS X 10.11), it's not anymore possible to install file in /usr/bin and /usr/lib

Installation layout should be:
/Applications/heimdall-frontend.app
/Library/Extensions/heimdall.kext
/usr/local/bin/heimdall
/usr/local/lib/libusb-1.0.dylib

d5ve, 2017-02-02:
I ran into this as well on macOS Sierra. I tried to rebuild the APK, but couldn't get it to work.

What did work was to create a new partition on my hard drive, and install Heimdall on that.

I followed the "How to create a partition on your Mac" steps from http://www.imore.com/how-partition-your-mac to create an 8GB partition. This is a non-destructive process, as it just resizes the existing partition (as long as you have enough free disk space). It didn't affect my OS install at all.

Then I ran the Heimdall installation process and chose "Change Install Location", where I was able to select my new partition, where it installed fine. The application seems to work as expected.

ehartford, 2017-05-31:
i cant install heimdall in Sierra

db17, 2018-02-25:
@d5ve, to manually extract files from the .pkg installer, you can also run the following commands (once Heimdall's image disk is open):
```
mkdir /tmp/heimdall/
cd /tmp/heimdall/
xar -xf /Volumes/Heimdall\ Suite\ 1.4.0/Heimdall\ Suite\ 1.4.0.pkg Payload
sudo tar -xf Payload    # sudo is for extracting owner/mode
```

Then to install, the same commands as you proposed:
```
sudo cp -R Applications/heimdall-frontend.app /Applications/
sudo cp -R System/Library/Extensions/heimdall.kext /Library/Extensions/
sudo cp usr/bin/heimdall /usr/local/bin/heimdall
sudo usr/lib/libusb-1.0.dylib /usr/local/lib/libusb-1.0.dylib
```