Benjamin-Dobell/Heimdall #370: "Odin protocol version error!" on Samsung Galaxy S7 (SM-G930A)

auxiliary-character, 2016-10-04:
This is on Arch Linux, installed from the AUR just a few hours ago (version `1.4.1.r37.gb6fe7f8-1`) with the `heimdall-git` package.

Here's the log:

```
[austin@wolfram samsung-tools]$ sudo heimdall print-pit --usb-log-level debug
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.011193] [00000b92] libusb: debug [libusb_get_device_list] 
[ 0.011240] [00000b92] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.011258] [00000b92] libusb: debug [libusb_get_device_descriptor] 
[ 0.011266] [00000b92] libusb: debug [libusb_get_device_descriptor] 
[ 0.011273] [00000b92] libusb: debug [libusb_get_device_descriptor] 
[ 0.011286] [00000b92] libusb: debug [libusb_open] open 4.82
[ 0.011329] [00000b92] libusb: debug [usbi_add_pollfd] add fd 9 events 4
[ 0.011340] [00000b92] libusb: debug [libusb_get_device_descriptor] 
[ 0.011350] [00000b92] libusb: debug [libusb_get_config_descriptor] index 0
Claiming interface...
[ 0.011380] [00000b92] libusb: debug [libusb_detach_kernel_driver] interface 1
[ 0.011396] [00000b92] libusb: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 0.011447] [00000b92] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0

Initialising protocol...
[ 0.011729] [00000b92] libusb: debug [libusb_alloc_transfer] transfer 0x1fd9998
[ 0.011748] [00000b92] libusb: debug [libusb_submit_transfer] transfer 0x1fd9998
[ 0.011759] [00000b92] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.011777] [00000b92] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 0.011808] [00000b92] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.011819] [00000b92] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.011831] [00000b92] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.012041] [00000b92] libusb: debug [handle_events] poll() returned 1
[ 0.012090] [00000b92] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=4
[ 0.012103] [00000b92] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.012113] [00000b92] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 0.012122] [00000b92] libusb: debug [disarm_timerfd] 
[ 0.012133] [00000b92] libusb: debug [usbi_handle_transfer_completion] transfer 0x1fd9998 has callback 0x7f1c25c90b90
[ 0.012142] [00000b92] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.012160] [00000b92] libusb: debug [libusb_free_transfer] transfer 0x1fd9998
[ 0.012175] [00000b92] libusb: debug [libusb_alloc_transfer] transfer 0x1fd9998
[ 0.012184] [00000b92] libusb: debug [libusb_submit_transfer] transfer 0x1fd9998
[ 0.012192] [00000b92] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.012203] [00000b92] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 7
[ 0.012227] [00000b92] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.012236] [00000b92] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.012683] [00000b92] libusb: debug [handle_events] poll() returned 1
[ 0.012733] [00000b92] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=4
[ 0.012746] [00000b92] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.012756] [00000b92] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 0.012765] [00000b92] libusb: debug [disarm_timerfd] 
[ 0.012776] [00000b92] libusb: debug [usbi_handle_transfer_completion] transfer 0x1fd9998 has callback 0x7f1c25c90b90
[ 0.012784] [00000b92] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.012795] [00000b92] libusb: debug [libusb_free_transfer] transfer 0x1fd9998
Protocol initialisation successful.

Beginning session...
[... 139 lines cut ...]
[21.811745] [00000b92] libusb: debug [handle_events] poll() returned 1
[21.811852] [00000b92] libusb: debug [handle_events] timerfd triggered
[21.811870] [00000b92] libusb: debug [libusb_cancel_transfer] transfer 0x1fc66b8
[21.812173] [00000b92] libusb: debug [disarm_timerfd] 
[21.812217] [00000b92] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[21.812234] [00000b92] libusb: debug [handle_events] poll() returned 1
[21.812250] [00000b92] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[21.812261] [00000b92] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[21.812270] [00000b92] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
[21.812279] [00000b92] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[21.812289] [00000b92] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[21.812298] [00000b92] libusb: debug [disarm_timerfd] 
[21.812307] [00000b92] libusb: debug [usbi_handle_transfer_completion] transfer 0x1fc66b8 has callback 0x7f1c25c90b90
[21.812316] [00000b92] libusb: debug [sync_transfer_cb] actual_length=0
[21.812330] [00000b92] libusb: debug [libusb_free_transfer] transfer 0x1fc66b8
Releasing device interface...
[21.812373] [00000b92] libusb: debug [libusb_release_interface] interface 1

[21.812423] [00000b92] libusb: debug [libusb_close] 
[21.812443] [00000b92] libusb: debug [usbi_remove_pollfd] remove fd 9
[21.812470] [00000b92] libusb: debug [libusb_exit] 
[21.812480] [00000b92] libusb: debug [libusb_exit] destroying default context
```

If there's anything else I can provide, let me know.

charliegreen, 2017-10-13:
I'm having a similar issue, on Debian 9.2, with heimdall v1.4.2 (built from commit 9bcc42d, which at the time I post this is HEAD). I'm also using a Samsung Galaxy S7 SM-G930A, and getting the error message "Odin protocol version error!" on it. Here's my log:

```
$ ./bin/heimdall version
v1.4.2
$ ./bin/heimdall print-pit --usb-log-level debug
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.003532] [00000b3c] libusb: debug [libusb_get_device_list] 
[ 0.003549] [00000b3c] libusb: debug [libusb_get_device_descriptor] 
[ 0.003551] [00000b3c] libusb: debug [libusb_get_device_descriptor] 
[ 0.003552] [00000b3c] libusb: debug [libusb_get_device_descriptor] 
[ 0.003554] [00000b3c] libusb: debug [libusb_get_device_descriptor] 
[ 0.003556] [00000b3c] libusb: debug [libusb_get_device_descriptor] 
[ 0.003559] [00000b3c] libusb: debug [libusb_open] open 1.13
[ 0.003589] [00000b3c] libusb: debug [usbi_add_pollfd] add fd 9 events 4
[ 0.003593] [00000b3c] libusb: debug [libusb_get_device_descriptor] 
[ 0.003596] [00000b3c] libusb: debug [libusb_get_config_descriptor] index 0
Claiming interface...
[ 0.003609] [00000b3c] libusb: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 0.003629] [00000b3c] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0

Initialising protocol...
[ 0.003924] [00000b3c] libusb: debug [libusb_alloc_transfer] transfer 0x55f02cdb2020
[ 0.003929] [00000b3c] libusb: debug [libusb_submit_transfer] transfer 0x55f02cdb2020
[ 0.003931] [00000b3c] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.003936] [00000b3c] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 0.003946] [00000b3c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.003951] [00000b3c] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.003962] [00000b3c] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.003981] [00000b3c] libusb: debug [handle_events] poll() returned 1
[ 0.003986] [00000b3c] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=4
[ 0.003990] [00000b3c] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.003994] [00000b3c] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 0.003999] [00000b3c] libusb: debug [disarm_timerfd] 
[ 0.004002] [00000b3c] libusb: debug [usbi_handle_transfer_completion] transfer 0x55f02cdb2020 has callback 0x7f408cd59ff0
[ 0.004004] [00000b3c] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.004015] [00000b3c] libusb: debug [libusb_free_transfer] transfer 0x55f02cdb2020
[ 0.004021] [00000b3c] libusb: debug [libusb_alloc_transfer] transfer 0x55f02cdb2020
[ 0.004025] [00000b3c] libusb: debug [libusb_submit_transfer] transfer 0x55f02cdb2020
[ 0.004028] [00000b3c] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.004032] [00000b3c] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 7
[ 0.004038] [00000b3c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.004042] [00000b3c] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.004870] [00000b3c] libusb: debug [handle_events] poll() returned 1
[ 0.004876] [00000b3c] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=4
[ 0.004879] [00000b3c] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.004883] [00000b3c] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 0.004886] [00000b3c] libusb: debug [disarm_timerfd] 
[ 0.004891] [00000b3c] libusb: debug [usbi_handle_transfer_completion] transfer 0x55f02cdb2020 has callback 0x7f408cd59ff0
[ 0.004895] [00000b3c] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.004899] [00000b3c] libusb: debug [libusb_free_transfer] transfer 0x55f02cdb2020
Protocol initialisation successful.

Beginning session...
[... 140 lines cut ...]
[21.801897] [00000b3c] libusb: debug [handle_events] timerfd triggered
[21.801905] [00000b3c] libusb: debug [libusb_cancel_transfer] transfer 0x55f02cdb9d40
[21.801964] [00000b3c] libusb: debug [disarm_timerfd] 
[21.801978] [00000b3c] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
[21.801991] [00000b3c] libusb: debug [handle_events] poll() returned 1
[21.802007] [00000b3c] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[21.802019] [00000b3c] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[21.802027] [00000b3c] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
[21.802037] [00000b3c] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[21.802048] [00000b3c] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[21.802059] [00000b3c] libusb: debug [disarm_timerfd] 
[... cut ...]
[21.802103] [00000b3c] libusb: debug [libusb_free_transfer] transfer 0x55f02cdb9d40
Releasing device interface...
[21.802143] [00000b3c] libusb: debug [libusb_release_interface] interface 1

[21.802182] [00000b3c] libusb: debug [libusb_close] 
[21.802203] [00000b3c] libusb: debug [usbi_remove_pollfd] remove fd 9
[21.802230] [00000b3c] libusb: debug [libusb_exit] 
[21.802240] [00000b3c] libusb: debug [libusb_exit] destroying default context
$
```

Please let me know if there's any other information I have that may be useful!

Chris-Slade, 2018-01-20:
Having the same issue on a Galaxy S7 G930A. Running Manjaro Linux on a x86_64 laptop (kernel version 4.14.14-1-MANJARO), with libusb version 1.0.21-2. Tried both the Heimdall versions in the AUR (git and non-git), and compiled from source. In all cases `print-pit` would fail with `
ERROR: libusb error -7 whilst receiving bulk transfer`, and Download mode on the phone would print `Odin protocol version error!`.

sci4me, 2018-03-15:
Same exact problem but on Windows. Phone says `Odio protocl version error`. Heimdall says `libusbx: error [init_device] device '\\.\USB#VID_1532&PID_0504&MI_03#7&124BE690&0&0003' is no longer connected!`

Blefish, 2018-03-16:
I have a patch for this, is @Benjamin-Dobell still maintaining this project?

Benjamin-Dobell, 2018-03-16:
@Blefish Yes, and PRs are welcome. They just need to be thorough. Fixes for one device tend to break them for others.

Blefish, 2018-03-23:
There is a patch available here, can anyone confirm whether it works? #459

anonveni98, 2023-01-05:
Can i install twrp with heimdall as i tried to install twrp with herolte odin 3.14 but it failed. After googling I found that there is a problem with the SM-G930A. If it shows Qualcomm secure boot enabled then we can't install twrp. So odin failed to install twrp and shows an error SECURE CHECK FAIL : Recovery
Thanks in advance.

DaddyW123, 2023-09-01:
> Can i install twrp with heimdall as i tried to install twrp with herolte odin 3.14 but it failed. After googling I found that there is a problem with the SM-G930A. If it shows Qualcomm secure boot [...]

Did you ever find a resolution for this?  I am trying to flash an S10 with Lineage, and I'm getting the "Odin protocol version error!", and mine too is saying "Qualcomm secureboot : enabled".  Which I have a feeling is causing my problem.
This device was already rather bricked because using the stock recovery, I did a "wipe and restore to factory defaults", but that process didn't work right because it just boot loops into recovery now.  So I cannot boot in to the stock android to turn on developer mode.  I was basically just hoping that I could salvage this phone but I have a feeling it's going to have to go to e-waste.