Benjamin-Dobell/Heimdall #376: Samsung galaxy S4 (GT-I9506) : Failed to begin session

cdwijs, 2016-11-14:
Hi All, I am trying to install CyanogenMod on my galaxy S4 following this guide:
https://wiki.cyanogenmod.org/w/Install_CM_for_ks01lte
I cannot flash the custom recovery, I get the error message: The device does not recognize the command

Steps I have taken:
-boot the phone into ODIN mode (home-volume-down-power, then volume up to accept
-connect the phone
-run zadig, replace driver (keep zadig running)
-run usbpcap, filename: S4
-run heimdall
-open S4 file in wireshark
[S4.zip](https://github.com/Benjamin-Dobell/Heimdall/files/589219/S4.zip)

Messages of Heimdall:
heimdall.exe flash --RECOVERY twrp-3.0.2-0-ks01lte.img --no-reboot
Heimdall v1.4.0
Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/
This software is provided free of charge. Copying and redistribution is
encouraged.
If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/
Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...
Initialising protocol...
Protocol initialisation successful.
Beginning session...
libusbx: error [winusbx_submit_bulk_transfer] ReadPipe/WritePipe failed: [22] The device does not recognize the command.
libusbx: error [winusbx_submit_bulk_transfer] ReadPipe/WritePipe failed: [22] The device does not recognize the command.
libusbx: error [winusbx_submit_bulk_transfer] ReadPipe/WritePipe failed: [22] The device does not recognize the command.
libusbx: error [winusbx_submit_bulk_transfer] ReadPipe/WritePipe failed: [22] The device does not recognize the command.
libusbx: error [winusbx_submit_bulk_transfer] ReadPipe/WritePipe failed: [22] The device does not recognize the command.
ERROR: Failed to begin session!
Releasing device interface...

Messages of the phone after booting into ODIN mode:
ODIN MODE
PRODUCTNAME: GT-I9506
CURRENT BINARY: Samsung Official
SYSTEM STATUS: Official
KNOX WARRANTY VOID: 0x0
QUALCOMM SECUREBOOT: ENABLE
AP SWREV: S1, T1, A1, A1, P1
SECURE DOWNLOAD : ENABLE
UDC START

In wireshark, the phone has source: 2.1.0, but wireshark shows it as a galaxy SII:
Frame 1239: 46 bytes on wire (368 bits), 46 bytes captured (368 bits)
USB URB
    [Source: 2.1.0]
    [Destination: host]
    USBPcap pseudoheader length: 28
    IRP ID: 0xfffffa80091c1530
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_CONTROL_TRANSFER (0x0008)
    IRP information: 0x01, Direction: PDO -> FDO
    URB bus id: 2
    Device address: 1
    Endpoint: 0x80, Direction: IN
    URB transfer type: URB_CONTROL (0x02)
    Packet Data Length: 18
    [Request in: 1238]
    [Time from request: 0.001000000 seconds]
    Control transfer stage: Data (1)
DEVICE DESCRIPTOR
    bLength: 18
    bDescriptorType: 0x01 (DEVICE)
    bcdUSB: 0x0200
    bDeviceClass: Communications and CDC Control (0x02)
    bDeviceSubClass: 2
    bDeviceProtocol: 0
    bMaxPacketSize0: 64
    idVendor: Samsung Electronics Co., Ltd (0x04e8)
    idProduct: GT-I9100 Phone [Galaxy S II] (Download mode) (0x685d)
    bcdDevice: 0x0100
    iManufacturer: 1
    iProduct: 2
    iSerialNumber: 0
    bNumConfigurations: 1

My versions:
Heimdall v1.4.0
Windows 7 Ultimate SP1 64bit, 8GB ram
USBPcap 1.0.0.7 http://desowin.org/usbpcap/
wireshark 2.2.1 https://www.wireshark.org/

iamantony, 2016-11-15:
Try to use Heimdall v1.4.1 (see Releases). Or download this repo and build Heimdall from source.

cdwijs, 2016-11-15:
Hi Iamantony,
Where is the compiled windows version of v1.4.1? I can't find them in these locations:
https://bitbucket.org/benjamin_dobell/heimdall/downloads
https://github.com/Benjamin-Dobell/Heimdall/releases/tag/v1.4.1

Cheers,
Cedric

cdwijs, 2016-12-03:
I have compiled Heimdall v1.4.1 on arch linux. After using a better USB cable I succeeded in installing the custom recovery. 
Thank you for your work on this program. If I can be of assistance for testing something on my GT-I9506 phone, please let me know. 
Best, Cedric

iamantony, 2016-12-05:
Good news) 
I would suggest you to close this issue because you have solved problem.

closed 2016-12-05 by cdwijs