Benjamin-Dobell/Heimdall #377: Build Heimdall on windows and msys2: Could NOT find libusb (missing: LIBUSB_LIBRARY)

cdwijs, 2016-11-15:
I am trying to build Heimdall from source, but I get the following error:
CMake Error at D:/msys64/mingw64/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find libusb (missing: LIBUSB_LIBRARY)
Call Stack (most recent call first):
  D:/msys64/mingw64/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/Findlibusb.cmake:30 (find_package_handle_standard_args)
  heimdall/CMakeLists.txt:9 (find_package)

Steps I have taken:
Download msys2 http://repo.msys2.org/distrib/x86_64/msys2-x86_64-20161025.exe
Install to D:\msys64 (instead of c:\msys64)
run msys2, then run the following commands:
pacman -Sy pacman 
<< restart msys2 >>
pacman -Sy pacman
pacman -Syu
pacman -Syu
<< restart msys2 >>
pacman -Su

then run the commands from here:
https://github.com/Benjamin-Dobell/Heimdall/tree/master/Win32
pacman -Syu
pacman -S mingw-w64-x86_64 mingw-w64-x86_64-clang mingw-w64-x86_64-cmake mingw-w64-x86_64-libusb mingw-w64-x86_64-qt5-static make
pacman -S git
git clone https://github.com/Benjamin-Dobell/Heimdall.git
cd Heimdall/
mkdir build
cd build/
pacman -S gcc
cmake -G "MSYS Makefiles" -DCMAKE_BUILD_TYPE=Release -DQt5Widgets_DIR=/d/msys64/mingw64/qt5-static/lib/cmake/Qt5Widgets ..

Full buildlog: 
******************
cedric@Ce MSYS ~
$ pacman -Sy pacman
:: Synchronizing package databases...
 mingw32                  303.1 KiB   698K/s 00:00 [#####################] 100%
 mingw32.sig               96.0   B  0.00B/s 00:00 [#####################] 100%
 mingw64                  302.8 KiB  1033K/s 00:00 [#####################] 100%
 mingw64.sig               96.0   B  0.00B/s 00:00 [#####################] 100%
 msys                     136.8 KiB   871K/s 00:00 [#####################] 100%
 msys.sig                  95.0   B  0.00B/s 00:00 [#####################] 100%
warning: pacman-5.0.1-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) pacman-5.0.1-1

Total Download Size:    6.81 MiB
Total Installed Size:  33.72 MiB
Net Upgrade Size:       0.00 MiB

:: Proceed with installation? [Y/n]
:: Retrieving packages...
 pacman-5.0.1-1-x86_64      6.8 MiB  1093K/s 00:06 [#####################] 100%
(1/1) checking keys in keyring                     [#####################] 100%
(1/1) checking package integrity                   [#####################] 100%
(1/1) loading package files                        [#####################] 100%
(1/1) checking for file conflicts                  [#####################] 100%
(1/1) checking available disk space                [#####################] 100%
:: Processing package changes...
(1/1) reinstalling pacman                          [#####################] 100%

cedric@Ce MSYS ~
$
******************
restart msys2
******************
cedric@Ce MINGW64 ~
[... 1253 lines cut ...]
-- The CXX compiler identification is GNU 5.3.0
-- Check for working C compiler: D:/msys64/usr/bin/gcc.exe
-- Check for working C compiler: D:/msys64/usr/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: D:/msys64/usr/bin/g++.exe
-- Check for working CXX compiler: D:/msys64/usr/bin/g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at D:/msys64/mingw64/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find libusb (missing: LIBUSB_LIBRARY)
Call Stack (most recent call first):
  D:/msys64/mingw64/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/Findlibusb.cmake:30 (find_package_handle_standard_args)
  heimdall/CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "D:/msys64/home/cedric/Heimdall/build/CMakeFiles/CMakeOutput.log".

cedric@Ce MINGW64 ~/Heimdall/build
$

cdwijs, 2016-12-04:
I have asked the LIBUSB error to the msys2 folks, the answer is here: I did not test it myself yet.
https://github.com/Alexpux/MSYS2-packages/issues/765

Cheers,
Cedric

TravisCHoward, 2016-12-05:
Unfortunately, they do not actually provide a solution...