Benjamin-Dobell/Heimdall #378: Compiling error "CMake Error at heimdall-frontend/CMakeLists.txt:11"

tamirciat, 2016-11-19:
Hi,
I did this in OpenSuse -64 bit 42.1:
git clone https://github.com/Benjamin-Dobell/Heimdall.git
cd Heimdall
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Release ..

and I've got this error:

"-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found libusb: /usr/lib64/libusb-1.0.so  
-- Checking if large (64-bit) file support is available...
-- Checking if large (64-bit) file support is available - yes
CMake Error at heimdall-frontend/CMakeLists.txt:11 (find_package):
  By not providing "FindQt5Widgets.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Qt5Widgets", but CMake did not find one.

  Could not find a package configuration file provided by "Qt5Widgets" with
  any of the following names:

    Qt5WidgetsConfig.cmake
    qt5widgets-config.cmake

  Add the installation prefix of "Qt5Widgets" to CMAKE_PREFIX_PATH or set
  "Qt5Widgets_DIR" to a directory containing one of the above files.  If
  "Qt5Widgets" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/abid/Heimdall/build/CMakeFiles/CMakeOutput.log"."

What a problem here?

iamantony, 2016-11-20:
Check that you have installed this package: qt5-default

tamirciat, 2016-11-20:
qt5-default not exist in opensuse repo, I have installed libqt5-creator instead.

iamantony, 2016-11-20:
libqt5-creator is IDE for qt/c++ projects. Try to search for package qt-core or something like this.

tamirciat, 2016-11-21:
This packages installed in my system opensuse 42.1. Does it not enough?

[image]

iamantony, 2016-11-21:
Hmm... see [this question on stackoverflow](http://stackoverflow.com/questions/18722329/cmake-does-not-find-qt-5-1-1). I hope it will help.

tamirciat, 2016-11-27:
I don't have Qt5Widgets folder in my system.

iamantony, 2016-11-28:
Try to run this command as root user:
_find / -type d -name 'Qt5Widgets'_

In Ubuntu Qt5Widgets folder located here:
_/usr/lib/x86_64-linux-gnu/cmake/Qt5Widgets_

As a final variant - try to install Qt from qt.io. Download .run-installer from [here](https://www.qt.io/download-open-source/). This installer contains all qt files and libraries that you need.

BenjaminDeuter, 2017-05-27:
In case someone is getting here via Google, this problem sounds a lot like the one I just solved on a recent installation of OpenSUSE Tumbleweed.

@tamirciat If this issue is still of interest to you, the package you were likely missing is `libQt5Widgets-devel`.

@Benjamin-Dobell I believe this resolves the issue, at least it did for me, so this might be closed now.

closed 2017-05-27 by Benjamin-Dobell

dandv, 2017-10-13:
I'm on Ubuntu 16.04, and the `libQt5Widgets-devel` package wasn't found. `qt5-default` worked instead. To make things easy for Ubuntu users:

    sudo apt install -y build-essential cmake zlib1g-dev qt5-default libusb-1.0-0-dev