Benjamin-Dobell/Heimdall #380: Error Compiling

Travis-Ecc, 2016-11-29:
I have followed the instructions at https://wiki.cyanogenmod.org/w/Install_and_compile_Heimdall#Windows
However when I use the last command I get an error

$ cmake -G "MSYS Makefiles" -DCMAKE_BUILD_TYPE=Release -DQt5Widgets_DIR=/mingw64/qt5-static/lib/cmake/Qt5Widgets ..
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:3 (project):
  The CMAKE_C_COMPILER:

    gcc.exe

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:3 (project):
  The CMAKE_CXX_COMPILER:

    g++.exe

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error: CMAKE_AR was not found, please set to archive program. CMAKE_AR-NOTFOUND
-- Configuring incomplete, errors occurred!

I'm new to this so step by step stupid instructions are required

cdwijs, 2016-12-03:
Hi Travis,
I had the same error, this error means there's no compiler installed. This can be corrected by issuing the following command:
pacman -S gcc

The next problem you could run into is that you will get this error, see also this issue:
https://github.com/Benjamin-Dobell/Heimdall/issues/377
Could NOT find libusb (missing: LIBUSB_LIBRARY)

I have not found a way around this one on windows, on arch linux the compile goes without problems.
Hope this helps,
Cedric

cdwijs, 2016-12-04:
Hi Travis,
I have asked the LIBUSB error to the msys2 folks, the answer is here: I did not test it myself
https://github.com/Alexpux/MSYS2-packages/issues/765

Cheers,
Cedric

TravisCHoward, 2016-12-04:
I'm having this problem as well. Followed the steps from win32/readme with the addition of a gcc installation.

`CMake Error at C:/msys64/mingw64/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find libusb (missing: LIBUSB_LIBRARY)
Call Stack (most recent call first):
  C:/msys64/mingw64/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/Findlibusb.cmake:30 (find_package_handle_standard_args)
  heimdall/CMakeLists.txt:9 (find_package)
`

diepes, 2017-12-11:
Found this searching for a Linux make, error's
BridgeManager.cpp:(.text+0xxxx): undefined reference to `libusb_xxxxxx'

Solution was to edit 
~/git/Heimdall/build [master|…1] 
17:45 $ gvim heimdall/CMakeFiles/heimdall.dir/link.txt

and add a flag -lusb-1.0 to the end.