Benjamin-Dobell/Heimdall #381: Download pit for S5 G900F fails with heimdall-frontend but succeeds with command-line binaries

lope, 2016-11-29:
Both binaries were compiled from the source I downloaded from master today.
I didn't save the error, but it gave the typical error seen in other issues regarding not being able to end the transfer successfully etc.

Benjamin-Dobell, 2016-11-29:
It does seem a bit off that Heimdall Grontend would fail where Heimdall command line succeeds, as the former utilises the latter.

My guess is that you ended up with multiple `heimdall` CLI commands on your computer and Heimdall Frontend was calling the wrong one. If you go into the about screen you can see what version of Heimdall command line the Frontend is using.

lope, 2016-11-29:
Oh that explains it. I thought it seemed a little crazy/unusual that the frontend would re-implement the code from the CLI. I mean at first it looked like it was just dumping the CLI output.

After reading your response now, I realized what happened.

The newly compiled heimdall-frontend must have been using the ubuntu 14.04 **installed** Heimdall 1.4, which experiences that problem.
Of course when I ran the new binary, I ran the binary directly.

BTW, after running make, I tried to run `sudo make install` but it said there's no install stuff as part of the build script. Then I scratched around in some make file and I saw some text "INSTALL" something, but then figured "whatever" and then just ran it from the terminal.