Benjamin-Dobell/Heimdall #382: S4 Mini (GTI9195) 4.4.2 Flash fails on Linux

mikmikmik, 2016-12-03:
I tried from Fedora and Debian, from repos and from sources.
I tried with SaburoJiro branch, at different commits heads.
Always get a end of file error when I try to flash.
I also tried with 2 different usb cables on all my usb ports, both of my computers...

iamantony, 2016-12-05:
Could you provide logs?

mikmikmik, 2016-12-06:
I'll check again later and put logs here. Thanks for your interest.

mikmikmik, 2016-12-06:
I uploaded the .img directly and got ERROR: Failed to unpack received packet.
```
sudo heimdall flash --RECOVERY recovery.img --no-reboot --verbose
Heimdall v1.4.1
```
(...)
```
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
(x19)
PIT file download successful.
```

```
Uploading RECOVERY
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
0%WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

7%
(...)
100%
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: Failed to unpack received packet.

ERROR: Failed to confirm end of file transfer sequence!
ERROR: RECOVERY upload failed!

Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Releasing device interface...
```
If I try with a SIGNED .zip I get:
`ERROR: libusb error -7 whilst receiving packet. Retrying...`

mikmikmik, 2016-12-06:
Sorry for this issue it's working now, the problem was my recovery.img file... I tried with another one and it worked...

closed 2016-12-06 by mikmikmik