Benjamin-Dobell/Heimdall #384: Downloading PIT file on samsung galaxy s3 (I747) failes with frontend and CLI

GNUGradyn, 2016-12-10:
Trying to reinstall stock firmware from cyanogenmod so I can network unlock my phone, but im getting this error

```
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
PIT file download successful.

ERROR: Partition "file" does not exist in the specified PIT.
Ending session...
Rebooting device...
Releasing device interface...

```

GNUGradyn, 2016-12-10:
Figured it out, i had the syntax wrong. Im assuming if im replacing the firmware, i write over the SYSTEM partition?

closed 2017-05-22 by GNUGradyn

JonnyTech, 2017-05-22:
Extract the tar file, there you will find various .img files. To install a new rom you will need somethng like this:

```
heimdall flash --BOOT boot.img  --no-reboot
heimdall flash --SYSTEM system.img 
```

alex63w5, 2024-04-13:
I have a similar problem when I go to flash a boot image on Samsung using heimdall and error partition boot does not exist in the specified pit