Benjamin-Dobell/Heimdall #388: "WARNING: Empty bulk transfer after sending packet failed" when trying to flash a Samsung Galaxy S4 Mini GT-I9195 using compiled-from-source Heimdall 1.4.1

breversa, 2017-01-15:
Hi,

Following @lope's advice here : https://github.com/Benjamin-Dobell/Heimdall/issues/348, I built Heimdall from source and tried to flash a Samsung Galaxy S4 Mini (GT-I9195), to no avail :

`sudo ./heimdall flash --RECOVERY recovery-twrp-2.6.3.0-serranoltexx.img --verbose --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after receiving packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
PIT file download successful.

Uploading RECOVERY
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
0%WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

14%

28%

42%

57%

71%

85%

100%
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
RECOVERY upload successful

Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Releasing device interface...
`

00joshi, 2017-03-12:
Can confirm:

Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after receiving packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
PIT file download successful.

Uploading RECOVERY
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
0%WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

7%

14%

22%

29%

37%

44%

51%

59%

66%

74%

81%

88%

96%

100%
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: Failed to unpack received packet.

ERROR: Failed to confirm end of file transfer sequence!
ERROR: RECOVERY upload failed!

Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Releasing device interface...

Arroloff, 2017-03-19:
i got the same issue, s4 mini, build from source, mint 18 and with the same output.

stratosmacker, 2017-06-15:
has there been any movement on this issue? I'm stuck and cannot flash kernels

alanpoon, 2017-06-23:
Same issue on GT-I9500.

SIN47, 2017-07-08:
Same issue Archlinux, Heimdall 1.4.2, Samsung S5 G900V. Trying to flash TWRP

aliensonneptune, 2017-08-24:
Trying TWRP, confirm Elementary OS 0.4, Heimdall 1.4.1, please fix, have been trying to flash TWRP for like 2 days.

paulsommer, 2018-01-19:
Still no solution for this issue?
I have two Note 3 with the same messages like in the 1st post.
Is there any workaround?

tflucke, 2018-02-18:
Same issue here, and I've been trying for quite a while.  
Had similar issues with Odin on Windows so it might be a Samsung issue rather than a heimdall issue.

louisdoe, 2022-12-13:
same here, any solution ? Thanks