Benjamin-Dobell/Heimdall #392: Invalid argument: –-RECOVERY

monnier, 2017-02-19:
I'm using Debian's heimdall-flash package (version 1.4.1) and can't get the command line to accept my request to flash:

````
% heimdall flash --verbose --no-reboot –-RECOVERY recovery.img 
Invalid argument: –-RECOVERY

Action: flash
Arguments:
    [--<partition name> <filename> ...]
    [--<partition identifier> <filename> ...]
    [--pit <filename>] [--verbose] [--no-reboot] [--resume] [--stdout-errors]
    [--usb-log-level <none/error/warning/debug>]
  or:
    --repartition --pit <filename> [--<partition name> <filename> ...]
    [--<partition identifier> <filename> ...] [--verbose] [--no-reboot]
    [--resume] [--stdout-errors] [--usb-log-level <none/error/warning/debug>]
Description: Flashes one or more firmware files to your phone. Partition names
    (or identifiers) can be obtained by executing the print-pit action.
Note: --no-reboot causes the device to remain in download mode after the action
      is completed. If you wish to perform another action whilst remaining in
      download mode, then the following action must specify the --resume flag.
WARNING: If you're repartitioning it's strongly recommended you specify
        all files at your disposal.
```

I've tried moving arguments around, but to no avail: it doesn't even try to fetch the PIT in order to figure out if the partition name is right or not.

JonnyTech, 2017-02-19:
There should he two hyphens, it looks like you have some sort of dash there

`heimdall flash --RECOVERY filename.img`

monnier, 2017-02-19:
Duh!  Indeed, I had some kind of dash character there.  No idea how it got there (and how I managed to preserve it through the various edits I did to try various alternatives).  I just replaced it with a proper hyphen and lo and behold: it worked!

Thanks!

closed 2017-02-19 by monnier