Benjamin-Dobell/Heimdall #393: Unable to flash modem to my N910T

doesntcount, 2017-03-02:
Recently used Heimdall to flash TWRP onto my N910T. That worked great. I then used TWRP to flash the new Lineage trltetmo build. This ROM, however, doesn't work well with the modem I have, but Heimdall seems to have a bug that is preventing me from being able to flash the modem. Take a look below. Seems like a bug in Heimdall?

/usr/local/bin/heimdall flash --MODEM modem.bin --APNHLOS NON-HLOS.bin  --verbose --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
PIT file data received, read 'empty' packet...
PIT file data receive completed, send kRequestEndTransfer...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
PIT file download successful.

Uploading MODEM
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
0%WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

1%

3%

5%

6%

8%

10%

12%

13%

15%

17%

18%

20%

22%

24%

25%

27%

29%

30%

32%

34%

36%

37%

39%

41%

43%

44%

46%

48%

49%

51%
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

53%

55%

56%

58%

60%

61%

63%

65%

67%

68%

70%

72%

74%

75%

77%

79%

80%

82%

84%

86%

87%

89%

91%

92%

94%

96%

98%

99%

100%
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
MODEM upload successful

Uploading APNHLOS
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
0%WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

5%

11%

17%

22%

28%

34%

40%

45%

51%

57%

63%

68%

74%

80%

85%

91%

97%

100%
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: Failed to unpack received packet.

ERROR: Failed to confirm end of file transfer sequence!
ERROR: APNHLOS upload failed!

Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer.
ERROR: Failed to receive session end confirmation!
Releasing device interface...

doesntcount, 2017-03-02:
I should also mention that this is using the very latest heimdall. Immediately before this, I cloned this repo and built it.

fat-tire, 2017-03-15:
Here's what works with my Note 4.  I've been using this for like 2 years, but I see the original source repository from Saburo Jiro is gone, so I just made this repo.

https://github.com/fat-tire/Heimdall

See if this works for ya.

fat-tire, 2017-03-15:
(Also see [here](https://github.com/Benjamin-Dobell/Heimdall/issues/307#issuecomment-157267616) for build instructions.)