Benjamin-Dobell/Heimdall #394: Heimdall-frontend on El Capitan?

NewbiefromNJ, 2017-03-05:
Hello, 

I found an closed issue that contains Heimdall-1.4.1-Unofficial-Signed, but that's only command line. 

Is there any way to install Heimdall-frontend on OS X 10.11.6 El Capitan?

damionfury, 2017-03-18:
I know it's been 14 days since you wrote this and you may have found another solution, but for anyone who happens to see this, I'll offer my two cents.

I haven't seen any other official dmg but it's incredibly easy to build from source. If you've never done anything like that, here are the steps:

- [ ] **If you don't already have it, install Homebrew and Cask**: https://www.howtogeek.com/211541/homebrew-for-os-x-easily-installs-desktop-apps-and-terminal-utilities/
- [ ] **If you don't already have it, install git**
  - Open your command line and type: `brew install git` (hit enter after any command in this list).

> Helpful hint: When typing any file name or directory name, you can type the first couple characters and hit the tab key. The terminal will attempt to automatically complete what you're typing. This also works with commands but isn't very useful for the very short commands we're using.

- [ ] **Clone the Heimdall repository onto your computer.** 
   1. Using your command line, create a directory for your code: `mkdir ~/code/`
   1. Change to that location: `cd ~/code`
   1. Type `git clone https://github.com/Benjamin-Dobell/Heimdall.git`
- [ ] **Follow the steps in the OSX readme.** 
   1. On your command line, type: `cd Heimdall`
   1. Type: `tail OSX/README.txt`
   1. Follow the steps listed - you could even open a second terminal window and copy-paste the commands.

> Helpful hint: The readme has you install several software packages. If you aren't as comfortable in the terminal, you might find it easier to install them one at a time so that it's easier to figure out if something went wrong. Homebrew is pretty good about describing errors and it will be easier to troubleshoot this way.
>
> Instead of running `brew install thing1 thing2 thing3` try `brew install thing1` and then `brew install thing2`, etc.

- [ ] **Run Heimdall**
   1. Your current directory should be something like ~/code/Heimdall/build. Verify this by typing: `pwd`
   1. Assuming that you are still in the build directory, type: `cd bin/heimdall-frontend.app/Contents/MacOS/`
   1. Launch Heimdall: `./heimdall-frontend`

db17, 2018-02-25:
You can also extract the heimdall-frontend for the official dmg (version 1.4.0). See #369.