Benjamin-Dobell/Heimdall #396: SAMSUNG Galaxy Tab 8.4 SMT-320 protocol initialization failed

hexag1, 2017-03-12:
Trying to flash TWRP onto my Samsung Galaxy Tab 8.4 SMT-320

I keep getting "protocol initialization failed".

I also sometimes get 
>"Downloading device's PIT file...
>ERROR: Failed to send request to end PIT file transfer!
>ERROR: Failed to download PIT file!

etc

Here's my verbose / debug output. IDK what else to do.

> @KTRON-PC Downloads]$ heimdall flash --RECOVERY twrp-3.1.0-0-mondrianwifiue.img --no-reboot --verbose --usb-log-level debug

> Heimdall v1.4.1
> 
> Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
> http://www.glassechidna.com.au/
> 
> This software is provided free of charge. Copying and redistribution is
> encouraged.
> 
> If you appreciate this software and you would like to support future
> development please consider donating:
> http://www.glassechidna.com.au/donate/
> 
> Initialising connection...
> Detecting device...
> [timestamp] [threadID] facility level [function call] <message>
> --------------------------------------------------------------------------------
> [ 0.007568] [0000067f] libusb: debug [libusb_get_device_list] 
> [ 0.007600] [0000067f] libusb: debug [discovered_devs_append] need to increase capacity
> [ 0.007680] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007688] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007692] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007695] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007699] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007705] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007710] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007748] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007753] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007756] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007760] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007764] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.007772] [0000067f] libusb: debug [libusb_open] open 1.65
> [ 0.115550] [0000067f] libusb: debug [usbi_add_pollfd] add fd 10 events 4
> [ 0.115594] [0000067f] libusb: debug [libusb_get_device_descriptor] 
> [ 0.115617] [0000067f] libusb: debug [libusb_alloc_transfer] transfer 0x1503120
> [ 0.115626] [0000067f] libusb: debug [libusb_submit_transfer] transfer 0x1503120
> [ 0.115648] [0000067f] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
> [ 0.115691] [0000067f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 0.115704] [0000067f] libusb: debug [handle_events] poll fds modified, reallocating
> [ 0.115725] [0000067f] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
> [ 0.115787] [0000067f] libusb: debug [handle_events] poll() returned 1
> [ 0.115805] [0000067f] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
> [ 0.115815] [0000067f] libusb: debug [handle_control_completion] handling completion status 0
> [ 0.115827] [0000067f] libusb: debug [disarm_timerfd] 
> [ 0.115837] [0000067f] libusb: debug [usbi_handle_transfer_completion] transfer 0x1503120 has callback 0x7f6959665170
> [ 0.115847] [0000067f] libusb: debug [sync_transfer_cb] actual_length=4
> [ 0.115870] [0000067f] libusb: debug [libusb_free_transfer] transfer 0x1503120
> [ 0.115886] [0000067f] libusb: debug [libusb_alloc_transfer] transfer 0x1503120
> [ 0.115896] [0000067f] libusb: debug [libusb_submit_transfer] transfer 0x1503120
> [ 0.115905] [0000067f] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
> [ 0.115922] [0000067f] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 0.115932] [0000067f] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
> [ 0.116052] [0000067f] libusb: debug [handle_events] poll() returned 1
> [ 0.116074] [0000067f] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=18
> [ 0.116083] [0000067f] libusb: debug [handle_control_completion] handling completion status 0
> [ 0.116090] [0000067f] libusb: debug [disarm_timerfd] 
[... 81 lines cut ...]
> [ 1.117637] [0000067f] libusb: debug [libusb_cancel_transfer] transfer 0x1503120
> [ 1.117733] [0000067f] libusb: debug [disarm_timerfd] 
> [ 1.117749] [0000067f] libusb: debug [handle_events] poll() 3 fds with timeout in 0ms
> [ 1.117757] [0000067f] libusb: debug [handle_events] poll() returned 1
> [ 1.117764] [0000067f] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
> [ 1.117769] [0000067f] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
> [ 1.117774] [0000067f] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
> [ 1.117777] [0000067f] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
> [ 1.117781] [0000067f] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
> [ 1.117785] [0000067f] libusb: debug [disarm_timerfd] 
> [ 1.117788] [0000067f] libusb: debug [usbi_handle_transfer_completion] transfer 0x1503120 has callback 0x7f6959665170
> [ 1.117793] [0000067f] libusb: debug [sync_transfer_cb] actual_length=0
> [ 1.117809] [0000067f] libusb: debug [libusb_free_transfer] transfer 0x1503120
> ERROR: Failed to receive handshake response. Result: -7
> ERROR: Protocol initialisation failed!
> 
> Releasing device interface...
> [ 1.117866] [0000067f] libusb: debug [libusb_release_interface] interface 1
> 
> [ 1.117897] [0000067f] libusb: debug [libusb_close] 
> [ 1.117909] [0000067f] libusb: debug [usbi_remove_pollfd] remove fd 10
> [ 1.117925] [0000067f] libusb: debug [libusb_exit] 
> [ 1.117930] [0000067f] libusb: debug [libusb_exit] destroying default context

>

hexag1, 2017-03-12:
Also this is on Manjaro Linux. But I have the same exact problems on Windows 10.

jmayday, 2017-03-26:
I'm facing same issue with I9506 on Fedora:

```
[/s/a/I9506XXSDPL2_I9506XXUDOJ2_bootmodem]─(1)-> sudo heimdall detect
Device detected
[/s/a/I9506XXSDPL2_I9506XXUDOJ2_bootmodem]─> sudo heimdall print-pit
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...
```

```
[/s/a/I9506XXSDPL2_I9506XXUDOJ2_bootmodem]─> sudo heimdall print-pit --verbose
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Releasing device interface...
```

jmayday, 2017-03-26:
I got it working by compiling heimdall from sources. 
```
dnf install cmake zlib-devel libusb-devel gcc gcc-c++ kf5-kconfigwidgets-devel
```
Then I followed exactly Linux\README. Using heimdall compiled from sources caused no issue like one mentioned in previous posts here and I was able to run it with _print-pit_ option.