Benjamin-Dobell/Heimdall #399: SM-G710L support

ildar, 2017-05-10:
Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
[heimdall.dbg.txt](https://github.com/Benjamin-Dobell/Heimdall/files/990850/heimdall.dbg.txt)
Possible to fix?

JonnyTech, 2017-05-10:
I had the same problem. Built both Heimdall and libusb from source. Working perfectly now.

ildar, 2017-05-11:
Indeed it worked. Thanks a lot!

closed 2017-05-11 by ildar

Benjamin-Dobell, 2017-05-11:
@ildar This is actually quite interesting. It'd be nice to narrow down whether libusb or Heimdall was the issue.

Prior to compiling libusb, had you compiled Heimdall yourself?

JonnyTech, 2017-05-11:
@Benjamin-Dobell on Debian testing I just had to compile Heimdall, but on Ubuntu 14.04 I had to compile both. Either way, Heimdall needs recompiling for most Samsung devices that I have tried. I shall be reinstalling my system in the next few days, is there anything that you would like me to test?

Benjamin-Dobell, 2017-05-11:
@JonnyTech Thanks for the offer. I started writing out instructions, then realised I'm just being lazy and should just do it myself. The information in this thread is useful nonetheless and gives me a couple of leads.

JonnyTech, 2017-05-11:
The offer stands should you require it.

ildar, 2017-05-11:
I didn't build libusb, just Heimdall.
I suggest that some commits after the last release (1.4.1) did the job.
May I suggest making the minor-minor release like 1.4.1.1 so that all the
fixes be released?

Benjamin-Dobell, 2017-05-11:
@ildar Unfortunately, making a minor release (1.4.2) won't necessarily resolve the problem, as I don't maintain the Debian/Ubuntu package. In fact, the package in question (`heimdall-flash`) actually seems like it might have some weird patches applied against it, that might be why it's failing.

Realistically the Debian package author should not be patching Heimdall, if there's something wrong the patch should be submitted as a pull request. If I don't merge said pull request, then there's a good reason for it.

Benjamin-Dobell, 2017-05-11:
1.4.2, ahoy!

Of course, the package maintainer(s) will need to update their package(s). Feel free to reach out to them to let them know 1.4.2 has dropped and that if the package is for an officially supported platform (e.g. Linux) then they shouldn't be applying source code patches.

JonnyTech, 2017-05-11:
Thanks for the quick response on this. Why would package maintainers modify your code? If I could understand that it would give me a better approach for asking them to leave it alone.

Off-topic: I notice the the download always contains a large file (zadig.exe) for Windows users. Would you consider smaller, source only, releases without that binary for the rest of us?