Benjamin-Dobell/Heimdall #401: Protocol initialisation failed on SM-P600 !!

vincegre, 2017-05-19:
Hi

Tried Heimdall on Windows but it get stucks on Initialisation of protocol and on Linux it doesn't detect any device in download mode !!
Samsung Galaxy Tab 10.1 2014 (Europe, Wifi only vension).
Displayed on Tab in download mode:
ODIN MODE
PRODUCT NAME: SM-P600
CURRENT BINARY: Samsung Official
SYSTEM STATUS: Official
Secure Download: Enabled
KNOX WARRANTY VOID: 0
RP SWREV: A3

Thanks,

Vincèn

JoshD522, 2017-06-21:
I have the same problem with my SM-P600 (USA) using Windows 10.  I get this output:

Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "SAMSUNG"
           Product: "Gadget Serial"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
WARNING: Control transfer #2 failed. Result: -7
WARNING: Control transfer #3 failed. Result: -7
WARNING: Control transfer #4 failed. Result: -7
WARNING: Control transfer #5 failed. Result: -7
WARNING: Control transfer #6 failed. Result: -9
ERROR: Failed to send data: "ODIN"
Releasing device interface...

JonnyTech, 2017-06-22:
Please try the latest release v1.4.2

JoshD522, 2017-06-22:
Is there a 1.4.2 build?  Or, do I need to setup a build environment and build from source?  I see the http://www.glassechidna.com.au/ page still has 1.4.0.