Benjamin-Dobell/Heimdall #402: Firmware Backup/Dump

hy-l, 2017-06-22:
hi, all
can I use heimdall to dump the current firmware on the phone to a file? ( e.g. /system, /efs, /data, etc. ) 
at this point, i just want to see what inside the /efs from bootloader, and make a backup  without rooting the device. Is it possible? How?
If heimdall dose not have the backup feature, could you please add this in the future?

thanks. :)

Benjamin-Dobell, 2017-06-22:
It's not possible unfortunately - although for security reasons I'm somewhat thankful.

Heimdall did at one point have _theoretical_ support for dumping firmware, however no publicly available phones actually supported the feature. Heimdall communicates with Loke software running on the device, so Heimdall can only support what Loke supports - all publicly released versions of Loke (that I've investigated) do not support this feature.

closed 2017-06-22 by Benjamin-Dobell

hy-l, 2017-06-22:
okay, maybe the only way is rooting my phone.
also, i never expect to get an answer so fast. thank you. ;)