Benjamin-Dobell/Heimdall #405: Heimdall frontend ERROR: Failed to start Heimdall!

MurKit, 2017-07-12:
I'm on Mint 18.1, all the libs are installed.
When I compile release 1.4.2 or newest from GitHub, I later get an error:
FRONTEND ERROR: Failed to start Heimdall!

Build log:
```
$ make
Scanning dependencies of target pit
[  3%] Building CXX object libpit/CMakeFiles/pit.dir/source/libpit.cpp.o
[  6%] Linking CXX static library libpit.a
[  6%] Built target pit
Scanning dependencies of target heimdall
[ 10%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Arguments.cpp.o
[ 13%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o
In file included from /home/machine/Downloads/si9500/Heimdall-1.4.2/heimdall/source/BridgeManager.cpp:50:0:
/home/machine/Downloads/si9500/Heimdall-1.4.2/heimdall/source/SendFilePartPacket.h: In constructor ‘Heimdall::SendFilePartPacket::SendFilePartPacket(FILE*, unsigned int)’:
/home/machine/Downloads/si9500/Heimdall-1.4.2/heimdall/source/SendFilePartPacket.h:49:44: warning: ignoring return value of ‘size_t fread(void*, size_t, size_t, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
     (void)fread(data, 1, bytesToRead, file);
                                            ^
[ 16%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/ClosePcScreenAction.cpp.o
[ 20%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DetectAction.cpp.o
[ 23%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DownloadPitAction.cpp.o
[ 26%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/FlashAction.cpp.o
[ 30%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/HelpAction.cpp.o
[ 33%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/InfoAction.cpp.o
[ 36%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Interface.cpp.o
[ 40%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/main.cpp.o
[ 43%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/PrintPitAction.cpp.o
/home/machine/Downloads/si9500/Heimdall-1.4.2/heimdall/source/PrintPitAction.cpp: In function ‘int Heimdall::PrintPitAction::Execute(int, char**)’:
/home/machine/Downloads/si9500/Heimdall-1.4.2/heimdall/source/PrintPitAction.cpp:142:64: warning: ignoring return value of ‘size_t fread(void*, size_t, size_t, FILE*)’, declared with attribute warn_unused_result [-Wunused-result]
   (void)fread(pitFileBuffer, 1, localPitFileSize, localPitFile);
                                                                ^
[ 46%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Utility.cpp.o
[ 50%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/VersionAction.cpp.o
[ 53%] Linking CXX executable ../bin/heimdall
[ 53%] Built target heimdall
Scanning dependencies of target heimdall-frontend_automoc
[ 56%] Automatic moc for target heimdall-frontend
Generating moc_aboutform.cpp
Generating moc_mainwindow.cpp
[ 56%] Built target heimdall-frontend_automoc
[ 60%] Generating qrc_mainwindow.cpp
RCC: Warning: No resources in '/home/machine/Downloads/si9500/Heimdall-1.4.2/heimdall-frontend/mainwindow.qrc'.
[ 63%] Generating ui_mainwindow.h
[ 66%] Generating ui_aboutform.h
Scanning dependencies of target heimdall-frontend
[ 70%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/aboutform.cpp.o
[ 73%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/Alerts.cpp.o
[ 76%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/FirmwareInfo.cpp.o
[ 80%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/main.cpp.o
[ 83%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/mainwindow.cpp.o
[ 86%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/PackageData.cpp.o
[ 90%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/Packaging.cpp.o
[ 93%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/qrc_mainwindow.cpp.o
[ 96%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/heimdall-frontend_automoc.cpp.o
[100%] Linking CXX executable ../bin/heimdall-frontend
[100%] Built target heimdall-frontend
```

aletheus, 2017-08-01:
i'm getting the same.

bugith, 2017-09-13:
So do I on ubuntu 14.04.5 : I first installed from repo (1.4.0-2) both heimdall-flash & heimdall-flash-frontend.
As I could do anything in my SM-N7505 with this version I compiled 1.4.2 sources as of today 2017:09:13 then I succeeded in command line both in print-pit > file.pit.txt and download-pit --ouput file.pit, as frontend failed
So far so good, trying to go further with frontend, I noticed checking Help/About Heimdall that was reported "Heimdall Frontend V.1.4.2 Copyright 2010-2017..." and "Heimdall (command line) V.1.4.0 Copyright 2010-2017..." !!!! It's I forgot to uninstall the 2 v1.4.0 packages with my packages manager.... then so did I ... then the "Heimdall frontend ERROR: Failed to start Heimdall!".
Now, looking Help/About Heimdall: "Heimdall Frontend V.1.4.2 Copyright © 2010-2017..." and "Heimdall (command line) [NO VERSION HERE] Copyright © 2010-2017..." 
It seems frontend can't find the CL binary ....

aletheus, 2017-09-13:
I was able to get cl working with changes from the gs5 branch, and that was
good enough for me. Thanks for all your work. I truly appreciate having a
linux tool for samsung devices,

-c

bugith, 2017-09-14:
And I was able to have frontend reporting 1.4.2 both for itself and cl by copying the 2 bins in /usr/bin :)
Then I could flash TWRP to my phone with the heimdall GUI

No more "ERROR: Failed to start Heimdall!"

MurKit, 2017-09-15:
@bugith Could you please tell what exactly did you copy?

bugith, 2017-09-15:
both 2 heimdall & heimdall-fronted files that compilation created into ~/Heimdall/heimdall/build/bin (copied with **sudo cp path/heim* /usr/bin/** on my ubuntu 14.04 which put them in the path variable, no more requiring typing the path (./heim.. or ~/Heim..../heim...)

dandv, 2017-10-14:
This looks like a dupe of [#356](https://github.com/Benjamin-Dobell/Heimdall/issues/356#issuecomment-336617581), for which I left a comment with a simple working solution.

rlabrovi, 2017-10-26:
I did same think @bugith did. After compiling I copied 2 files from ~/heimdall/build/bin to /usr/bin and run them with sudo privileges. No more "ERROR: Failed to start Heimdall!"