Benjamin-Dobell/Heimdall #406: Samsung Galaxy Core Prime (SM-G360F) ERROR: Failed to receive handshake response. Result: -7

FinKsu, 2017-07-22:
When I run `sudo heimdall print-pit --verbose` I get this:
`Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Releasing device interface...
`
Using Ubuntu 16.10

shaqaruden, 2017-07-24:
When I get that I restart my device into Download Mode, just press and hold the Pwr, vol Down, and home button until the warning screen comes up again, then it should work.

FinKsu, 2017-07-24:
I built 1.4.2, and it worked. download-pit didn't work first time, but worked after phone reboot. It gave me some errors but still successfully downloaded pit.

closed 2017-07-26 by FinKsu