Benjamin-Dobell/Heimdall #407: Samsung Galaxy A520W

shaqaruden, 2017-07-24:
I'm trying to root my device but I'm getting this error when trying to save the .PIT file. I have installed all required dependencies, tried different USB ports. I am running Lubuntu 17.04

> sudo heimdall download-pit --output  ~/Desktop/A520W.pit --verbose --no-reboot --stdout-errors
Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...
libusb: error [op_set_interface] setintf failed error -1 errno 71
ERROR: Setting up interface failed!
ERROR: Setting up interface failed!
Releasing device interface...

shaqaruden, 2017-07-24:
So I built Heimdall from source (upgrading to 1.4.2) and I got a step further but not sure if the .PIT was created correctly because I am given a warning. I'm guessing not.

sudo ./heimdall download-pit --output  ~/Desktop/A520W.pit --verbose --no-reboot --stdout-errors
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "SAMSUNG"
           Product: "Gadget Serial"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
WARNING: Empty bulk transfer after receiving packet failed. Continuing anyway...
WARNING: Empty bulk transfer after receiving packet failed. Continuing anyway...
PIT file download successful.

Ending session...
Releasing device interface...

shaqaruden, 2017-07-24:
Here is the contents of my .PIT file

./heimdall print-pit --file ~/Desktop/A520W.pit 
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Entry Count: 30
Unknown 1: 1598902083
Unknown 2: 844251476
Unknown 3: 21324
Unknown 4: 14153
Unknown 5: 14392
Unknown 6: 48
Unknown 7: 0
Unknown 8: 0

--- Entry #0 ---
Binary Type: 0 (AP)
Device Type: 2 (MMC)
Identifier: 80
Attributes: 2 (STL Read-Only)
Update Attributes: 1 (FOTA)
Partition Block Size/Offset: 0
Partition Block Count: 8192
File Offset (Obsolete): 0
File Size (Obsolete): 0
Partition Name: BOOTLOADER
Flash Filename: sboot.bin
FOTA Filename: 

--- Entry #1 ---
Binary Type: 0 (AP)
Device Type: 2 (MMC)
Identifier: 90
Attributes: 2 (STL Read-Only)
Update Attributes: 1 (FOTA)
Partition Block Size/Offset: 0
Partition Block Count: 8192
File Offset (Obsolete): 0
File Size (Obsolete): 0
Partition Name: CM
Flash Filename: cm.bin
FOTA Filename: 

--- Entry #2 ---
Binary Type: 0 (AP)
Device Type: 2 (MMC)
Identifier: 91
Attributes: 2 (STL Read-Only)
Update Attributes: 1 (FOTA)
Partition Block Size/Offset: 6144
Partition Block Count: 2048
File Offset (Obsolete): 0
File Size (Obsolete): 0
Partition Name: ECT
Flash Filename: ect.bin
FOTA Filename: 

--- Entry #3 ---
Binary Type: 0 (AP)
Device Type: 2 (MMC)
[... 349 lines cut ...]
Device Type: 2 (MMC)
Identifier: 24
Attributes: 5 (Read/Write)
Update Attributes: 1 (FOTA)
Partition Block Size/Offset: 13346816
Partition Block Count: 2048
File Offset (Obsolete): 0
File Size (Obsolete): 0
Partition Name: NAD_REFER
Flash Filename: nad_refer.bin
FOTA Filename: 

--- Entry #29 ---
Binary Type: 0 (AP)
Device Type: 2 (MMC)
Identifier: 25
Attributes: 5 (Read/Write)
Update Attributes: 5 (FOTA)
Partition Block Size/Offset: 13348864
Partition Block Count: 0
File Offset (Obsolete): 0
File Size (Obsolete): 0
Partition Name: USERDATA
Flash Filename: userdata.img
FOTA Filename: remained

shaqaruden, 2017-07-24:
Tried flashing SuperSU and I receive this. I am running as superuser

Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "SAMSUNG"
           Product: "Gadget Serial"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Beginning session...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer.
Releasing device interface...

Benjamin-Dobell, 2017-07-24:
Don't forget to make sure that after you've run a command with `--no-reboot` that the next command includes `--resume`. I think that might be the issue here.

shaqaruden, 2017-07-24:
i even tried with heimdall-frontend with those options selected and I still have the issue.

Benjamin-Dobell, 2017-07-24:
@shaqaruden That specific error you most recently reported just doesn't make a heap of sense seems as earlier you posted output demonstrating that beginning a session works fine. At that point the commands are yet to do anything differently.

So either the device was not ready to receive packets (need to reboot back into download mode) or there are USB reliability issues with your PC - because I can't see why it would have worked fine earlier but not now with the exact same binary.

It's probably worth noting that users have in past had issues with libusb and/or their kernel i.e. Users had no success with Heimdall until they updated their OS. I don't know specifically why that would be the case, but unfortunately whatever the reason it's out of my/Heimdall's hands.

I should note that I'm not actually sure whether Galaxy A5 works with Heimdall. However, the fact you were able to download your PIT is a positive sign; usually it's all or nothing.

Benjamin-Dobell, 2017-07-24:
Updating your OS may be a non-option - or perhaps it's already up-to-date. However, you can try downloading and building the latest official libusb release from source, then rebuild Heimdall. No promises, but it's worth a shot.

shaqaruden, 2017-07-24:
I will attempt it