Benjamin-Dobell/Heimdall #410: ERROR: Failed to detect compatible download-mode device (Galaxy S7)

stemy2, 2017-08-08:
Hello,

I have this message every time i try a detection, even if my phone is on download mode, as if it was not connected.

What have i to do to sojve this ?

Thanks.

JonnyTech, 2017-08-08:
Which version of heimdall and which operating system? Have you tried different USB port and USB cable?

stemy2, 2017-08-08:
I authorized my PC to debug my phone and then, it is detected. But now, i have a new error when i try to print the PIT file: «ERROR: Failed to send data!Releasing device interface...».

JonnyTech, 2017-08-08:
What command did you use? Please post your terminal contents. And also answer the previous questions.

stemy2, 2017-08-08:
I use the version 1.4.0. I'm on Ubuntu. I tried countless cables and ports and i have the same message.

`stem@Stem-PC:~$ sudo su
[sudo] password for stem:
root@Stem-PC:/home/stem# heimdall print-pit
Heimdall v1.4.0

Copyright (c) 2010-2013, Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...
libusbx: error [op_set_interface] setintf failed error -1 errno 71
ERROR: Setting up interface failed!
Releasing device interface...

root@Stem-PC:/home/stem#`

JonnyTech, 2017-08-08:
Try version 1.42

stemy2, 2017-08-08:
OK, but he latest version of the package bitbucket is 1.4.0. Does it means that i'm forced to compile it manually ?

JonnyTech, 2017-08-08:
Yes, it is not difficult, see appendix B from https://raw.githubusercontent.com/Benjamin-Dobell/Heimdall/master/Linux/README

stemy2, 2017-08-09:
OK. So i compiled the app, but «heimdall version» was still returning «1.4.0». So i uninstalled heimdall and i compiled it again. Now, the commad gives me «bash: heimdall : command not found».

JonnyTech, 2017-08-09:
Well done. You have built the new version. But it is not installed in the system. To run it go to folder where new built file is and use ./heimdall

stemy2, 2017-08-09:
OK. I tried the command «adb devices» and now, i have this error message»

`adb server version (31) doesn't match this client (39); killing...`

stemy2, 2017-08-09:
Ok, i'm now in the recovery mode and tried to wipe. Now, it shws me some error messages.

`Could not mount /data and unable to find crypte footer.
failed to mount '/data' (invalid argument)
unable to recreate /data/media folder.
Updating partition details...
failed to mount '/data' (invalid argument)
..done
Unable to mount storage
failed to mount '/data' (invalid argument)
Full SElinux support is present
failed to mount '/data/media/TWRP/.twrps
MTP enabled
Formatting system using make_ext4fs...
failed to mount '/data' (invalid argument)
Unable to wipe /data
Updating partition details...
failed to mount '/data' (invalid argument)
...done
Unable to mount storage`

JonnyTech, 2017-08-09:
Heimdall only works in bootloader (aka fastboot) mode, not in recovery. What commands are you typing? Please copy and paste everything from the terminal session.

stemy2, 2017-08-09:
OK.

In fact, i followed all the steps mentioned [on this page](https://wiki.lineageos.org/devices/herolte/install). After preparing the install, i booted my device in Download mode and then i typed this:

`heimdall flash --RECOVERY twrp-3.1.1-0-herolte.img --no-reboot`

I waited for a little time, thinking that the blue bar would desappear, but even after some minutes, it didn't, so i rebooted into recovery mode. I typed the following command

`adb push lineage-14.1-20170801-nightly-herolte-signed.zip /sdcard/`

and then i clicked on «Advanced Wipe» and tried to wipe Cache, System and Data partitions. But it failed and now, i'm stuck at this step.

So according your comment, the howto i mentioned in an URL is totajjy wrong, am i right ?

JonnyTech, 2017-08-09:
OK, so if I understand correctly, Heimdall worked and you have been able to successfully install TWRP.

You only need `--no-reboot` if you are going to issue another command. So just `./heimdall flash --RECOVERY twrp-3.1.1-0-herolte.img` will flash the partition and automatically reboot. Remember to hold down: Volume Up, Home and Power - to get to recovery mode.

If you cannot mount /data then that is a TWRP bug where it cannot handle the encrypted partition correctly (I have already reported the problem because it happened to me). To access data you need to format it: In TWRP go to WIPE >> FORMAT DATA then type yes to wipe it.

So the correct order with your current setup is:

- boot to TWRP
- if you have the error of mounting /data then format it as shown above (this will wipe your files!)
- select WIPE
- swipe to factory reset
- copy lineageos file to /sdcard with adb
- optionally copy gapps to /sdcard with adb
- select INSTALL
- choose the lineage file (and optionally the gapps zip file)
- select INSTALL IMAGE
- reboot system

stemy2, 2017-08-09:
OK, now it is installed and works fine.

Lot of thanks.

closed 2017-08-09 by stemy2