Benjamin-Dobell/Heimdall #413: ERROR: Protocol initialisation failed!

aliensonneptune, 2017-08-23:
Have been trying for 3 hours to flash TWRP to a SMA500FU, and I don't know if it's a problem with the fact it's an A5, or if it's the build I'm using or what?

Here's the output of detect, and download-pit.

```bash
aliensonneptune@tinfoilhats:/$ sudo heimdall detect
Device detected
aliensonneptune@tinfoilhats:/$ sudo heimdall download-pit --output pit.pit
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Protocol initialisation failed!

Releasing device interface...

aliensonneptune@tinfoilhats:/$ 
```

Can anyone help me?

PS, I'm an android noob, only been using it for 4 weeks.

PPS, I'm on a USB3, and the phone is on the 6.0.1 UD.

JonnyTech, 2017-08-23:
Use version 1.4.2 - I recommend compiling yourself

aliensonneptune, 2017-08-24:
@JonnyTech I've built it from the repo, and I still get 1.4.1, built yesterday? When was 1.4.2 pushed?

JonnyTech, 2017-08-24:
Have you tried rebooting the phone and another USB port and/or cable?

Which version of libusb do you have installed? Version 1.0.21 works for me:

```
user@computer:~$ dpkg -l libusb-1*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  libusb-1.0-0:a 2:1.0.21-2   amd64        userspace USB programming library
ii  libusb-1.0-0-d 2:1.0.21-2   amd64        userspace USB programming library
ii  libusb-1.0-doc 2:1.0.21-2   all          documentation for userspace USB p
```

Make sure that you do not run Heimdall as root/sudo - I have had problems:

```
user@computer:~/Heimdall-master/build/bin$ ./heimdall info
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Heimdall utilises libusbx for all USB communication:
    http://www.libusb.org/

libusbx is licensed under the LGPL-2.1:
    http://www.gnu.org/licenses/licenses.html#LGPL
```
What are the output of these commands:

```
user@computer:~/Heimdall-master/build/bin$ ./heimdall detect
user@computer:~/Heimdall-master/build/bin$ ./heimdall download-pit --output pit.pit
user@computer:~/Heimdall-master/build/bin$ cat pit.pit
```

aliensonneptune, 2017-08-24:
Tried reboot, other USB cable, ./heimdall detect came up with nothing, tried the 1st one again, and like before it detected on the original cable. Turns out my backup cable is faulty. After a complete sudo apt update && sudo apt upgrade, IT GOT THE PIT FILE! Here's my proof, though the .pit looks messy as hell, don't know if that's good?
```bash
aliensonneptune@tinfoilhats:~/musicphone/Heimdall/bin$ ./heimdall detect
Device detected

aliensonneptune@tinfoilhats:~/musicphone/Heimdall/bin$  dpkg -l libusb-1*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                    Version          Architecture     Description
+++-=======================-================-================-====================================================
ii  libusb-1.0-0:amd64      2:1.0.20-1       amd64            userspace USB programming library
ii  libusb-1.0-0-dbg:amd64  2:1.0.20-1       amd64            userspace USB programming library development files
ii  libusb-1.0-0-dev:amd64  2:1.0.20-1       amd64            userspace USB programming library development files
ii  libusb-1.0-doc          2:1.0.20-1       all              documentation for userspace USB programming

aliensonneptune@tinfoilhats:~/musicphone/Heimdall/bin$ ./heimdall download-pit --output pit.pit
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
PIT file download successful.

Ending session...
Rebooting device...
Releasing device interface...

aliensonneptune@tinfoilhats:~/musicphone/Heimdall/bin$ cat pit.pit
v�4COM_TAR2MSM8916 xAPNHLOSNON-HLOS.bin���MODEMmodem.bin�cSBL1sbl1.mbn�g@DDR�gABOOTaboot.mbn�wRPMrpm.mbn�{QSEEtz.mbn�QHEEhyp.mbn	��FSG
�� secsec.dat
                �� TPAD
@pEFSefs.img.ext4�MODEMST1nvrebuild1.bin�MODEMST2nvrebuild2.bin�hBOOTboot.imgHxRECOVERYrecovery.img�(FOTA��7BACKUP�FSC�7SSD8@PERSISTpersist.img.ext4xHPERSDATApersdata.img.ext4�KSYSTEMsystem.img.ext4�P@CACHEcache.img.ext4W@HIDDENhidden.img.ext4@]USERDATAuserdata.img.ext4remainedF"PGPTpgpt.imgG"PITMSM8974.pitH2 MD5md5.imgI���!SGPTsgpt.imgI�W!����%C��4=��
                                                        	��o!���p�%�<\T��Rq�
                                                                                      �I']enw��T?8�Z�{E��cV	��E7O&���Q��B��4��4�����=
                         ό�'d�T}?��O���28��~�ڐe�V)G��<��`���y���d�"���A�9e�!Q�'���)�&!ũ��Գ�M��Mq}�iah���H�
�2�樦E�<�K6��OP�`G8�qА�?Ȁ ؖ,���ͫ����d5�+����0�
```

aliensonneptune, 2017-08-24:
I sent pretty much that command to my phone, and it rebooted, actually I tried doing it using --<Numeric Identifier> instead of partition name, and accidentally flashed to 18, which was BACKUP, the phone still boots though. Here is the output of the console command, now how do I boot into the recovery?

```
aliensonneptune@tinfoilhats:~/musicphone/Heimdall$ ./bin/heimdall flash --RECOVERY ort.img 
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
PIT file download successful.

Uploading RECOVERY
100%
RECOVERY upload successful

Ending session...
Rebooting device...
Releasing device interface...

aliensonneptune@tinfoilhats:~/musicphone/Heimdall$ 
```

JonnyTech, 2017-08-24:
Well done! To boot to recovery after flashing press Power, Volume Up and Home Buttons at the same time. If the phone boots normally then you pressed the buttons too late. At this point your newly flashed firmware has been overwritten by the system. So you need to flash it again from Heimdall in bootloader mode then hold the the buttons as it finished and restarts.

Or you can use the `--no-reboot` flag to not reboot automatically which makes it easier to get ready and power down the device manually and boot up holding down the keys mentioned above.

aliensonneptune, 2017-08-24:
I have tried booting, but it just shows the boot logo of my phone normally and some text in the top left. I thought it was loading TWRP in, but I left it for 20-25 mins then lost hope and held down all the buttons to boot normally? Is it the image or what? PS I'm using the latest image from [here](http://techerrata.com/browse/twrp2/twrp), have no idea if that's what I'm meant to be using?

aliensonneptune, 2017-08-24:
Okay, found an Odin-flashable TWRP .tar.md5, unpacked it, and found an .img file, whick looked flashable. Flashed it, worked flawlessly, booted TWRP, found my SuperSU .zip, installed.... aaaaaaaaand now I'm stuck on the loading screen :/

spmp, 2017-08-24:
Has anyone else got this issue? I am having exactly the same thing:
```
./heimdall print-pit --verbose --no-reboot
...
--- Entry #28 ---
Binary Type: 0 (AP)
Device Type: 2 (MMC)
Identifier: 73
Attributes: 5 (Read/Write)
Update Attributes: 1 (FOTA)
Partition Block Size/Offset: 61071327
Partition Block Count: 33
File Offset (Obsolete): 0
File Size (Obsolete): 0
Partition Name: SGPT
Flash Filename: sgpt.img
FOTA Filename: 

Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...

Releasing device interface...
```
And my `libusb`
```
ii  libusb-0.1-4:amd64                                          2:0.1.12-30                                       amd64        userspace USB programming library
ii  libusb-1.0-0:amd64                                          2:1.0.21-1                                        amd64        userspace USB programming library
ii  libusb-1.0-0:i386                                           2:1.0.21-1                                        i386         userspace USB programming library
ii  libusb-1.0-0-dev:amd64                                      2:1.0.21-1                                        amd64        userspace USB programming library development files
ii  libusb-1.0-doc                                              2:1.0.21-1                                        all          documentation for userspace USB programming
ii  libusb-dev                                                  2:0.1.12-30                                       amd64        userspace USB programming library development files
```
Heimdall buit from git, ubuntu 17.04.

The phone is a Samsung Galaxy Note 3, SM-N900V. Any other solution would be amazing!

aliensonneptune, 2017-08-24:
Wiped dalvik/cache/tried hard resetting, SOMEBODY HELP, PLEASE, MY PHONE ISN'T LOADING IN.

JonnyTech, 2017-08-24:
Did it boot after flashing TWRP, before installing SuperSU?

aliensonneptune, 2017-08-24:
Yes, and in the time since I last commented, I figured out that if TWRP still booted in, I could use ADB sideload to load in a new ROM, which could either solve my problems or make them worse.

I'm an optimist. 

SO.

I sideloaded a Lineage OS 14 ROM for the A500xx series, crossed my fingers and booted. Soon enough I was greeted by the amazing sight of a Lineage OS welcome screen. Currently sideloading opengapps for my device. 

Somehow flashing TWRP to root, and failing, then trying to fix it gave me face unlock and a stock (finally) theme? Anyway, couldn't have done it without your help, thanks a lot @JonnyTech

JonnyTech, 2017-08-24:
@aliensonneptune - great result, glad that I could help, please close this issue now

@spmp - do not hijack issues, please start your own using the new issue button

spmp, 2017-08-24:
@JonnyTech Sorry buddy, I thought it was the same issue as having the same error etc. Will start a new issue 8)

dandv, 2017-10-14:
Same problem here, "Protocol initialisation failed".

I've noticed though that the first `heimdall` command did execute successfully... so I kept rebooting the phone out of download mode (hold all buttons for ~5 seconds) then back into download mode, then issue the next `heimdall` command.

**TL;DR** only the first command after booting into download mode works!

AchrafKassioui, 2017-11-24:
Thanks dandv! Indeed, only first command after booting into download works.

NacreousDawn596, 2021-10-03:
um, help? I have the same error but I'm trying with the kernel `sudo heimdall flash --kernel zImage` but it told me :
`Initialising protocol...
ERROR: Failed to send handshake!ERROR: Protocol initialisation failed!

Releasing device interface...

`

is there any solution please? _I'm trying to root my samsung A2 core_

JonnyTech, 2021-10-03:
Did you issue that command directly after booting the phone into download mode? If you read this thread you will learn that only one command after each reboot works.

NacreousDawn596, 2021-10-03:
owh, okay

NacreousDawn596, 2021-10-03:
I have tried everything here, I always get:
```
Initialising protocol...
ERROR: Protocol initialisation failed!

Releasing device interface...
```

JonnyTech, 2021-10-04:
Are you sure that you are in odin/download mode?: https://www.androidsettings.net/samsung-galaxy-a2-core-download-mode/

[image]

amo13, 2021-10-04:
You might also want to try a [newer build](https://github.com/amo13/Heimdall/releases/tag/v1.4.2) from a fork with a few fixes. It may or may not help...

PsySc0rpi0n, 2024-03-01:
I am trying "only" to `print-pit` for now a Samsung Galaxy S21 (G991B), under Debian Bookworm, and I also get the same `libusb` errors:
```
Initialising protocol...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer.
ERROR: Failed to send handshake!ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!
```

Heimdall version 1.4.2
```
$ heimdall version
v1.4.2
$
```

and libusb versions:
```
$ dpkg -l libusb*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                     Version      Architecture Description
+++-========================-============-============-=====================================================================
ii  libusb-0.1-4:amd64       2:0.1.12-32  amd64        userspace USB programming library
ii  libusb-1.0-0:amd64       2:1.0.26-1   amd64        userspace USB programming library
ii  libusb-1.0-0-dev:amd64   2:1.0.26-1   amd64        userspace USB programming library development files
ii  libusb-1.0-doc           2:1.0.26-1   all          documentation for userspace USB programming
ii  libusb-dev               2:0.1.12-32  amd64        userspace USB programming library development files
ii  libusbredirhost1:amd64   0.13.0-2     amd64        Implementing the usb-host (*) side of a usbredir connection (runtime)
ii  libusbredirparser1:amd64 0.13.0-2     amd64        Parser for the usbredir protocol (runtime)
```

Heimdall can see the device:
```
$ heimdall detect
Device detected
```

```

A bit weird here but it's there
$ lsusb
Bus 002 Device 008: ID 04e8:685d Samsung Electronics Co., Ltd GT-I9100 Phone [Galaxy S II] (Download mode)
```

Anything else I can try?

julianfoad, 2024-05-25:
I have written more about this error here: [Troubleshooting: Heimdall Fails at Initialising Protocol](https://wrily.foad.me.uk/troubleshooting-heimdall-fails-at-initialising-protocol)

In short:

- Scenario 1: the first command in bootloader/download mode works, and subsequent commands fail -> either it's a limitation of the device, or properly using Heimdall's "--no-reboot" and "--resume" options may overcome it.
- Scenario 2: every command (except "detect") in bootloader/download mode fails and you're on an OS similar to Ubuntu (that may well include Debian) -> there is a fix that's not in v1.4.2 in this repo, but that is available in various other people's updated versions of Heimdall, such as [Grimler's Heimdall](https://git.sr.ht/~grimler/Heimdall) .

julianfoad, 2024-05-25:
Possibly same issue as fixed by PR #478