Benjamin-Dobell/Heimdall #414: ERROR: libusb error -7 whilst receiving bulk transfer

spmp, 2017-08-24:
I am having a similar issue to https://github.com/Benjamin-Dobell/Heimdall/issues/413 trying to install TWRP on a _Samsung Galaxy Note 3_, _SM-N900V_ (Verizon version)
The USB device is USB 3. 
I have USB2 devices if this will help.

Original log of trying to flash is gone, but here are the snippets of trying to get the PIT file:
Note: I have changed the permissions fo the USB device so I can perform this as my normal user as I have heard that doing this as root can be problematic.  
```
./heimdall print-pit --verbose --no-reboot
...
--- Entry #28 ---
Binary Type: 0 (AP)
Device Type: 2 (MMC)
Identifier: 73
Attributes: 5 (Read/Write)
Update Attributes: 1 (FOTA)
Partition Block Size/Offset: 61071327
Partition Block Count: 33
File Offset (Obsolete): 0
File Size (Obsolete): 0
Partition Name: SGPT
Flash Filename: sgpt.img
FOTA Filename: 

Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...

Releasing device interface...
```
And my `libusb`
```
ii  libusb-0.1-4:amd64                                          2:0.1.12-30                                       amd64        userspace USB programming library
ii  libusb-1.0-0:amd64                                          2:1.0.21-1                                        amd64        userspace USB programming library
ii  libusb-1.0-0:i386                                           2:1.0.21-1                                        i386         userspace USB programming library
ii  libusb-1.0-0-dev:amd64                                      2:1.0.21-1                                        amd64        userspace USB programming library development files
ii  libusb-1.0-doc                                              2:1.0.21-1                                        all          documentation for userspace USB programming
ii  libusb-dev                                                  2:0.1.12-30                                       amd64        userspace USB programming library development files
```
Heimdall built from git (pulled yesterday), ubuntu 17.04.

JonnyTech, 2017-08-24:
Can you try again and post the complete log including the command that you entered. The fact that you are getting the PIT is a good sign. Have you tried actually flashing the recovery image? I would try with a USB2 port.

spmp, 2017-08-24:
Did this actually work???
Doing it remote on a mates phone so cannot reboot to check 8)
```
./heimdall flash --RECOVERY ../twrp-3.1.1-0-hlte.img --no-reboot --verbose | tee /tmp/heimdall.log
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
PIT file download successful.

Uploading RECOVERY
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
0%WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

100%
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: Failed to unpack received packet.

ERROR: Failed to confirm end of file transfer sequence!
ERROR: RECOVERY upload failed!

Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...

Releasing device interface...
```

JonnyTech, 2017-08-24:
Very strange. It does indeed look like it completed the transfer as it managed to get to 100% but whether the phone accepted it is uncertain because it showed that the upload failed.

It smells of a usb setup problem, I recommend building libusb from source, get it from https://sourceforge.net/projects/libusb/files/

Also, if you are using --no-reboot then remember to use --resume on the next command otherwise you will experience problems.

JonnyTech, 2017-08-24:
PS: I notice that you are trying to install an hlte file on an hltevzw device, if your original post is correct, download the correct image from https://twrp.me/devices/samsunggalaxynote3verizon.html

spmp, 2017-08-24:
Whilst I am waiting for the dist-upgrade to get deps for libusb...
```
./heimdall flash --RECOVERY  twrp-3.0.2-0-hltevzw-4.4.img --resume --no-reboot --verbose | tee /tmp/heimdall.log
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer.
Releasing device interface...
```
Less joy than the first time. Thanks fo rhte heads up. I was looking at lineageos and they seemed to think that was the twrp to get.. hope I havn't bricked it 8)

I guess it may be because I didnt o a `--resume` on my second command...

JonnyTech, 2017-08-24:
If you did not use --resume then you must reboot the phone before trying again. I never use --no-reboot as it always seems to cause me problems but I guess that you have no choice if you are working remotely. Get someone local to the device to reboot it back into bootloader mode.

spmp, 2017-08-24:
aaah crap. TO make life easy for me, can you please instruct as to how to build an isolated version of libusb and use this with Heimdall. Ubuntu has too many dependencies on `libusb-1.0.0`

JonnyTech, 2017-08-24:
Before trying that, did you reboot the phone and try flashing without --no-reboot? I shall be back after sleep zzzzz....

spmp, 2017-08-25:
Yes. Cheers for the help @JonnyTech 
WIll smash at it a bit more... This phone is a real PITA!

spmp, 2017-08-25:
I cheated and just ripped libusb-1.0-0 out with dpkg --force-all... still the same errors and less progress than before again...
```
./heimdall flash --RECOVERY  twrp-3.0.2-0-hltevzw-4.4.img --resume --no-reboot --verbose
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Beginning session...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
ERROR: libusb error -7 whilst receiving bulk transfer.
Releasing device interface...
```

JonnyTech, 2017-08-25:
Recompile Heimdall after replacing libusb. Keep the command line simple. Reboot both phone and computer then try: `./heimdall flash --RECOVERY twrp-3.0.2-0-hltevzw-4.4.img`

If that does not work can you explain what you did here:

> I have changed the permissions fo the USB device so I can perform this as my normal user

Heimdall should work as a normal user without any system modification.

spmp, 2017-08-29:
I tried all I could With libusb as directed and rebooting every combination.... My friend gave up... maybe he will be back with the phone in a week or so...

By changing permissions i mean
```
chmod myuser:myuser /dev/path/to/usb/device
```
Where the device is what heimdall reports that it does not have permission to write to.

JonnyTech, 2017-08-29:
Thanks for replying - hopefully you can try the device locally next time too...

Megver83, 2017-11-27:
I'm having exactly the same issue in Parabola GNU/Linux-libre (Arch-based, the heimdall pkg comes from Arch).
I also tried with different USB cables but nothing changed. The strange thing is that when printing/downloading the PIT I simply got the `WARNING: Empty bulk transfer after sending packet failed.` warning, but when I tried to flash something I got the error from the title, and then when trying to print the PIT again I got the libusb error, not the typical warning. Then just re-rebooted in odin mode and I could print the PIT almost without problem (I say "almost" because of the warning)

TealG, 2017-12-10:
I ran into this error using Heimdall 4.1.2 with an I9105P device
when the `--resume` argument was present
in any of the commands (print-pit, download-pit, flash):

<pre>
# heimdall print-pit --verbose --no-reboot --resume
[...]

Beginning session...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
</pre>

Without `--resume` the first command sent to the device works (with some warnings):

<pre>
# heimdall print-pit --verbose --no-reboot
[...]
Initialising protocol...     
Protocol initialisation successful.
                                     
Beginning session...          
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after receiving packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
PIT file download successful.
[...]
</pre>

Note that with the `--no-reboot` argument alone (without `--resume` which does not work)
it is not possible to issue further commands, that would result in errors:

<pre>
Initialising protocol...
ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!
</pre>

LucN31, 2018-04-08:
Hi,
I am same issue to Galaxy S5 and Galaxy S7 on debian 9.
Debian package does not work. 
Heimdall v1.4.2 from github, and locally compiled works only for first command. For example herolte (Galaxy S7) need this command : 
heimdall flash --BOOT twrp-x.x.x-x-herolte.img --RECOVERY twrp-x.x.x-x-herolte.img
BOOT flash pass, RECOVERY don't, it never ends.
**Workarround is flashing BOOT first, rebooting device in download mode, and flash RECOVERY.**

Problem is identical when you pass 2 commands without reboot : first pass, second fails.
It should be more comfortable to avoid to reboot between commands.

**Context :** 
=============================
**LINUX :** 
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch
**libusb :** 
ii  libusb-0.1-4:amd64                        2:0.1.12-30
ii  libusb-1.0-0:amd64                        2:1.0.21-1
ii  libusb-1.0-0-dev:amd64                    2:1.0.21-1
ii  libusb-1.0-doc                            2:1.0.21-1
ii  libusbmuxd4:amd64                         1.0.10-3+b1
ii  libusbredirhost1:amd64                    0.7.1-1
ii  libusbredirparser1:amd64                  0.7.1-1
**heimdall :** 
Heimdall v1.4.2 locally compiled

**I Attach file from 2 print-pit commands, and their equivalents including debug messages**

[first_print-pit.log](https://github.com/Benjamin-Dobell/Heimdall/files/1887422/first_print-pit.log)
[first_print-pit_debug.log](https://github.com/Benjamin-Dobell/Heimdall/files/1887423/first_print-pit_debug.log)
[second_print-pit.log](https://github.com/Benjamin-Dobell/Heimdall/files/1887424/second_print-pit.log)
[second_print-pit_debug.log](https://github.com/Benjamin-Dobell/Heimdall/files/1887425/second_print-pit_debug.log)

Command used are : 
sudo heimdall print-pit --no-reboot --verbose --usb-log-level debug --stdout-errors  | tee first_print-pit.log
and
sudo heimdall print-pit --resume --no-reboot --verbose --usb-log-level debug --stdout-errors  | tee second_print-pit.log

Thx,
Luc.

MartinMiddleton, 2018-06-03:
I duplicated this same issue on two separate Samsung Galaxy Note 4 phones. Compiled 1.4.2 on Ubuntu 16.04 today.

```
$ bin/heimdall download-pit --output /tmp/910t.pit --verbose
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
PIT file download successful.

Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...

Rebooting device...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Releasing device interface...
```

Just trying to print the PIT results in:

```
$ bin/heimdall print-pit --verbose --no-reboot
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...

Some devices may take up to 2 minutes to respond.
Please be patient!

WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
Session begun.

Downloading device's PIT file...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: Failed to unpack received packet.
ERROR: Failed to receive PIT file size!
ERROR: Failed to download PIT file!
Ending session...
WARNING: Empty bulk transfer after sending packet failed. Continuing anyway...
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...

Releasing device interface...
```

The boot sequence following this shows:
```
ODIN: flash read failure
ERROR: There is no pit binary
ODIN: flash read failure
```
I'm willing to test potential code fixes.

starbrights, 2022-07-23:
I'll have same problem **ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...** on my Tab S5e.
I do have TWRP on device, want to flash in Download mode the original LOS recovery. Any idea how that can work out of download mode? Sideload in TWRP seems to work. 
Used heimdall on a Linux machine.

LucN31, 2022-07-23:
Hi,
> I'll have same problem **ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...** on my Tab S5e.

As I can remember, debian package didn't work and I compiled heimdall from github sources.
Only first command pass without resetting usb (not possible to flash after print-pit)

I had to unplug usb, reboot in download mode and re-plug usb cable, before launching another heimdall command.

And after flashing recovery, you have to boot in recovery mode before a normal boot.
Luc.

LucN31, 2022-07-23:
> I had to unplug usb, reboot in download mode and re-plug usb cable, before launching another heimdall command.

Only unplug and replug usb cable works fine with --resume.
Luc.

starbrights, 2022-07-24:
I test that with --resume and with replug cable with no other result.  I can't see any difference between first or following commands. I would assume something's wrong with my device, but adb sideload (when in TWRP) seems to work.
There is happened a lot in LInux whuile plug in the cable, most about PCI bridge, but that is what finally comes about USB:
```

[ 1735.975125] xhci_hcd 0000:37:00.0: Host supports USB 3.1 Enhanced SuperSpeed
[ 1735.976616] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.04
[ 1735.976618] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[ 1735.976620] usb usb3: Product: xHCI Host Controller
[ 1735.976622] usb usb3: Manufacturer: Linux 5.4.0-122-generic xhci-hcd
[ 1735.976623] usb usb3: SerialNumber: 0000:37:00.0
[ 1735.976797] hub 3-0:1.0: USB hub found
[ 1735.976809] hub 3-0:1.0: 2 ports detected
[ 1735.977556] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.04
[ 1735.977559] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[ 1735.977561] usb usb4: Product: xHCI Host Controller
[ 1735.977563] usb usb4: Manufacturer: Linux 5.4.0-122-generic xhci-hcd
[ 1735.977564] usb usb4: SerialNumber: 0000:37:00.0
[ 1735.977706] hub 4-0:1.0: USB hub found
[ 1735.977716] hub 4-0:1.0: 2 ports detected
[ 1736.232685] usb 3-2: new high-speed USB device number 2 using xhci_hcd
[ 1736.384886] usb 3-2: New USB device found, idVendor=04e8, idProduct=685d, bcdDevice= 2.1b
[ 1736.384894] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1736.384898] usb 3-2: Product: SDM710
[ 1736.384902] usb 3-2: Manufacturer: Samsung

```

caliston, 2023-05-29:
I had this problem with a Galaxy Note 4 (SM-N910F) on Ubuntu 22.04 on a USB 3 port (USB-C to MicroUSB cable, no USB 2 ports available).  I could download the PIT but any firmware download failed with this error and the debug log had failures with "urb status -2".

I got it to work with:
- compiling myself from source, not using the Ubuntu package
- not using `--resume`
- booting the phone into download mode, then plugging it in immediately before running the flash command

The full command:
```
Heimdall/build/bin/heimdall flash --APNHLOS NON-HLOS.bin --ABOOT aboot.mbn --BOOT boot.img \
  --CACHE cache.img.ext4 --HIDDEN hidden.img.ext4 --MODEM modem.bin --RECOVERY recovery.img \
  --RPM rpm.mbn --SBL1 sbl1.mbn --DBI sdi.mbn --SYSTEM system.img.ext4 --TZ tz.mbn --usb-log-level debug
```

tbaumann, 2023-06-26:
```
❯ ./bin/heimdall print-pit --no-reboot --usb-log-level debug
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.012280] [00009f57] libusb: debug [libusb_get_device_list]
[ 0.012290] [00009f57] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.012294] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012296] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012297] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012298] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012299] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012300] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012301] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012302] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012303] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012304] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012305] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012309] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012313] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012314] [00009f57] libusb: debug [libusb_open] open 7.25
[ 0.012336] [00009f57] libusb: debug [usbi_add_event_source] add fd 7 events 4
[ 0.012339] [00009f57] libusb: debug [libusb_get_device_descriptor]
[ 0.012341] [00009f57] libusb: debug [libusb_get_config_descriptor] index 0
Claiming interface...
[ 0.012350] [00009f57] libusb: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 0.012363] [00009f57] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0

Initialising protocol...
[ 0.013176] [00009f57] libusb: debug [libusb_submit_transfer] transfer 0x55e887aa2680
[ 0.013178] [00009f57] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.013182] [00009f57] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 0.013190] [00009f57] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.013192] [00009f57] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.013194] [00009f57] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 1.013223] [00009f57] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 1.013249] [00009f57] libusb: debug [libusb_cancel_transfer] transfer 0x55e887aa2680
[ 1.013422] [00009f57] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 1.013443] [00009f57] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.013447] [00009f57] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 1.013452] [00009f57] libusb: debug [usbi_wait_for_events] poll() returned 1
[ 1.013458] [00009f57] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[ 1.013461] [00009f57] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[ 1.013464] [00009f57] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
[ 1.013466] [00009f57] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[ 1.013469] [00009f57] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[ 1.013472] [00009f57] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 1.013475] [00009f57] libusb: debug [usbi_handle_transfer_completion] transfer 0x55e887aa2680 has callback 0x7fc4d5fea0c0
[ 1.013477] [00009f57] libusb: debug [sync_transfer_cb] actual_length=0
[ 1.013481] [00009f57] libusb: debug [libusb_free_transfer] transfer 0x55e887aa2680
[ 1.263557] [00009f57] libusb: debug [libusb_submit_transfer] transfer 0x55e887ab6040
[ 1.263580] [00009f57] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 1.263587] [00009f57] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 1.263610] [00009f57] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.263615] [00009f57] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
[ 2.263592] [00009f57] libusb: debug [usbi_wait_for_events] poll() returned 1
[... 136 lines cut ...]
[10.765416] [00009f57] libusb: debug [libusb_unref_device] destroy device 7.8
[10.765420] [00009f57] libusb: debug [libusb_unref_device] destroy device 7.7
[10.765423] [00009f57] libusb: debug [libusb_unref_device] destroy device 7.5
[10.765428] [00009f57] libusb: debug [libusb_unref_device] destroy device 7.3
[10.765433] [00009f57] libusb: debug [libusb_unref_device] destroy device 7.9
[10.765438] [00009f57] libusb: debug [libusb_unref_device] destroy device 7.1
[10.765443] [00009f57] libusb: debug [libusb_unref_device] destroy device 6.1
[10.765449] [00009f57] libusb: debug [libusb_unref_device] destroy device 5.1
[10.765453] [00009f57] libusb: debug [libusb_unref_device] destroy device 4.1
[... cut ...]
[10.765472] [00009f57] libusb: debug [libusb_unref_device] destroy device 1.2
[10.765477] [00009f57] libusb: debug [libusb_unref_device] destroy device 1.1
[10.765483] [00009f57] libusb: debug [usbi_remove_event_source] remove fd 4
[10.765492] [00009f57] libusb: debug [usbi_remove_event_source] remove fd 3
```
Compiled from source on Ubuntu 22.04

```
❯ ./bin/heimdall version
v1.4.2
```

I get the same error with the distro package. And also if I use the five year old build from https://hub.docker.com/r/groupsky/heimdall

tbaumann, 2023-06-26:
Works with https://github.com/Benjamin-Dobell/Heimdall/pull/478 via https://github.com/Benjamin-Dobell/Heimdall/issues/504

Please merge upstream

lukaprincic, 2025-02-03:
Hey all, had a big problem flashing a Samsung Galaxy A52s 5G with the exact same error:
```
ERROR: libusb error -7 whilst receiving bulk transfer. Retrying...
```
but solved it by compiling a 2.1.0 version from https://git.sr.ht/~grimler/Heimdall