Benjamin-Dobell/Heimdall #416: ERROR: Failed to confirm end of file transfer sequence!

padraig2, 2017-09-02:
I get the error :  "ERROR: Failed to confirm end of file transfer sequence! ERROR: RECOVERY upload failed!"

Thank you in advance for your help.

**Heimdall version** - 1.4.2
**USB version**        - 1.0.21
**USB Cord**            - Brand new Anker PowerLine USB (n.b. I have tried multiple USB ports on back of Mac)
**Phone**                 - Samsung Galaxy Core Prime (model SM-G360T1)
**Computer**          - IMac 21.5 Late 2015  macOS 10.12.6

**Command**

./heimdall flash --RECOVERY Unofficial_TWRP_G360T1.tar.md5 --no-reboot --usb-log-level debug 2> error_log2

_(n.b. I get the same errors if I expand the above file into an .img file)_

**Error Log**

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000547] [00000307] libusbx: debug [libusb_get_device_list] 
[ 0.002051] [00000307] libusbx: debug [process_new_device] allocating new device for location 0x14000000
[ 0.002083] [00000307] libusbx: debug [darwin_cache_device_descriptor] device descriptor:
[ 0.002090] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDescriptorType:    0x01
[ 0.002095] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bcdUSB:             0x0100
[ 0.002100] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDeviceClass:       0x09
[ 0.002105] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDeviceSubClass:    0xff
[ 0.002110] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDeviceProtocol:    0x03
[ 0.002115] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bMaxPacketSize0:    0x08
[ 0.002120] [00000307] libusbx: debug [darwin_cache_device_descriptor]  idVendor:           0x05ac
[ 0.002125] [00000307] libusbx: debug [darwin_cache_device_descriptor]  idProduct:          0x8007
[ 0.002130] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bcdDevice:          0x0198
[ 0.002134] [00000307] libusbx: debug [darwin_cache_device_descriptor]  iManufacturer:      0x00
[ 0.002139] [00000307] libusbx: debug [darwin_cache_device_descriptor]  iProduct:           0x00
[ 0.002144] [00000307] libusbx: debug [darwin_cache_device_descriptor]  iSerialNumber:      0x00
[ 0.002149] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bNumConfigurations: 0x01
[ 0.002164] [00000307] libusbx: debug [darwin_check_configuration] active config: 0, first config: 1
[ 0.002179] [00000307] libusbx: debug [process_new_device] found device with address 0 port = 0 parent = 0x7fb93bc09548 at 0x0
[ 0.002560] [00000307] libusbx: debug [process_new_device] allocating new device for location 0x14300000
[ 0.002623] [00000307] libusbx: debug [darwin_cache_device_descriptor] device descriptor:
[ 0.002638] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDescriptorType:    0x01
[ 0.002643] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bcdUSB:             0x0200
[ 0.002648] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDeviceClass:       0x09
[ 0.002654] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDeviceSubClass:    0x00
[ 0.002659] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDeviceProtocol:    0x01
[ 0.002664] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bMaxPacketSize0:    0x40
[ 0.002669] [00000307] libusbx: debug [darwin_cache_device_descriptor]  idVendor:           0x1a40
[ 0.002675] [00000307] libusbx: debug [darwin_cache_device_descriptor]  idProduct:          0x0101
[ 0.002680] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bcdDevice:          0x0111
[ 0.002688] [00000307] libusbx: debug [darwin_cache_device_descriptor]  iManufacturer:      0x00
[ 0.002693] [00000307] libusbx: debug [darwin_cache_device_descriptor]  iProduct:           0x01
[ 0.002699] [00000307] libusbx: debug [darwin_cache_device_descriptor]  iSerialNumber:      0x00
[ 0.002704] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bNumConfigurations: 0x01
[ 0.002738] [00000307] libusbx: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002748] [00000307] libusbx: debug [process_new_device] found device with address 4 port = 3 parent = 0x7fb93bc08008 at 0x7fb93bc094b0
[ 0.003041] [00000307] libusbx: debug [process_new_device] allocating new device for location 0x14340000
[ 0.003095] [00000307] libusbx: debug [darwin_cache_device_descriptor] device descriptor:
[ 0.003103] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDescriptorType:    0x01
[ 0.003108] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bcdUSB:             0x0210
[ 0.003113] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDeviceClass:       0x00
[ 0.003119] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDeviceSubClass:    0x00
[ 0.003135] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bDeviceProtocol:    0x00
[ 0.003143] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bMaxPacketSize0:    0x40
[ 0.003149] [00000307] libusbx: debug [darwin_cache_device_descriptor]  idVendor:           0x1058
[ 0.003154] [00000307] libusbx: debug [darwin_cache_device_descriptor]  idProduct:          0x0830
[ 0.003159] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bcdDevice:          0x1065
[ 0.003164] [00000307] libusbx: debug [darwin_cache_device_descriptor]  iManufacturer:      0x02
[ 0.003179] [00000307] libusbx: debug [darwin_cache_device_descriptor]  iProduct:           0x03
[ 0.003184] [00000307] libusbx: debug [darwin_cache_device_descriptor]  iSerialNumber:      0x01
[ 0.003188] [00000307] libusbx: debug [darwin_cache_device_descriptor]  bNumConfigurations: 0x01
[... 1548 lines cut ...]
[ 6.296120] [00000307] libusbx: debug [handle_events] poll() returned 1
[ 6.296130] [00000307] libusbx: debug [op_handle_events] checking fd 4 with revents = 0
[ 6.296136] [00000307] libusbx: debug [op_handle_events] checking fd 6 with revents = 1
[ 6.296145] [00000307] libusbx: debug [darwin_handle_callback] handling bulk completion with kernel status -536854447
[ 6.296151] [00000307] libusbx: warning [darwin_transfer_status] transfer error: timed out
[ 6.296157] [00000307] libusbx: debug [usbi_handle_transfer_completion] transfer 0x7fb93bf009a0 has callback 0x10c60a452
[ 6.296163] [00000307] libusbx: debug [bulk_transfer_cb] actual_length=0
[ 6.296177] [00000307] libusbx: debug [ep_to_pipeRef] converting ep address 0x81 to pipeRef and interface
[ 6.296184] [00000307] libusbx: debug [ep_to_pipeRef] pipe 1 on interface 1 matches
[ 6.296238] [00000307] libusbx: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!
[ 6.296246] [00000307] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 6.296252] [00000307] libusbx: debug [handle_events] poll() 2 fds with timeout in 60000ms
[ 6.317138] [00000b07] libusbx: debug [darwin_async_io_callback] an async io operation has completed
[ 6.317215] [00000307] libusbx: debug [handle_events] poll() returned 1
[ 6.317244] [00000307] libusbx: debug [op_handle_events] checking fd 4 with revents = 0
[ 6.317251] [00000307] libusbx: debug [op_handle_events] checking fd 6 with revents = 1
[ 6.317259] [00000307] libusbx: debug [darwin_handle_callback] handling bulk completion with kernel status 0
[ 6.317267] [00000307] libusbx: debug [usbi_handle_transfer_completion] transfer 0x7fb93be00ee0 has callback 0x10c60a452
[ 6.317274] [00000307] libusbx: debug [bulk_transfer_cb] actual_length=8
[ 6.317299] [00000307] libusbx: debug [libusb_release_interface] interface 1
[ 6.317668] [00000307] libusbx: debug [libusb_close] 
[ 6.317755] [00000307] libusbx: debug [usbi_remove_pollfd] remove fd 6
[ 6.317790] [00000307] libusbx: debug [libusb_unref_device] destroy device 20.21
[ 6.317799] [00000307] libusbx: debug [libusb_exit] 
[ 6.317804] [00000307] libusbx: debug [libusb_exit] destroying default context

JonnyTech, 2017-09-02:
The USB port is not recognised.

The extracted img file should be used.

Can you run as su and provide a standard log without the debug info?

padraig2, 2017-09-02:
**Command** 

sudo ./heimdall flash --RECOVERY recovery.img --no-reboot

_n.b. **recovery.img** is the expanded version of Unofficial_TWRP_G360T1.tar.md5_

**Output**

Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
PIT file download successful.

Uploading RECOVERY
100%
ERROR: Failed to confirm end of file transfer sequence!
ERROR: RECOVERY upload failed!

Ending session...
Releasing device interface...

JonnyTech, 2017-09-02:
Looks like it worked. What was displayed on the phone? The error message may be because of the --no-reboot flag. Did you boot directly to TWRP afterwards? (Note that booting to the Android system will overwrite the new recovery and you must flash again. Booting to TWRP directly after flashing makes it permanent)

padraig2, 2017-09-03:
The phone shows the bootloader screen. 

This is what I did

1. Unplug the USB cable
2. Take out the battery _(the phone powers off)_
3. Boot into recovery mode _up volume, home button, power_

I see the stock recovery 

It says : 
**Android system recovery <3e>**
**LMY47X.G360T1UVU2AQD1**
_followed by other options such as reboot, apply update from ADB, etc_

It's the same recovery mode as before.

However, at the bottom, in amber letters, it says :

**# MANUAL MODE #**
**-- Applying Multi-CSC**
**Applied the CSC-code : TMB**
**Successfully applied multi-CSC**

I'm not sure if that was there before or not (before trying to flash anything)

Here is another clue, perhaps - something odd in the debug output : 

**libusbx: warning [darwin_transfer_status] transfer error: timed out**

A few lines below that I see 
**libusbx: debug [libusb_get_next_timeout] no URB with timeout or all handled by OS; no timeout!**

Another few lines it says : 
**ERROR: Failed to confirm end of file transfer sequence!
ERROR: RECOVERY upload failed!**

Anyway, these are messages in the debug output I included above. I don't think it worked since heimdall is telling me that "Recovery upload failed" and I also see the same recovery as before.

I get the same errors no matter if I try --no-reboot or not.

JonnyTech, 2017-09-03:
Hmm, are you sure that the bootloader is unlocked? If so, a long shot, but try hack from #347.

padraig2, 2017-09-04:
I don't know for a fact that the bootloader is unlocked. However I see many tutorials on how to root my particular device (MetroPCS variant of SM-G360T1). They all use ODIN as an upload tool.

I'm not sure how to find out for certain whether the phone is locked or not. I will have to research that some more. I can also just try to use ODIN and see if that works. 

Thanks anyway for your help. If the ODIN option doesn't work I will try that hack.

sarevok-anchev, 2018-05-03:
Right, just got to the bottom of this.

May it be as simple for you guys..

I noticed that the twrp .img file (https://eu.dl.twrp.me/gts210vewifi/twrp-3.2.1-0-gts210vewifi.img) was very small all of a sudden; one less later, and I realize it's just HTML!

The file was downloaded with wget, seems you have to actually click it in-browser.. and then a ~14.3MB file is received.

That I could flash, and no errors.

D'oh!