Benjamin-Dobell/Heimdall #417: Heimdall fails to fully flash a large image file

garretyoder, 2017-09-07:
While heimdall technically does not support newer devices such as the GS6,7, etc, it does actually work, more or less. However, the only failure I've encountered is when you ask it to flash a image that is greater than (I'm assuming from previous issues) 2GB. A 1.8gb system.img.ext4 flashes fine, however a 4GB system.img.ext4 fails to flash, silently. It uploads inside of a few seconds, obviously too fast to be transferring 4GB, and the phone fails it for secure check fail. Obviously it didn't transfer the entire image. I've doublechecked that large file support is enabled, and I'm decently sure it is as I compiled the most recent sources myself.

Any ideas? My little project will work flawlessly if I can just convince it to flash these huge 4GB images from newer phones.

Benjamin-Dobell, 2017-09-08:
Which version of Heimdall are you using?

Precisely how big are the files?

If you're building from source then LFS (large file support) should be enabled. However, the protocol itself is 32-bit, so you can't flash anything larger than 4,294,967,296 bytes.

If Samsung are doing that now, then the protocol must have been updated somehow.

garretyoder, 2017-09-08:
I'm using the latest source from master. I can check the exact byte count when I'm near the computer with the image tomorrow, but if I recall it was 4.1GB, so it should be within that limit. Heimdall doesn't produce any errors, using usb logging doesn't produce any warnings that aren't present all the time, even when it successfully flashes a smaller image. I know it's not the system images fault because I tried with multiple phones and the issue is reproducible. A smaller (1.8GB) system image from a combination file flashes perfectly fine.

It doesn't "fail" from heimdall's side, it uploads a 4GB image in a few seconds, which obviously isn't all 4GB. The phone fails it for security check because it obviously wasn't the full image, and then heimdall looses it's ability to communicate with the phone until it's been rebooted. Being as that no output changes between a working/not working flash, and the only thing different is the size of the file, I'm at a loss. Is there anything I can test to help either of us troubleshoot the issue?

garretyoder, 2017-09-08:
In addition - it almost seems like heimdall is misreading the size of the image and only sending the first few mb, but I'm not sure how to test that theory.

garretyoder, 2017-09-09:
I seem to have narrowed the issue. It's not related to the file size, but rather whether the image is system.img.ext4 versus system.img I believe. A 5GB system.img successfully flashed, but no system.img.ext4 images would flash.