Benjamin-Dobell/Heimdall #418: Problem flashing system image to i9100

tssk, 2017-09-07:
Hello,

I run into issue with flashing system image (to the FACTORYFS partition) generated in postmarketOS project -> https://github.com/postmarketOS/pmbootstrap/issues/422

Can I please ask you if you could help me identify where the problem could be?

I will be happy to provide any information or test anything necessary.

Thank you

Benjamin-Dobell, 2017-09-08:
Please provide some information - Heimdall version, command line output, your OS etx.

tssk, 2017-09-08:
@Benjamin-Dobell Hello,

I am running up-to-date amd64 Debian GNU/Linux
```

# heimdall flash --FACTORYFS samsung-i9100.img --verbose
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
           Product: "Gadget Serial"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 83
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
PIT file download successful.

Uploading FACTORYFS
0%
1%

2%

3%

4%

5%

6%

7%

8%

9%
ERROR: Failed to unpack received packet.

ERROR: Failed to confirm end of file transfer sequence!
ERROR: FACTORYFS upload failed!

Ending session...
Rebooting device...
Releasing device interface...

```

tssk, 2017-09-08:
I would like to ask if there is a possibility that there is some problem with the .img file itself and the device rejects it. As suggested in the comment https://github.com/Benjamin-Dobell/Heimdall/issues/191#issuecomment-317183924

tssk, 2017-09-18:
Closing this as there exists working description how to install on i9100 in wiki - https://wiki.postmarketos.org/wiki/Samsung_Galaxy_SII_(samsung-i9100)#Step_by_step_guide_for_Flashing_to_device_memory

tssk, 2017-09-19:
Sorry, I accidentally closed this instead of https://github.com/postmarketOS/pmbootstrap/issues/422

montvid, 2018-01-22:
So what other information might be needed to fix this bug?

Benjamin-Dobell, 2018-01-22:
@montvid A working flash capture from Kies/Odin is unfortunately pretty much the only thing that can help. However, I've flashed an I9100T _many_ times and (I could be wrong but) I believe they're essentially the same phone except for the modem.

Are you sure your device has an unlocked bootloader?

montvid, 2018-01-22:
My device is i9100 and it is unlocked. I uploaded the img file of postmarketos I am trying to flash with heimdal 1.4.2 unsuccessfully. It seems that I am not the only one that can't flash a pmos img file see, maybe you can try and see maybe the file format is bad or something:  
https://we.tl/LnFtupNAhs
Unfortunately I don't have a Windows machine to run Kies/Odin - does it work on wine? I should somehow get a log from odin to see what is happening?

montvid, 2018-02-03:
I mean i can flash the kernel with heimdal but i can't flash the data img with heimdal.