Benjamin-Dobell/Heimdall #420: Creating Odin3 protocol spec

KOLANICH, 2017-09-18:
https://github.com/KOLANICH/kaitai_struct_formats/tree/odin3

Benjamin-Dobell, 2017-09-18:
Okay. But why have you opened an _issue_?

KOLANICH, 2017-09-18:
1 I expect you to check it and add the missing information. I have created it from heimdall source code, but I guess you may have more info agout the protocol than you have the code.
2 I guess that the code generated (that specs are intended for autogenerating parsers using kaitai struct compiler) from that spec can be useful in Heimdall. It can be suitable for parsing, no serialization for now. I guess it is not ready for production since not all c++-related issues are fixed in compiler and runtimes, but it may be time to start integrating the generated code into Heimdall.

Benjamin-Dobell, 2017-09-18:
You're certainly welcome to do those things yourself, however I don't have the time or interest to do so. Heimdall is of course open source, so you're welcome to base your work on Heimdall.

However, in general Github's issue tracker is not the appropriate place to be requesting other people to contribute to your projects. As no issue with Heimdall has been identified I'm closing this ticket.

closed 2017-09-18 by Benjamin-Dobell