Benjamin-Dobell/Heimdall #421: unable to compile from source

Huby1, 2017-09-19:
Hi, 
I am working on Ubuntu 14.04 LTS. I found out that from "apt-get" the repository is not up to date. It is the version 1.4.0 that is installing. So I removed all to install from source the latest stable version witch is, except my mistake, 1.4.1.
Whilst installing the source I fail at the last line: $ sudo cp bin/* /usr/local/bin
Here is the error I get: 
cp: impossible to evaluate «bin/*»: no files or folder of this type

Many Thanks for your help ;)
Cheers,

JonnyTech, 2017-09-19:
Can you paste the output from the terminal after the `make` command. Are there any errors?

If you type `ls` you should see a folder named bin. You can try:
```
cd bin
cp * /usr/local/bin/
```

Huby1, 2017-09-19:
Hi JonnyTech, 
Many Thanks for your return. I searching and searching but probably missing something.. There is no "bin" folder into the downloaded source installation folder.
Any Idea ??? 

Many Thanks

JonnyTech, 2017-09-19:
As requested, please post the outputs of the other commands here - especially the `make`.

Huby1, 2017-09-19:
OK: 
`$ make`
`Scanning dependencies of target pit
[100%] Building CXX object CMakeFiles/pit.dir/source/libpit.cpp.o
Linking CXX static library libpit.a
[100%] Built target pit
$ sudo cp bin/* /usr/local/bin
cp: evaluation impossible «bin/*»: No file or folder of this type`

JonnyTech, 2017-09-19:
Are you sure that you typed all of the lines correctly?

```
$ git clone git://github.com/Benjamin-Dobell/Heimdall.git
$ mkdir -p Heimdall/build
$ cd Heimdall/build
$ cmake -DCMAKE_BUILD_TYPE=Release ..
$ make
$ sudo cp bin/* /usr/local/bin
```

Note that the `cmake` line ends in two dots.

Please provide output from all of the commands.

Huby1, 2017-09-19:
Hi JohnnyTech, 

I have decided to remove and start from scratch compiling. So I have something really different now:
I stuck at the `cmake -DCMAKE_BUILD_TYPE=Release ..` line.
Here is the error I get:
`CMake Error at CMakeLists.txt:9 (find_package):`
  `By not providing "Findlibusb.cmake" in CMAKE_MODULE_PATH this project has`
  `asked CMake to find a package configuration file provided by "libusb", but`
  `CMake did not find one.`

  `Could not find a package configuration file provided by "libusb" with any`
  `of the following names:`

   ` libusbConfig.cmake`
 `   libusb-config.cmake`

  `Add the installation prefix of "libusb" to CMAKE_PREFIX_PATH or set`
 ` "libusb_DIR" to a directory containing one of the above files.  If "libusb" `
`  provides a separate development package or SDK, be sure it has been`
`  installed.`

`-- Configuring incomplete, errors occurred!`
`See also "/home/sungardas/HeimdallFromSources/Heimdall/build/CMakeFiles/CMakeOutput.log".
`
**Here is the log file content:**

The system is: Linux - 4.4.0-96-generic - i686
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags: 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/sungardas/HeimdallFromSources/Heimdall/build/CMakeFiles/2.8.12.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags: 

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/sungardas/HeimdallFromSources/Heimdall/build/CMakeFiles/2.8.12.2/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/sungardas/HeimdallFromSources/Heimdall/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec1935004529/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec1935004529.dir/build.make CMakeFiles/cmTryCompileExec1935004529.dir/build
make[1]: entrant dans le répertoire « /home/sungardas/HeimdallFromSources/Heimdall/build/CMakeFiles/CMakeTmp »
/usr/bin/cmake -E cmake_progress_report /home/sungardas/HeimdallFromSources/Heimdall/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec1935004529.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTryCompileExec1935004529.dir/testCCompiler.c.o   -c /home/sungardas/HeimdallFromSources/Heimdall/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTryCompileExec1935004529
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec1935004529.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTryCompileExec1935004529.dir/testCCompiler.c.o  -o cmTryCompileExec1935004529 -rdynamic 
make[1]: quittant le répertoire « /home/sungardas/HeimdallFromSources/Heimdall/build/CMakeFiles/CMakeTmp »

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/sungardas/HeimdallFromSources/Heimdall/build/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/make "cmTryCompileExec1905720692/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec1905720692.dir/build.make CMakeFiles/cmTryCompileExec1905720692.dir/build
[... 185 lines cut ...]
    arg [CMakeFiles/cmTryCompileExec222260960.dir/CMakeCXXCompilerABI.cpp.o] ==> ignore
    arg [-lstdc++] ==> lib [stdc++]
    arg [-lm] ==> lib [m]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [-lc] ==> lib [c]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [/usr/lib/gcc/i686-linux-gnu/4.8/crtend.o] ==> ignore
    arg [/usr/lib/gcc/i686-linux-gnu/4.8/../../../i386-linux-gnu/crtn.o] ==> ignore
  remove lib [gcc_s]
  remove lib [gcc]
  remove lib [gcc_s]
  remove lib [gcc]
  collapse library dir [/usr/lib/gcc/i686-linux-gnu/4.8] ==> [/usr/lib/gcc/i686-linux-gnu/4.8]
  collapse library dir [/usr/lib/gcc/i686-linux-gnu/4.8/../../../i386-linux-gnu] ==> [/usr/lib/i386-linux-gnu]
  collapse library dir [/usr/lib/gcc/i686-linux-gnu/4.8/../../../../lib] ==> [/usr/lib]
  collapse library dir [/lib/i386-linux-gnu] ==> [/lib/i386-linux-gnu]
  collapse library dir [/lib/../lib] ==> [/lib]
  collapse library dir [/usr/lib/i386-linux-gnu] ==> [/usr/lib/i386-linux-gnu]
  collapse library dir [/usr/lib/../lib] ==> [/usr/lib]
  collapse library dir [/usr/lib/gcc/i686-linux-gnu/4.8/../../..] ==> [/usr/lib]
  implicit libs: [stdc++;m;c]
  implicit dirs: [/usr/lib/gcc/i686-linux-gnu/4.8;/usr/lib/i386-linux-gnu;/usr/lib;/lib/i386-linux-gnu;/lib]
  implicit fwks: []

JonnyTech, 2017-09-19:
You do not have the development package of `libusb` installed.

As per the wiki (https://github.com/Benjamin-Dobell/Heimdall/wiki/Build-instructions-(Linux)) you need to install the following dependencies:

```
$ sudo apt-get install build-essential cmake zlib1g-dev qt5-default libusb-1.0-0-dev libgl1-mesa-glx libgl1-mesa-dev
```

You should also find `libusb-dev` in your package manager, or you can install that from source too from http://libusb.info/.

Benjamin-Dobell, 2017-09-19:
If you just want the command line tool then compilation (and download times) can be sped up. _Exact_ commands for Ubuntu 14.04 are:

```
sudo apt-get update
sudo apt-get install -y build-essential cmake git libusb-1.0-0-dev

git clone git://github.com/Benjamin-Dobell/Heimdall.git

mkdir ./Heimdall/build
cd /Heimdall/build

cmake -DCMAKE_BUILD_TYPE=Release -DDISABLE_FRONTEND=ON ..
make
sudo cp bin/heimdall /usr/local/bin
```

I've just verified this with a fresh virtual machine.

JonnyTech, 2017-09-19:
@Benjamin-Dobell would you mind adding the `-DDISABLE_FRONTEND=ON` flag to the wiki?

Benjamin-Dobell, 2017-09-19:
@JonnyTech You're welcome to add it and reorganise as desired.

To be honest, I didn't even know that stuff was there in the wiki! I tend to prefer referring people to the official README; which admittedly is also missing the information. Pull requests are welcome!

JonnyTech, 2017-09-19:
Ah, I just noticed the big "EDIT" button...

Huby1, 2017-09-19:
Well, Many Many Thanks @Benjamin-Dobell and @JonnyTech , That time it works as a charm.
regarding **heimdall-frontend**, should I do something else to have it work ?

Benjamin-Dobell, 2017-09-19:
@Huby1 I _think_ you just need the following additional dependencies for Ubuntu 14.04:

```
sudo apt-get install qtbase5-dev zlib1g-dev
```

Then (assuming you're in the cloned `Heimdall` root directory):

```
cd ./build
cmake -DCMAKE_BUILD_TYPE=Release ..
make
sudo cp bin/* /usr/local/bin
```

Note that this time we omitted `-DDISABLE_FRONTEND=ON`.

Benjamin-Dobell, 2017-09-19:
If that doesn't work then try:

```
apt-get install qt5-default libgl1-mesa-glx libgl1-mesa-dev
```

Honestly my memory isn't great regarding the precise required Qt5 dependencies for Ubuntu 14.04. Heimdall certainly doesn't utilise OpenGL directly, it just depends on how Ubuntu's qt5 packages have been setup.

Huby1, 2017-09-19:
Ok so from the command line I was able to finally download the *.pit file.
Now I did all the latest steps to have frontend and at the command:
`$ heimdall-frontend`
- I get the following : 
`The software « heimdall-frontend » is not yet installed. You can install it by typing bellow :`
`sudo apt-get install heimdall-flash-frontend`

Benjamin-Dobell, 2017-09-20:
@Huby1 Instructions for compiling `heimdall-frontend` were given above.

Please take the time to thoroughly read both what @JonnyTech and I have written above, as well as the documentation (READMEs). Flashing your phone isn't something to rush into; I can understand having a bit of difficulty compiling Heimdall, but realistically all the information you've asked for is already available. You need to read, understand and follow instructions, or you could end up damaging your device.

Seems as you were able to compile Heimdall (i.e. No _issue_ with Heimdall), I'm going to close this issue. Github issues is for bug reports / feature requests etc. and not really for technical support. For technical support I'd recommend the XDA Developer forums, where lots of people will see and be able to help.

closed 2017-09-20 by Benjamin-Dobell

Huby1, 2017-09-20:
Indeed I am not rushing, really.. yesterday I searched for hours on the net after I went over the READMEs, and the WIKI available on GIT.. and still did not found how to compile frontend except using the apt-get.. but it is not the latest version.

Many Thanks anyway @Benjamin-Dobell and @JonnyTech for all your help