Benjamin-Dobell/Heimdall #422: How to uninstall heimdall installed from source?

siulkilulki, 2017-09-19:
How to uninstall heimdall installed from source? The makefiles are pretty big and I don't want to analyze them and track every file I need to delete. Can someone tell me what should I delete to remove heimdall and all of its installed components. I'm on Linux.

Hypexed, 2017-09-27:
How did you install it? The Makefile I generated didn't have an install rule. So I ran it from the build dir,

siulkilulki, 2017-09-27:
I have also ran i from the build dir, just like in README:
```
 mkdir build
 cd build
 cmake -DCMAKE_BUILD_TYPE=Release ..
 make
```

Hypexed, 2017-09-27:
By the looks of it you just need to delete the master folder extracted and it should be gone. I see a small  main heimdall binary created in build/bin. Could also look in FileSystem on desktop for heimdall.