Benjamin-Dobell/Heimdall #423: unspecified reference to the destination compilation errors

RedKugelblitz, 2017-09-26:
GNU/Linux 64-bit Debian gcc version 6.3.0

when trying to link the binary, gcc aborts:
[ 10%] Linking CXX executable ../bin/heimdall
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::FindDeviceInterface()”:
BridgeManager.cpp:(.text+0x2c): määrittelemätön viite kohteeseen ”libusb_get_device_list”
BridgeManager.cpp:(.text+0x95): määrittelemätön viite kohteeseen ”libusb_get_device_descriptor”
BridgeManager.cpp:(.text+0xbf): määrittelemätön viite kohteeseen ”libusb_ref_device”
BridgeManager.cpp:(.text+0xd3): määrittelemätön viite kohteeseen ”libusb_free_device_list”
BridgeManager.cpp:(.text+0xe9): määrittelemätön viite kohteeseen ”libusb_open”
BridgeManager.cpp:(.text+0xff): määrittelemätön viite kohteeseen ”libusb_get_device_descriptor”
BridgeManager.cpp:(.text+0x124): määrittelemätön viite kohteeseen ”libusb_get_config_descriptor”
BridgeManager.cpp:(.text+0x2c2): määrittelemätön viite kohteeseen ”libusb_free_config_descriptor”
BridgeManager.cpp:(.text+0x45c): määrittelemätön viite kohteeseen ”libusb_get_string_descriptor_ascii”
BridgeManager.cpp:(.text+0x487): määrittelemätön viite kohteeseen ”libusb_get_string_descriptor_ascii”
BridgeManager.cpp:(.text+0x4b2): määrittelemätön viite kohteeseen ”libusb_get_string_descriptor_ascii”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::ClaimDeviceInterface()”:
BridgeManager.cpp:(.text+0x59a): määrittelemätön viite kohteeseen ”libusb_claim_interface”
BridgeManager.cpp:(.text+0x5ca): määrittelemätön viite kohteeseen ”libusb_detach_kernel_driver”
BridgeManager.cpp:(.text+0x5e4): määrittelemätön viite kohteeseen ”libusb_claim_interface”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::SetupDeviceInterface()”:
BridgeManager.cpp:(.text+0x61d): määrittelemätön viite kohteeseen ”libusb_set_interface_alt_setting”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::ReleaseDeviceInterface()”:
BridgeManager.cpp:(.text+0x67a): määrittelemätön viite kohteeseen ”libusb_release_interface”
BridgeManager.cpp:(.text+0x69a): määrittelemätön viite kohteeseen ”libusb_attach_kernel_driver”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::~BridgeManager()”:
BridgeManager.cpp:(.text+0x724): määrittelemätön viite kohteeseen ”libusb_close”
BridgeManager.cpp:(.text+0x732): määrittelemätön viite kohteeseen ”libusb_unref_device”
BridgeManager.cpp:(.text+0x75e): määrittelemätön viite kohteeseen ”libusb_release_interface”
BridgeManager.cpp:(.text+0x79e): määrittelemätön viite kohteeseen ”libusb_attach_kernel_driver”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::DetectDevice()”:
BridgeManager.cpp:(.text+0x7c4): määrittelemätön viite kohteeseen ”libusb_init”
BridgeManager.cpp:(.text+0x7f6): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0x80a): määrittelemätön viite kohteeseen ”libusb_get_device_list”
BridgeManager.cpp:(.text+0x862): määrittelemätön viite kohteeseen ”libusb_get_device_descriptor”
BridgeManager.cpp:(.text+0x886): määrittelemätön viite kohteeseen ”libusb_free_device_list”
BridgeManager.cpp:(.text+0x8e8): määrittelemätön viite kohteeseen ”libusb_free_device_list”
BridgeManager.cpp:(.text+0x926): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0x933): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0x946): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0x956): määrittelemätön viite kohteeseen ”libusb_set_debug”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::SendBulkTransfer(unsigned char*, int, int, bool) const”:
BridgeManager.cpp:(.text+0x995): määrittelemätön viite kohteeseen ”libusb_bulk_transfer”
BridgeManager.cpp:(.text+0x9e4): määrittelemätön viite kohteeseen ”libusb_bulk_transfer”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::InitialiseProtocol()”:
BridgeManager.cpp:(.text+0xb44): määrittelemätön viite kohteeseen ”libusb_bulk_transfer”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::Initialise(bool)”:
BridgeManager.cpp:(.text+0xc2d): määrittelemätön viite kohteeseen ”libusb_init”
BridgeManager.cpp:(.text+0xc66): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0xc96): määrittelemätön viite kohteeseen ”libusb_claim_interface”
BridgeManager.cpp:(.text+0xcbf): määrittelemätön viite kohteeseen ”libusb_set_interface_alt_setting”
BridgeManager.cpp:(.text+0xd23): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0xd46): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0xd6e): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0xd96): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0xdd2): määrittelemätön viite kohteeseen ”libusb_detach_kernel_driver”
BridgeManager.cpp:(.text+0xdec): määrittelemätön viite kohteeseen ”libusb_claim_interface”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::ReceiveBulkTransfer(unsigned char*, int, int, bool) const”:
BridgeManager.cpp:(.text+0xe79): määrittelemätön viite kohteeseen ”libusb_bulk_transfer”
BridgeManager.cpp:(.text+0xec5): määrittelemätön viite kohteeseen ”libusb_bulk_transfer”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::EndSession(bool) const”:
BridgeManager.cpp:(.text+0x1255): määrittelemätön viite kohteeseen ”libusb_bulk_transfer”
BridgeManager.cpp:(.text+0x1371): määrittelemätön viite kohteeseen ”libusb_bulk_transfer”
BridgeManager.cpp:(.text+0x148e): määrittelemätön viite kohteeseen ”libusb_bulk_transfer”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o:BridgeManager.cpp:(.text+0x14f3): lisää määrittelemättömiä viitteitä kohteeseen ”libusb_bulk_transfer” seuraa
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::~BridgeManager()”:
BridgeManager.cpp:(.text+0x741): määrittelemätön viite kohteeseen ”libusb_exit”
CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o: Funktiossa ”Heimdall::BridgeManager::SetUsbLogLevel(Heimdall::BridgeManager::UsbLogLevel)”:
BridgeManager.cpp:(.text+0x5d86): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0x5d96): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0x5da3): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0x5db6): määrittelemätön viite kohteeseen ”libusb_set_debug”
BridgeManager.cpp:(.text+0x5dc6): määrittelemätön viite kohteeseen ”libusb_set_debug”
collect2: error: ld returned 1 exit status
heimdall/CMakeFiles/heimdall.dir/build.make:408: recipe for target 'bin/heimdall' failed
make[2]: *** [bin/heimdall] Error 1
CMakeFiles/Makefile2:140: recipe for target 'heimdall/CMakeFiles/heimdall.dir/all' failed
make[1]: *** [heimdall/CMakeFiles/heimdall.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

sorry for the finnish output, "määrittelemätön viite kohteeseen" -> "unspecified reference to the destination"

FinKsu, 2017-09-28:
Do you have libusb? Btw, I'm Finnish too :D

JonnyTech, 2017-09-28:
IIRC you need the libusb-dev (development) libraries to compile and build.

RedKugelblitz, 2017-09-28:
already have it. maybe the libusb1.0-0-something-something aren't compatible with the libusb-something?

FinKsu, 2017-09-30:
What version of Debian are you using?

RedKugelblitz, 2017-09-30:
9.10 stretch

RedKugelblitz, 2017-09-30:
Could be relevant: https://stackoverflow.com/questions/23549833/linking-against-libusb-on-ubuntu-doesnt-work

RedKugelblitz, 2017-09-30:
I just don't know how cmake works, gotta do a little bit of reading

RedKugelblitz, 2017-09-30:
Update: I tried to download the repo again, this time using "git clone" (previously a zip of the master branch) and it worked. I think someone pushed a fix for the issue already. I'm sorry i can't provide more info on the issue. Thanks for the help everyone!

FinKsu, 2017-10-01:
Nice that it works!

FinKsu, 2017-10-01:
I think you can now close the issue, as you don't have it anymore

closed 2017-10-23 by RedKugelblitz

diepes, 2017-12-11:
Solved a similar issue by editing
~/git/Heimdall/build [master|…1] 
17:45 $ gvim heimdall/CMakeFiles/heimdall.dir/link.txt

adding -lusb-1.0  to the end.

found the flag with
$ pkg-config --list-all | grep usb
$ pkg-config --libs libusb-1.0

jbruce12000, 2023-01-24:
This should not be closed. This is still a bug and the suggestion above fixed it for me. I had to manually edit

./build/heimdall/CMakeFiles/heimdall.dir/link.txt

and change -lusb to -lusb-1.0