Benjamin-Dobell/Heimdall #426: Cannot retrieve PIT from samsung GT-I9195

milari, 2017-10-04:
Hi.
Here verbose output of "detect" command:
# heimdall detect --verbose --usb-log-level debug
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.002148] [0000128e] libusb: debug [libusb_get_device_list] 
[ 0.002192] [0000128e] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.002204] [0000128e] libusb: debug [libusb_get_device_descriptor] 
[ 0.002211] [0000128e] libusb: debug [libusb_get_device_descriptor] 
Device detected
[ 0.002232] [0000128e] libusb: debug [libusb_exit] 
[ 0.002239] [0000128e] libusb: debug [libusb_exit] destroying default context

It seems to be ok, but when I ask to print-pit:

# heimdall print-pit --usb-log-level debug
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.005591] [0000129e] libusb: debug [libusb_get_device_list] 
[ 0.005660] [0000129e] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.005693] [0000129e] libusb: debug [libusb_get_device_descriptor] 
[ 0.005705] [0000129e] libusb: debug [libusb_get_device_descriptor] 
[ 0.005718] [0000129e] libusb: debug [libusb_open] open 2.4
[ 0.005752] [0000129e] libusb: debug [usbi_add_pollfd] add fd 11 events 4
[ 0.005777] [0000129e] libusb: debug [libusb_get_device_descriptor] 
[ 0.005789] [0000129e] libusb: debug [libusb_get_config_descriptor] index 0
Claiming interface...
[ 0.005819] [0000129e] libusb: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 0.005866] [0000129e] libusb: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0

Initialising protocol...
[ 0.006203] [0000129e] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.006240] [0000129e] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 0.006286] [0000129e] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.006301] [0000129e] libusb: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.006471] [0000129e] libusb: debug [handle_events] poll() returned 1
[ 0.006529] [0000129e] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=4
[ 0.006555] [0000129e] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.006575] [0000129e] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 0.006595] [0000129e] libusb: debug [disarm_timerfd] 
[ 0.006616] [0000129e] libusb: debug [usbi_handle_transfer_completion] transfer 0x9f18f28 has callback 0xb779dcd0
[ 0.006634] [0000129e] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.006670] [0000129e] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.006684] [0000129e] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 7
[ 0.006705] [0000129e] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.006718] [0000129e] libusb: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 1.006725] [0000129e] libusb: debug [handle_events] poll() returned 1
[ 1.006775] [0000129e] libusb: debug [handle_events] timerfd triggered
[ 1.006794] [0000129e] libusb: debug [libusb_cancel_transfer] 
[ 1.007136] [0000129e] libusb: debug [disarm_timerfd] 
[ 1.007171] [0000129e] libusb: debug [handle_events] poll() 4 fds with timeout in 0ms
[ 1.007186] [0000129e] libusb: debug [handle_events] poll() returned 1
[ 1.007201] [0000129e] libusb: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[ 1.007210] [0000129e] libusb: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[ 1.007219] [0000129e] libusb: debug [handle_bulk_completion] abnormal reap: urb status -2
[ 1.007227] [0000129e] libusb: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[ 1.007235] [0000129e] libusb: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[ 1.007243] [0000129e] libusb: debug [disarm_timerfd] 
[ 1.007251] [0000129e] libusb: debug [usbi_handle_transfer_completion] transfer 0x9f19190 has callback 0xb779dcd0
[ 1.007260] [0000129e] libusb: debug [sync_transfer_cb] actual_length=0
ERROR: Protocol initialisation failed!

Releasing device interface...
[ 1.007312] [0000129e] libusb: debug [libusb_release_interface] interface 1

[ 1.007352] [0000129e] libusb: debug [libusb_close] 
[ 1.007372] [0000129e] libusb: debug [usbi_remove_pollfd] remove fd 11
[ 1.007395] [0000129e] libusb: debug [libusb_exit] 
[ 1.007403] [0000129e] libusb: debug [libusb_exit] destroying default context

Can You help me?
Thanks.

JonnyTech, 2017-10-04:
Try version 1.4.2

milari, 2017-10-05:
No more luck...
---
$ sudo bin/heimdall detect
Device detected
$ sudo bin/heimdall print-pit --no-reboot                               
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...
ERROR: Failed to begin session!
Releasing device interface...

$ sudo bin/heimdall print-pit --no-reboot --usb-log-level debug         
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.005259] [00003d82] libusbx: debug [libusb_get_device_list] 
[ 0.005308] [00003d82] libusbx: debug [discovered_devs_append] need to increase capacity
[ 0.005325] [00003d82] libusbx: debug [libusb_get_device_descriptor] 
[ 0.005334] [00003d82] libusbx: debug [libusb_get_device_descriptor] 
[ 0.005345] [00003d82] libusbx: debug [libusb_open] open 2.9
[ 0.005385] [00003d82] libusbx: debug [usbi_add_pollfd] add fd 11 events 4
[ 0.005411] [00003d82] libusbx: debug [libusb_get_device_descriptor] 
[ 0.005420] [00003d82] libusbx: debug [libusb_get_config_descriptor] index 0
Claiming interface...
[ 0.005449] [00003d82] libusbx: debug [libusb_claim_interface] interface 1
Setting up interface...
[ 0.005488] [00003d82] libusbx: debug [libusb_set_interface_alt_setting] interface 1 altsetting 0

Initialising protocol...
[ 0.005747] [00003d82] libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.005794] [00003d82] libusbx: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 4
[ 0.005852] [00003d82] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.005888] [00003d82] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 0.005928] [00003d82] libusbx: debug [handle_events] poll() returned 1
[ 0.005963] [00003d82] libusbx: debug [reap_for_handle] urb type=3 status=0 transferred=4
[ 0.005995] [00003d82] libusbx: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[ 0.006027] [00003d82] libusbx: debug [handle_bulk_completion] last URB in transfer --> complete!
[ 0.006056] [00003d82] libusbx: debug [disarm_timerfd] 
[ 0.006089] [00003d82] libusbx: debug [usbi_handle_transfer_completion] transfer 0x1831e48 has callback 0x7ff9989fa7f0
[ 0.006121] [00003d82] libusbx: debug [sync_transfer_cb] actual_length=4
[ 0.006165] [00003d82] libusbx: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.006202] [00003d82] libusbx: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 7
[ 0.006246] [00003d82] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.006278] [00003d82] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 1.006244] [00003d82] libusbx: debug [handle_events] poll() returned 1
[ 1.006337] [00003d82] libusbx: debug [handle_events] timerfd triggered
[ 1.006380] [00003d82] libusbx: debug [libusb_cancel_transfer] 
[ 1.006671] [00003d82] libusbx: debug [disarm_timerfd] 
[ 1.006733] [00003d82] libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.006767] [00003d82] libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
[ 1.006807] [00003d82] libusbx: debug [handle_events] poll() returned 1
[ 1.006847] [00003d82] libusbx: debug [reap_for_handle] urb type=3 status=-2 transferred=0
[ 1.006880] [00003d82] libusbx: debug [handle_bulk_completion] handling completion status -2 of bulk urb 1/1
[ 1.006911] [00003d82] libusbx: debug [handle_bulk_completion] abnormal reap: urb status -2
[ 1.006942] [00003d82] libusbx: debug [handle_bulk_completion] abnormal reap: last URB handled, reporting
[ 1.006973] [00003d82] libusbx: debug [usbi_handle_transfer_cancellation] detected timeout cancellation
[ 1.007004] [00003d82] libusbx: debug [disarm_timerfd] 
[ 1.007035] [00003d82] libusbx: debug [usbi_handle_transfer_completion] transfer 0x1831e48 has callback 0x7ff9989fa7f0
[ 1.007066] [00003d82] libusbx: debug [sync_transfer_cb] actual_length=0
ERROR: Protocol initialisation failed!

Releasing device interface...
[ 1.007210] [00003d82] libusbx: debug [libusb_release_interface] interface 1

[ 1.007299] [00003d82] libusbx: debug [libusb_close] 
[ 1.007347] [00003d82] libusbx: debug [usbi_remove_pollfd] remove fd 11
[ 1.007453] [00003d82] libusbx: debug [libusb_exit] 
[ 1.007488] [00003d82] libusbx: debug [libusb_exit] destroying default context

julianfoad, 2024-05-25:
Duplicate of #413? I just added some help there.