Benjamin-Dobell/Heimdall #428: Can't compile on MacOS Sierra 10.12.6

derwentx, 2017-10-09:
Hi There, here are the steps I've used to try to compile Heimdall on my system from the latest unmodified version on GitHub as per the instructions in [appendix B](https://github.com/Benjamin-Dobell/Heimdall/blob/9bcc42da350bd2e1766980bcb77d806a82d56a1d/OSX/README.txt#L480)

With additional tips mentioned [here](https://github.com/Benjamin-Dobell/Heimdall/issues/311) when the vanilla instructions didn't work.

```bash
$ pwd
```
```
/Users/derwent/GitHub/Heimdall/build
```
```bash
$ brew install libusb qt5 cmake
```
```
Warning: libusb 1.0.21 is already installed
Warning: qt 5.9.1 is already installed
Warning: cmake 3.9.4 is already installed
```
```bash
$ cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release -DQt5Widgets_DIR=/usr/local/opt/qt5/lib/cmake/Qt5Widgets -DDISABLE_FRONTEND=ON ..
```
```

-- The C compiler identification is AppleClang 9.0.0.9000037
-- The CXX compiler identification is AppleClang 9.0.0.9000037
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found libusb: /usr/local/lib/libusb-1.0.a
-- Checking if large (64-bit) file support is available...
-- Checking if large (64-bit) file support is available - yes
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    Qt5Widgets_DIR

-- Build files have been written to: /Users/derwent/GitHub/Heimdall/build
```
```bash
$ LIBRARY_PATH=/usr/local/lib make
```
```
[ 12%] Built target pit
[ 18%] Linking CXX executable ../bin/heimdall
Undefined symbols for architecture x86_64:
  "_CFDataGetBytes", referenced from:
      _process_new_device in libusb-1.0.a(libusb_1_0_la-darwin_usb.o)
  "_CFDataGetLength", referenced from:
      _process_new_device in libusb-1.0.a(libusb_1_0_la-darwin_usb.o)
  "_CFDataGetTypeID", referenced from:
      _process_new_device in libusb-1.0.a(libusb_1_0_la-darwin_usb.o)
  "_CFGetTypeID", referenced from:
      _process_new_device in libusb-1.0.a(libusb_1_0_la-darwin_usb.o)
      _get_ioregistry_value_number in libusb-1.0.a(libusb_1_0_la-darwin_usb.o)
  "_CFNumberGetTypeID", referenced from:
      _get_ioregistry_value_number in libusb-1.0.a(libusb_1_0_la-darwin_usb.o)
  "_CFNumberGetValue", referenced from:
      _get_ioregistry_value_number in libusb-1.0.a(libusb_1_0_la-darwin_usb.o)
  "_CFRelease", referenced from:
[... 150 lines cut ...]
  /Users/derwent/.pyenv/shims/python3.3-config
  /Users/derwent/.pyenv/shims/python3.3m-config
  /Users/derwent/.pyenv/shims/python3.4-config
  /Users/derwent/.pyenv/shims/python3.4m-config
  /Users/derwent/.pyenv/shims/python3.6-config
  /Users/derwent/.pyenv/shims/python3.6m-config

Warning: You have a curlrc file
If you have trouble downloading packages with Homebrew, then maybe this
is the problem? If the following command doesn't work, then try removing
your curlrc:
  curl https://github.com
```
And some info about my system:
```
Distro: OS X 10.12.6
Shell: /bin/zsh
CPU: Intel Core i7-4960HQ CPU @ 2.60GHz
Memory: 16 GB
Disk: 91%
```

When I install using the installer, it works fine but I'm trying to install from source because I'm having issues with my Samsung Galaxy S5 klte not downloading the pit file on my windows or mac machines.

Thanks.

johnny13, 2017-10-15:
@derwentx I dont think you are using the latest release. Check the releases tab and download 1.4.2.
Im running Sierra 10.12.6 also and had the same error until I used the latest release.

```bash
cd ~/Downloads/Heimdall-1.4.2/
mkdir build
cd build
```

Then running this command `cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release -DQt5Widgets_DIR=/usr/local/opt/qt5/lib/cmake/Qt5Widgets ..` gave me this output...

```bash
-- The C compiler identification is AppleClang 9.0.0.9000037
-- The CXX compiler identification is AppleClang 9.0.0.9000037
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found libusb: /usr/local/lib/libusb-1.0.a  
-- Checking if large (64-bit) file support is available...
-- Checking if large (64-bit) file support is available - yes
-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.8") 
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/derekscott/Downloads/Heimdall-1.4.2/build
```

Then running `make` gave me this...

```bash
Scanning dependencies of target pit
[  3%] Building CXX object libpit/CMakeFiles/pit.dir/source/libpit.cpp.o
[  6%] Linking CXX static library libpit.a
[  6%] Built target pit
Scanning dependencies of target heimdall
[  9%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Arguments.cpp.o
[ 12%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o
[ 15%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/ClosePcScreenAction.cpp.o
[ 18%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DetectAction.cpp.o
[ 21%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DownloadPitAction.cpp.o
[ 24%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/FlashAction.cpp.o
[ 27%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/HelpAction.cpp.o
[ 30%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/InfoAction.cpp.o
[ 33%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Interface.cpp.o
[ 36%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/main.cpp.o
[ 39%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/PrintPitAction.cpp.o
[ 42%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Utility.cpp.o
[ 45%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/VersionAction.cpp.o
[ 48%] Linking CXX executable ../bin/heimdall
[ 48%] Built target heimdall
Scanning dependencies of target heimdall-frontend_autogen
[ 51%] Generating qrc_mainwindow.cpp
RCC: Warning: No resources in '/Users/derekscott/Downloads/Heimdall-1.4.2/heimdall-frontend/mainwindow.qrc'.
[ 54%] Generating ui_mainwindow.h
[ 57%] Generating ui_aboutform.h
[ 60%] Automatic MOC for target heimdall-frontend
Generating MOC source CAWN4ZDDTK/moc_aboutform.cpp
Generating MOC source CAWN4ZDDTK/moc_mainwindow.cpp
Generating MOC compilation mocs_compilation.cpp
[ 60%] Built target heimdall-frontend_autogen
Scanning dependencies of target heimdall-frontend
[ 63%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/aboutform.cpp.o
[ 66%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/Alerts.cpp.o
[ 69%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/FirmwareInfo.cpp.o
[ 72%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/main.cpp.o
[ 75%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/mainwindow.cpp.o
[ 78%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/PackageData.cpp.o
[ 81%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/Packaging.cpp.o
[ 84%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/qrc_mainwindow.cpp.o
[ 87%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/heimdall-frontend_autogen/mocs_compilation.cpp.o
[ 90%] Linking CXX executable ../bin/heimdall-frontend.app/Contents/MacOS/heimdall-frontend
[100%] Built target heimdall-frontend
```

mateuszk87, 2017-10-17:
Doesn't work for me either...

derwentx, 2017-10-26:
Hey @johnny13 ,
Looks like my repo was indeed on the very latest version in GitHub, it's just a more recent commit than yours:
```bash
git show
```
```
commit 9bcc42da350bd2e1766980bcb77d806a82d56a1d (HEAD -> master, origin/master, origin/HEAD)
Merge: 3b4f34e cc44b34
Author: Benjamin Dobell <benjamin.dobell+github@glassechidna.com.au>
Date:   Wed Sep 20 12:05:45 2017 +1000

    Merge pull request #408 from grayed/master

    Use library found instead of hardcoding compiler flag.

```
```bash
cat heimdall/source/Interface.cpp | grep 'version ='
```
```c
const char *version = "v1.4.2";
```
```bash
cat heimdall-frontend/aboutform.ui| grep Version
```
```ui
&lt;p style=&quot; margin-top:0px; margin-bottom:0px; margin-left:0px; margin-right:0px; -qt-block-indent:0; text-indent:0px;&quot;&gt;Version 1.4.2&lt;/p&gt;
```

The only difference between between our cmake outputs is that yours said `-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.8") ` so i ran brew install zlib and tried your make command:

```bash
cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release -DQt5Widgets_DIR=/usr/local/opt/qt5/lib/cmake/Qt5Widgets ..
```
```
-- The C compiler identification is AppleClang 9.0.0.9000037
-- The CXX compiler identification is AppleClang 9.0.0.9000037
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found libusb: /usr/local/lib/libusb-1.0.a
-- Checking if large (64-bit) file support is available...
-- Checking if large (64-bit) file support is available - yes
-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.11")
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/derwent/GitHub/Heimdall/build
```
The only difference now is that mine says `found version "1.2.11"` instead of `1.2.8`.
However `make` still fails:
```
Scanning dependencies of target pit
[  3%] Building CXX object libpit/CMakeFiles/pit.dir/source/libpit.cpp.o
[  6%] Linking CXX static library libpit.a
[  6%] Built target pit
Scanning dependencies of target heimdall
[  9%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/Arguments.cpp.o
[ 12%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/BridgeManager.cpp.o
[ 15%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/ClosePcScreenAction.cpp.o
[ 18%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DetectAction.cpp.o
[ 21%] Building CXX object heimdall/CMakeFiles/heimdall.dir/source/DownloadPitAction.cpp.o
[... 205 lines cut ...]
Scanning dependencies of target heimdall-frontend
[ 63%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/aboutform.cpp.o
[ 66%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/Alerts.cpp.o
[ 69%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/FirmwareInfo.cpp.o
[ 72%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/main.cpp.o
[ 75%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/mainwindow.cpp.o
[ 78%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/PackageData.cpp.o
[ 81%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/source/Packaging.cpp.o
[ 84%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/qrc_mainwindow.cpp.o
[ 87%] Building CXX object heimdall-frontend/CMakeFiles/heimdall-frontend.dir/heimdall-frontend_autogen/mocs_compilation.cpp.o
In file included from /Users/derwent/GitHub/Heimdall/build/heimdall-frontend/heimdall-frontend_autogen/mocs_compilation.cpp:2:
/Users/derwent/GitHub/Heimdall/build/heimdall-frontend/heimdall-frontend_autogen/CAWN4ZDDTK/moc_aboutform.cpp:9:10: fatal error:
      '../../../../../../Documents/GitHub/Heimdall/heimdall-frontend/source/aboutform.h' file not found
#include "../../../../../../Documents/GitHub/Heimdall/heimdall-frontend/source/aboutform.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [heimdall-frontend/CMakeFiles/heimdall-frontend.dir/heimdall-frontend_autogen/mocs_compilation.cpp.o] Error 1
make[1]: *** [heimdall-frontend/CMakeFiles/heimdall-frontend.dir/all] Error 2
make: *** [all] Error 2
```
However I am able to `make` with -DDISABLE_FRONTEND=ON on this version.

It seems like commit https://github.com/Benjamin-Dobell/Heimdall/commit/cc44b3497db0b705e5c337bce0a45416728189c9 #408 breaks the linking stage for my system, any thoughts @Benjamin-Dobell  @grayed ?

Thanks.

Fimagena, 2017-11-05:
Had the same issue. The problem is not libusb itself but that it requires the CoreFoundation and IOKit frameworks that are not explicitly linked in.

There might be better ways to fix this (I'm not a Mac expert) but adding the below to the `heimdall/CMakeLists.txt` file worked for me

    find_library(CF CoreFoundation)
    find_library(IOK IOKit)
    target_link_libraries(heimdall ${CF} ${IOK})

gaellalire, 2018-01-17:
Same problem.
Solved by adding the below in `heimdall/CMakeLists.txt`
```
set(CMAKE_EXE_LINKER_FLAGS "-lobjc -framework IOKit -framework CoreFoundation")
```
Same solution as the @Fimagena one except I don't see how it can found `-lobjc`

etweek, 2023-03-15:
For anyone coming from Mojave, you now need to add `Security` too:
Near the top of `heimdall/CMakeLists.txt`:
```
find_package(libusb REQUIRED)
find_library(CF CoreFoundation)
find_library(IOK IOKit)
find_library(Security Security)
```
(i.e. add three lines to the existing libusb)
and near the bottom:
```
target_link_libraries(heimdall PRIVATE pit)
target_link_libraries(heimdall PRIVATE ${LIBUSB_LIBRARY})
target_link_libraries(heimdall objc ${CF} ${IOK} ${Security})
```
(last line new)

and then it compiles