Benjamin-Dobell/Heimdall #442: Heimdall 1.4.0 can't be installed on MacOS 10.13.2

headsvk, 2017-12-21:


Benjamin-Dobell, 2017-12-21:
There's several issues, both open and closed, regarding this; please search before opening a new issue.

closed 2017-12-21 by Benjamin-Dobell

headsvk, 2017-12-21:
@Benjamin-Dobell  No, there's no issue logged for High Sierra, I did check.

Benjamin-Dobell, 2017-12-21:
1.4.0 isn't even the latest version of Heimdall - it's literally several years old. Please do some preliminary research first. There are several issues open regarding macOS and Heimdall.

headsvk, 2017-12-21:
@Benjamin-Dobell yes some unrelated issues, great. https://glassechidna.com.au/heimdall/#downloads isn't this the official website? why does it link to an old version then? what research do you mean?

Benjamin-Dobell, 2017-12-21:
TLDR; you need to build it yourself.

Because this is an open source project and I give up a huge amount of my life developing it and maintaining it, with almost zero community assistance, and little-to-no thanks. I don't have time to build new releases.

You're welcome to contribute - as is everyone else.

Like I said I'm busy enough as is, please do some research and use your own time, rather than wasting mine asking questions that can be easily answered in just a couple fo minute of searching.

headsvk, 2017-12-21:
I came from LineageOS install instructions that happily link to this project, so no wonder people come here and create new issues. I understand that you're not maintaining this project anymore but then it should be written on the homepage and nobody will lose time bothering you with it.

Benjamin-Dobell, 2017-12-21:
I am maintaining the project. Who said otherwise? I said I'm not building binaries because it's assumed that people flashing their phones have a clue about how to follow a README. Unfortunately, sometimes that's too much to ask.

headsvk, 2017-12-21:
On the website, there's zero information about having to build the binaries yourself and there's also no info about the version offered to download so people have no idea that it's old and doesn't work. If the download link would take us to your README instead then we could follow it right away.

mstarke, 2017-12-26:
I have to chime in for @Benjamin-Dobell it's easy to ask for minor changes in documentation or websites but this is sometimes the last place a maintainer works on since the main goal is to keep the software working. I have similar experiences and can fully relate to the frustration from issues being created by people unable or unwilling to properly research the project. The criticism by @headsvk is valid though but if external resources link to the project it gets even not complicated as this might lead to different expectations. I came here because the installer wasn't supported on macOS 10.12 and found multiple issues on GitHub so it's possible to find answers but it requires a bit of digging to get heimdall to work which is what Benjamin is asking for.