Benjamin-Dobell/Heimdall #447: Not connecting after PIT file recovery

martinsson, 2018-01-01:
I get "Disconnected" status in the UI just below "ID:COM".

Actually this only happens after it is rebooted by heimdall after recovering the PIT file. When it reboots and the phone goes into download mode then JOdin3 is not seeing the phone. Actually it seems that it is only the UI that is now working, because the logs says "Heimdall Device detected!"

Here's a log of the attempt

```
[VERBOSE]Heimdall Device detected!
[INTERACTION][SHOWYESNO] Title:Do you want me to obtain a PIT for you?
Message:This application requires what is known as a 'PIT file'.  The PIT file tells the application where to place files on your device.  If you don't have a PIT file, we can obtain one from your device
press: 

[VERBOSE]Heimdall Device detected!
[VERBOSE]obtaining pit
[VERBOSE]showing information message object
[INTERACTION][SHOWINFORMATION] Title:You will need to restart your device in Download Mode.
Message:In order to obtain a PIT file, the device will be rebooted.  Once it reboots, you will need to put it back into download mode.\\n\\nPro-Tip: hold the download mode combination and press OK to dismiss this dislog to reboot into download mode immediately
press: Empty

[VERBOSE]Heimdall Device detected!
[VERBOSE]Done with message object
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]TimeOut on /usr/local/bin/heimdall after 10000ms. Returning what was received.
[VERBOSE]found pit
[VERBOSE]showing information message object
[INTERACTION][SHOWINFORMATION] Title:Got it!
Message:We obtained the PIT file and everything is ready to flash\\n\\n Click the start button again when you're ready. 
press: Empty

[DEBUG]Attempting state change to 0 devices connected
[DEBUG]Device disconnected commanded
[DEBUG]Device Removed
[DEBUG]UI did not respond to state change, retrying.
[DEBUG]Attempting state change to 0 devices connected
[DEBUG]Device disconnected commanded
[DEBUG]Device Removed
[DEBUG]UI did not respond to state change, retrying.
[DEBUG]Attempting state change to 0 devices connected
[DEBUG]Device disconnected commanded
[DEBUG]Device Removed
[DEBUG]UI did not respond to state change, retrying.
[VERBOSE]Done with message object
[DEBUG]Attempting state change to 0 devices connected
[DEBUG]Device disconnected commanded
[DEBUG]Device Removed
[DEBUG]UI did not respond to state change, retrying.
[DEBUG]Attempting state change to 0 devices connected
[DEBUG]Device disconnected commanded
[DEBUG]Device Removed
[DEBUG]UI did not respond to state change, retrying.
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
```

I'm trying to flash a galaxy s7, model G930V and I'm using java jdk1.8.0_152 on MacOS Sierra 10.12.6 (16G1036)

Cheers

JonnyTech, 2018-01-01:
The log does not show Heimdall version, can you post full log from terminal including commands entered.

martinsson, 2018-01-01:
Sure, here's the full log and a caption of the UI

[image]

```
/Library/Java/JavaVirtualMachines/jdk1.8.0_152.jdk/Contents/Home/bin/java -jar app/JOdin3CASUAL.jar
Jan 01, 2018 6:43:43 PM com.sun.javafx.css.parser.CSSParser declaration
WARNING: CSS Error parsing '*{-fx-font: 18px "Arial"; -fx-background-color: rgba(0.0.0.0); -fx-text-fill: #AAAAFFFF; -fx-padding: 15;}: Expected '<number>' or '<percentage>' while parsing '-fx-background-color' at [1,53]
[DEBUG]Diagnostics OS:Mac OS X Version:10.12.6 Architecture:x86_64 Separator:/ | Java:Oracle Corporation Version:1.8.0_152 Website:http://java.oracle.com/ | WorkingDir: /Users/johan/Downloads/JOdin3CASUAL-r1142-dist CASUALDir:/var/folders/s8/snh3nx3n1xx7jdmjh7qsbqh00000gp/T/CASUALjohan-2018-01-01-18.43.43/ | CASUAL is handling this system as:Mac 64 bit -- End Diag
[DEBUG]native search /usr/local/bin/heimdall:true
[VERBOSE]Heimdall Device detected!
[VERBOSE]Starting: mode:Heimdall
Connected:true

[DEBUG]Beginning MandatoryThread counter.casual-dev.com
[VERBOSE]Heimdall Device detected!
[DEBUG]Attempting state change to 1 devices connected
[DEBUG]Device Connected
[DEBUG]@stateConnected
[INTERACTION][SHOWYESNO] Title:Do you want me to obtain a PIT for you?
Message:This application requires what is known as a` 'PIT file'.  The PIT file tells the application where to place files on your device.  If you don't have a PIT file, we can obtain one from your device
press: 

[VERBOSE]Heimdall Device detected!
[VERBOSE]obtaining pit
[VERBOSE]showing information message object
[INTERACTION][SHOWINFORMATION] Title:You will need to restart your device in Download Mode.
Message:In order to obtain a PIT file, the device will be rebooted.  Once it reboots, you will need to put it back into download mode.\\n\\nPro-Tip: hold the download mode combination and press OK to dismiss this dislog to reboot into download mode immediately
press: Empty

[VERBOSE]Heimdall Device detected!
[VERBOSE]Done with message object
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]TimeOut on /usr/local/bin/heimdall after 10000ms. Returning what was received.
[VERBOSE]found pit
[VERBOSE]showing information message object
[INTERACTION][SHOWINFORMATION] Title:Got it!
Message:We obtained the PIT file and everything is ready to flash\\n\\n Click the start button again when you're ready. 
press: Empty

[DEBUG]Attempting state change to 0 devices connected
[DEBUG]Device disconnected commanded
[DEBUG]Device Removed
[DEBUG]UI did not respond to state change, retrying.
[DEBUG]Attempting state change to 0 devices connected
[DEBUG]Device disconnected commanded
[DEBUG]Device Removed
[DEBUG]UI did not respond to state change, retrying.
[VERBOSE]Done with message object
[DEBUG]Attempting state change to 0 devices connected
[DEBUG]Device disconnected commanded
[DEBUG]Device Removed
[DEBUG]UI did not respond to state change, retrying.
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
[VERBOSE]Heimdall Device detected!
```

martinsson, 2018-01-01:
Actually I installed the version referenced in this issue #291, should be 1.4.1 I guess

JonnyTech, 2018-01-01:
This is a support area for Heimdall, please submit the error to the authors of JOdin3.

martinsson, 2018-01-01:
JOdin3 is the GUI I understand

JonnyTech, 2018-01-01:
Why not use Heindall or Heimdall frontend?

martinsson, 2018-01-01:
Because I didn't realise that it was more than a driver :). The command line interface seems great. Thanks! 👍

closed 2018-01-01 by martinsson