Benjamin-Dobell/Heimdall #448: Heimdall 1.41 fails to download PIT from Galaxy Note 3 (SM-900P) under Debian Testing (Buster)

CarlFink, 2018-01-02:
Hello. Trying to install LineageOS on my Galaxy Note 3 following the instructions here: https://wiki.lineageos.org/devices/hlte/install

Using Heimdall from the Debian repo.

I use the instructions and get:

```
carlf@debian-nitpicking:~/Downloads$ heimdall flash --RECOVERY twrp-3.2.1-0-hlte.img --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Unexpected handshake response!
ERROR: Protocol initialisation failed!

Releasing device interface...
```
Not always, though. Sometimes it's 

```
carlf@debian-nitpicking:~/Downloads$ heimdall flash --RECOVERY twrp-3.2.1-0-hlte.img --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
```
That's with a USB 2.0 cable. With USB3 (in a USB3 connector on the PC) Heimdall simply does not detect the phone.  When it tries download the PIT, one blue dot of the progress bar appears, then everything stops. Finally I have to power-down the phone, ignoring the scary **Do not turn off the target ...** message.

Suggestions?

myrdd, 2018-01-14:
Hi @CarlFink

I ran into the same issue on Debian 9 (Stretch) using Debian's version of Heimdall, which is `1.4.1`.

I resolved the issue by building the latest source (commit 5377b6286c10edb21e94d1c8efaa961bc53a7c8f).

If that does not work for you, use the command line options `--verbose` and `--usb-log-level debug`. Before I switched to the latest source I've been seeing `ERROR: libusb error -7 whilst sending bulk transfer`. Take a look at https://github.com/Benjamin-Dobell/Heimdall/issues/197#issuecomment-144864083, maybe experimenting with USB ports and devices helps you working around the issue.

CarlFink, 2018-01-14:
Thanks, but the actual phone had a hardware failure just after I posted that. The USB port physically failed. It's an old enough phone that I'm not going to fix it, so the entire thing became moot. I appreciate your following up, though.

closed 2018-01-14 by CarlFink