Benjamin-Dobell/Heimdall #453: Samsung Galaxy S8 userdata partition problem

TimoSairiala, 2018-02-06:
I think this is not actually an problem with Heimdall, but rather something around it. But as I understand here I would find many peoples who understands this area, maybe I can find some help.

So I am trying to generate and flash an userdata partition on galaxy S8 but the phone wont accept my image. I can do this without any problems for older samsung devices.

What I would do is
1) make an sparse image on ubuntu (with make_ext4fs from android-tools-fsutils package)
make_ext4fs -l [lenght] -s userdata.img [directory_to_my_content]
2) boot the device on Download mode
3) flash it with heimdall flash --userdata userdata.img 

Now, on galaxy S8 the flashing goes ok for some time, but in around 40% it just stops and I have to reboot the device manually.

I checked many samsung firmware packages and usually they doesn't contain userdata image, but this one has it:
https://samsung-firmware.org/download/Galaxy__S8__/4v72/NEE/G950FXXU1AQJ1/G950FOXM1AQJ1/
and this userdata image I can flash ok with heimdall. It takes hours and days to download it so I have copied the userdata image here: https://1drv.ms/u/s!AlTfydQsL14YhKIifEMOccrcobcj6A

I did simg2img it and after mounting I can see it has plenty of apps and cache and some other files in it. First I thought that the userdata image has have something like this in it or it will fail. I tried to make an new sparse image with this content included, but the flashing still fails. I tried to make it a bit smaller, but it always fails.

If I see 
```
file userdata.img
userdata.img: Android sparse image, version: 1.0, Total of 2621440 4096-byte output blocks in 447 input chunks.
```
Nothing special there, but if I see 
```
simg_dump userdata.img
simg_dump: userdata.img: The file header size was expected to be 28, but is 32.
```
it wont dump the file because header size is something else than expected? Maybe this is the reason why the device wont accept the file?

So I am not that sure if there is a way for me to generate flashable userdata image for this phone? Or maybe I am lost and this header mismatch is not the problem, but there is something else I should check or do? Do you have any hints or clues from where I could begin?

TimoSairiala, 2018-02-28:
Just in case somebody is also wondering the same thing, I came across another github project that has a D script to samsungify the images:
https://github.com/anestisb/android-simg2img/blob/master/simg_samsungify.d
So like expected, nothing wrong with the heimdall, it is just that samsung has their own format they like to use with some images. They have also better version of simg_dump tool that can dump images with 32 byte headers, so it is a nice project.

closed 2018-02-28 by TimoSairiala

TimoSairiala, 2018-03-13:
yeah, it all works with the samsung's own images. My problem was with generating the userdata image as it seems samsung has a bit special format.

opensrx, 2018-03-13:
You've used the flashing command without using the "userdata" argument? If not, try this way.
If you want to get your data back on the phone, there is many ways, the easier one is to use google backup system before. The second one if you didn't and if you've did a copy of your data image file is to unpack it and get your files back.

opensrx, 2018-03-13:
You can perform a compilation of the firmware.xml to change some value and try again too. Take a look on the readme file if you are interested to solve the flashing matter.