Benjamin-Dobell/Heimdall #454: v1.4.1 softbricking SPH-P100 tablets with the flash command "heimdall flash --KERNEL <filename>"

lsimplify, 2018-02-15:
I bricked a **SPH-P100** trying to flash the kernel partition with the boot.img file from this archive: cm-11-20140212-terenceng-p1c.zip (**sha1sum** is 1b80087efa191a992f2b0bdee0cfafab42d0d69f)
In case it matters, this is the "**Sprint firmware version**" of the SPH-P100.

(I think the tablet is actually in "download mode" because it has this little graphic on the screen, so that is good.)

Command that I used was basic flashing command. Just `heimdall flash --KERNEL <filename>`

**And I got this error:** ERROR: Failed to send file part packet!

The error message made me think following this advice will prevent the error:  #347 

I'm following the advice and will try to flash it tomorrow, but if this doesn't fix the flash functionality, I'll apply the changes to the source code that are mentioned here: #14

I'll use the verbose option next time until we get it to work.

lsimplify, 2018-02-16:
Update: `./heimdall detect --usb-log-level debug --verbose --stdout-errors` cannot detect the device. 

Plugging in the device doesn't cause any messages to show up in with `sudo dmesg --level debug`

The graphic on the screen only shows up when connecting it to the wall. (Maybe it will detect in Windows with Odin, but I don't know right now. Tips/advice is appreciated!

I have more working SPH-P100 tablets, but I can't just willy-nilly try stuff (and brick them). I need some suggestions on how to get heimdall to not fail in the first place like it did in my first post. _Suggestions should be on logical foundations._

P.S.: These are stock unrooted tablets in case that means anything in regard to why it is failing.

lsimplify, 2018-02-16:
I just came up with the idea of trying to flash non-critical partitions. If flashing fails, the SPH-P100 should still boot. Once flashing works, I'll try to flash the kernel partition.

lsimplify, 2018-02-16:
This was unexpected... the binary with the code in #347 seems to have not done anything when I tried to flash the modem, and when I used the latest unmodified source, the same 'ol brick resulted. The device is still detected, but I do not understand why a command to flash the modem partition results in a brick with the yellow triangle just displaying. 

```
~/Heimdall/build/bin/heimdall flash --MODEM modem.bin --no-reboot --usb-log-level debug --verbose --stdout-errors
Heimdall v1.4.2
...
Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.006488] [00000e27] libusb: debug [libusb_get_device_list] 
[ 0.006526] [00000e27] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.006539] [00000e27] libusb: debug [libusb_get_device_descriptor] 
[ 0.006546] [00000e27] libusb: debug [libusb_get_device_descriptor] 
[ 0.006550] [00000e27] libusb: debug [libusb_get_device_descriptor] 
[ 0.006558] [00000e27] libusb: debug [libusb_get_device_descriptor] 
[ 0.006568] [00000e27] libusb: debug [libusb_get_device_descriptor] 
[ 0.006578] [00000e27] libusb: debug [libusb_get_device_descriptor] 
[ 0.006587] [00000e27] libusb: debug [libusb_get_device_descriptor] 
[ 0.006597] [00000e27] libusb: debug [libusb_get_device_descriptor] 
[ 0.006608] [00000e27] libusb: debug [libusb_open] open 2.15
[ 0.006654] [00000e27] libusb: debug [usbi_add_pollfd] add fd 10 events 4
[ 0.006664] [00000e27] libusb: debug [libusb_get_device_descriptor] 
[ 0.006680] [00000e27] libusb: debug [libusb_alloc_transfer] transfer 0x55bf58852630
[ 0.006690] [00000e27] libusb: debug [libusb_submit_transfer] transfer 0x55bf58852630
[ 0.006703] [00000e27] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.006734] [00000e27] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.006744] [00000e27] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.006756] [00000e27] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.006821] [00000e27] libusb: debug [handle_events] poll() returned 1
[ 0.006839] [00000e27] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.006851] [00000e27] libusb: debug [handle_control_completion] handling completion status 0
[ 0.006864] [00000e27] libusb: debug [disarm_timerfd] 
[ 0.006876] [00000e27] libusb: debug [usbi_handle_transfer_completion] transfer 0x55bf58852630 has callback 0x7f156b89cb30
[ 0.006889] [00000e27] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.006914] [00000e27] libusb: debug [libusb_free_transfer] transfer 0x55bf58852630
[ 0.006928] [00000e27] libusb: debug [libusb_alloc_transfer] transfer 0x55bf588530f0
[ 0.006939] [00000e27] libusb: debug [libusb_submit_transfer] transfer 0x55bf588530f0
[ 0.006951] [00000e27] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.006970] [00000e27] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.006978] [00000e27] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.016805] [00000e27] libusb: debug [handle_events] poll() returned 1
[ 0.016831] [00000e27] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=16
[ 0.016836] [00000e27] libusb: debug [handle_control_completion] handling completion status 0
[ 0.016846] [00000e27] libusb: debug [disarm_timerfd] 
[ 0.016862] [00000e27] libusb: debug [usbi_handle_transfer_completion] transfer 0x55bf588530f0 has callback 0x7f156b89cb30
[ 0.016867] [00000e27] libusb: debug [sync_transfer_cb] actual_length=16
[ 0.016874] [00000e27] libusb: debug [libusb_free_transfer] transfer 0x55bf588530f0
      Manufacturer: "SAMSUNG"
[ 0.016889] [00000e27] libusb: debug [libusb_alloc_transfer] transfer 0x55bf588523d0
[ 0.016892] [00000e27] libusb: debug [libusb_submit_transfer] transfer 0x55bf588523d0
[ 0.016894] [00000e27] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.016915] [00000e27] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.016918] [00000e27] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.026807] [00000e27] libusb: debug [handle_events] poll() returned 1
[ 0.026828] [00000e27] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.026833] [00000e27] libusb: debug [handle_control_completion] handling completion status 0
[ 0.026840] [00000e27] libusb: debug [disarm_timerfd] 
[ 0.026845] [00000e27] libusb: debug [usbi_handle_transfer_completion] transfer 0x55bf588523d0 has callback 0x7f156b89cb30
[ 0.026849] [00000e27] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.026857] [00000e27] libusb: debug [libusb_free_transfer] transfer 0x55bf588523d0
[ 0.026864] [00000e27] libusb: debug [libusb_alloc_transfer] transfer 0x55bf58855160
[ 0.026870] [00000e27] libusb: debug [libusb_submit_transfer] transfer 0x55bf58855160
[ 0.026876] [00000e27] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.026891] [00000e27] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.026895] [00000e27] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.036807] [00000e27] libusb: debug [handle_events] poll() returned 1
[... cut ...]
[ 3.270388] [00000e27] libusb: debug [sync_transfer_cb] actual_length=0
[ 3.270394] [00000e27] libusb: debug [libusb_free_transfer] transfer 0x55bf5885bc30
[ 3.270416] [00000e27] libusb: debug [libusb_alloc_transfer] transfer 0x55bf5885bc30
[ 3.270421] [00000e27] libusb: debug [libusb_submit_transfer] transfer 0x55bf5885bc30
[ 3.270428] [00000e27] libusb: debug [add_to_flying_list] arm timerfd for timeout in 3000ms (first in line)
[ 3.270436] [00000e27] libusb: debug [submit_bulk_transfer] need 1 urbs
```

lsimplify, 2018-02-17:
This was the output from the command when #347 was applied.
(It didn't work. The phone started booting during the flashing.)
```
$ ~/Heimdall/build/bin/heimdall flash --MODEM modem.bin --resume --no-reboot --usb-log-level debug --verbose --stdout-errors

Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.006568] [00000c7c] libusb: debug [libusb_get_device_list] 
[ 0.006605] [00000c7c] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.006624] [00000c7c] libusb: debug [libusb_get_device_descriptor] 
[ 0.006639] [00000c7c] libusb: debug [libusb_get_device_descriptor] 
[ 0.006649] [00000c7c] libusb: debug [libusb_get_device_descriptor] 
[ 0.006658] [00000c7c] libusb: debug [libusb_get_device_descriptor] 
[ 0.006687] [00000c7c] libusb: debug [libusb_get_device_descriptor] 
[ 0.006700] [00000c7c] libusb: debug [libusb_get_device_descriptor] 
[ 0.006710] [00000c7c] libusb: debug [libusb_get_device_descriptor] 
[ 0.006719] [00000c7c] libusb: debug [libusb_get_device_descriptor] 
[ 0.006731] [00000c7c] libusb: debug [libusb_open] open 2.10
[ 0.006782] [00000c7c] libusb: debug [usbi_add_pollfd] add fd 10 events 4
[ 0.006793] [00000c7c] libusb: debug [libusb_get_device_descriptor] 
[ 0.006808] [00000c7c] libusb: debug [libusb_alloc_transfer] transfer 0x55d8864b4280
[ 0.006820] [00000c7c] libusb: debug [libusb_submit_transfer] transfer 0x55d8864b4280
[ 0.006834] [00000c7c] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.006870] [00000c7c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.006881] [00000c7c] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.006894] [00000c7c] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.016902] [00000c7c] libusb: debug [handle_events] poll() returned 1
[ 0.016925] [00000c7c] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.016929] [00000c7c] libusb: debug [handle_control_completion] handling completion status 0
[ 0.016934] [00000c7c] libusb: debug [disarm_timerfd] 
[ 0.016949] [00000c7c] libusb: debug [usbi_handle_transfer_completion] transfer 0x55d8864b4280 has callback 0x7ff7c63bab30
[ 0.016953] [00000c7c] libusb: debug [sync_transfer_cb] actual_length=4
[ 0.016984] [00000c7c] libusb: debug [libusb_free_transfer] transfer 0x55d8864b4280
[ 0.016998] [00000c7c] libusb: debug [libusb_alloc_transfer] transfer 0x55d8864b3320
[ 0.017001] [00000c7c] libusb: debug [libusb_submit_transfer] transfer 0x55d8864b3320
[ 0.017010] [00000c7c] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.017037] [00000c7c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.017042] [00000c7c] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.017136] [00000c7c] libusb: debug [handle_events] poll() returned 1
[ 0.017158] [00000c7c] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=16
[ 0.017165] [00000c7c] libusb: debug [handle_control_completion] handling completion status 0
[ 0.017171] [00000c7c] libusb: debug [disarm_timerfd] 
[ 0.017177] [00000c7c] libusb: debug [usbi_handle_transfer_completion] transfer 0x55d8864b3320 has callback 0x7ff7c63bab30
[ 0.017181] [00000c7c] libusb: debug [sync_transfer_cb] actual_length=16
[ 0.017193] [00000c7c] libusb: debug [libusb_free_transfer] transfer 0x55d8864b3320
      Manufacturer: "SAMSUNG"
[ 0.017213] [00000c7c] libusb: debug [libusb_alloc_transfer] transfer 0x55d8864a9180
[ 0.017217] [00000c7c] libusb: debug [libusb_submit_transfer] transfer 0x55d8864a9180
[ 0.017221] [00000c7c] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.017237] [00000c7c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.017241] [00000c7c] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.027132] [00000c7c] libusb: debug [handle_events] poll() returned 1
[ 0.027157] [00000c7c] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
[ 0.027161] [00000c7c] libusb: debug [handle_control_completion] handling completion status 0
[ 0.027168] [00000c7c] libusb: debug [disarm_timerfd] 
[ 0.027176] [00000c7c] libusb: debug [usbi_handle_transfer_completion] transfer 0x55d8864a9180 has callback 0x7ff7c63bab30
[... 868 lines cut ...]
[20.793464] [00000c7c] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 1024
[20.793474] [00000c7c] libusb: debug [submit_bulk_transfer] first URB failed, easy peasy
[20.793485] [00000c7c] libusb: debug [disarm_timerfd] 
[20.793494] [00000c7c] libusb: debug [libusb_free_transfer] transfer 0x55d8864ab760
ERROR: libusb error -4 whilst sending bulk transfer.ERROR: libusb error -4 whilst sending bulk transfer. Retrying...
 Retrying...
[22.043693] [00000c7c] libusb: debug [libusb_alloc_transfer] transfer 0x55d8864ab760
[... cut ...]
ERROR: libusb error -4 whilst sending bulk transfer.ERROR: libusb error -4 whilst sending bulk transfer.

ERROR: Failed to send end session packet!
ERROR: Failed to send end session packet!
Releasing device interface...
[22.043925] [00000c7c] libusb: debug [libusb_release_interface] interface 1

[22.043971] [00000c7c] libusb: debug [libusb_close] 
[22.043997] [00000c7c] libusb: debug [usbi_remove_pollfd] remove fd 10
[22.044043] [00000c7c] libusb: debug [libusb_exit] 
[22.044055] [00000c7c] libusb: debug [libusb_exit] destroying default context
```

JonnyTech, 2018-02-17:
My experience is that the `--no-reboot` and `--resume` commands only work on a very few devices. Try flashing in one go, one command per reboot such as this example:

`heimdall flash --FACTORYFS factoryfs.rfs --CACHE cache.rfs --DBDATA dbdata.rfs --IBL+PBL boot.bin --SBL Sbl.bin --PARAM param.lfs --KERNEL zImage --MODEM modem.bin`

Flash one or more partitions then reboot back to download mode and run another command and repeat until done. One single `heimdall` command per session.

lsimplify, 2018-02-20:
@JonnyTech The first time I tried flashing the recovery partition I did not use `--no-reboot` or `--resume`. My command is in the original post: `heimdall flash --KERNEL <filename>` It softbricked _without_ those two options. This is the basis of my bug report.

Also, when sending one partition fails, I really don't see how sending those eight partitions will result in zero failures. 

I'm wondering if this issue is a bug of Heimdall, or an artifact of the fact that the Samsung software is actively preventing firmware modifications. 

I might just try to flash back rooted stock firmware with Odin and proceed using ROM Manager.

lsimplify, 2018-03-16:
One more piece of info for whomever it may concern:
Here is the archive where I got the modem file from: http://www.mediafire.com/file/g9wrj3gg86ws60c/Sprint_EA24_modem.zip And the **actual sha1 sum** of the **modem.bin** file in the zip archive is: b3a0cf1c0b687b1cddd3a4ad717832b8e486b465

mahdi-cll, 2021-07-06:
Same problem 
OS: Ubuntu 21.04
```
sudo ./heimdall  flash --RECOVERY '/TWRP-3.3.1-trhpltexx-20190610.img'
[sudo] password for m: 
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
PIT file download successful.

Uploading RECOVERY
0%
ERROR: Failed to send file part packet!
ERROR: RECOVERY upload failed!

Ending session...
ERROR: Failed to receive session end confirmation!
Releasing device interface...
```

amo13, 2021-07-08:
On a few older samsung devices the recovery partition was not RECOVERY, but Kernel2, so try
`sudo ./heimdall  flash --Kernel2 twrp.img`
Otherwise, was does the pit file information say? You should be able to see an overview of the present partitions and this way find out which one is the recovery partition.

mahdi-cll, 2021-07-09:
> On a few older samsung devices the recovery partition was not RECOVERY, but Kernel2, so try
> `sudo ./heimdall flash --Kernel2 twrp.img`
> Otherwise, was does the pit file information say? You should be [...]

ERROR: Partition "Kernel2" does not exist in the specified PIT.
PIT output attached here: https://github.com/Benjamin-Dobell/Heimdall/issues/488