Benjamin-Dobell/Heimdall #467: libusb -7 timeout with s9 (SDM845)

zlice, 2018-05-04:
# heimdall print-pit --verbose
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Samsung"
           Product: "SDM845"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 021B
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 01
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 81
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer.
ERROR: Failed to send handshake!ERROR: Failed to receive handshake response. Result: -7
ERROR: Protocol initialisation failed!

Releasing device interface...

--------------------

gentoo (don't know if there's some driver i'm missing?)
heimdall-git
libusb-1.0.22

really don't know what else would be useful or what's wrong. used multiple cables, 1 of them i transfered 200GB over so i know it's fine.

TimoSairiala, 2018-05-04:
Is the device in the Download mode? I am not sure how S9 behaves, but usually the product is something like "gadget serial" or something more generic

zlice, 2018-05-04:
it was, but i've got it in a box next to me and i'm about to call samsung to return it. thanks for reminding me =p

lorc, 2018-05-07:
Have the same problem there. Looks like Samsung changed something in protocol...

v0idst4r, 2018-06-05:
I'm having the same problem with my Note 8. I'm running out of options as I can't get any machine I own to flash it.

ma-hab, 2022-02-13:
Same problem on SM-G960F with libusb-1.0.24-2

Demindiro, 2022-12-28:
I have a SM-G960F. It took me a while before I could flash TWRP but I found that unplugging and replugging the device may help if it times out.

julianfoad, 2024-05-25:
Duplicate of #413? I just added some help there.