Benjamin-Dobell/Heimdall #470: ERROR: Failed to send request to end PIT file transfer!

graylion, 2018-05-18:
An old favourite

Galaxy Note 3

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...

OS is Mint 18.3

#apt list heimdall-flash
Listing... Done
heimdall-flash/xenial,now 1.4.1-1 i386 [installed]

the same happens on windows 7

graylion, 2018-05-18:
Correction: Initialising protocol...
ERROR: Failed to send data!Releasing device interface...

Oh and installing the driver fails on Win10do

graylion, 2018-05-18:
heimdall print-pit --verbose --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
      Manufacturer: "Sasmsung"
           Product: "MSM8960"

            length: 18
      device class: 2
               S/N: 0
           VID:PID: 04E8:685D
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 02.02.01
       endpoint[0].address: 82
           max packet size: 0010
          polling interval: 09

interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 0A.00.00
       endpoint[0].address: 81
           max packet size: 0200
          polling interval: 00
       endpoint[1].address: 01
           max packet size: 0200
          polling interval: 00
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer. Retrying...
ERROR: libusb error -7 whilst sending bulk transfer.
ERROR: Failed to send handshake!ERROR: Expected: "LOKE"
Received: ""
ERROR: Unexpected handshake response!
ERROR: Protocol initialisation failed!

Releasing device interface...

gizmo15, 2018-05-24:
I have the same problem. I try different usb port no change.
same with --no-reboot

frodeseverin, 2018-05-30:
Similar problem for Samsung Galaxy S5 LTE (SM-G900F).

Logs from Debian Linux attatched.
Heimdall 1.4.1 installed from Debian Stretch repository.

;)Frode
[heimdall.log](https://github.com/Benjamin-Dobell/Heimdall/files/2055456/heimdall.log)
[heimdall_verbose.log](https://github.com/Benjamin-Dobell/Heimdall/files/2055457/heimdall_verbose.log)
[heimdall_verbose_debug.log](https://github.com/Benjamin-Dobell/Heimdall/files/2055458/heimdall_verbose_debug.log)

paskalito, 2021-06-09:
@frodeseverin did you ever find a soloution for this?

amo13, 2021-06-09:
try to avoid the print-pit-stuff and just issue the flashing command to install twrp (or whatever). I have seen a few devices that would accept only a single heimdall command after boot to download mode. That effectively means that you have to reboot to download mode after every command if there is another command to be executed. This is why I try to issue only the single command needed, which is flash.

paskalito, 2021-06-09:
Thank you for your quick answer! this is what i get:

````
heimdall flash --RECOVERY '/home/xxx/Downloads/lineage-18.1-20210606-recovery-klte.img' --no-reboot
Heimdall v1.4.1

Copyright (c) 2010-2014 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
Protocol initialisation successful.

Beginning session...

Some devices may take up to 2 minutes to respond.
Please be patient!

Session begun.

Downloading device's PIT file...
ERROR: Failed to send request to end PIT file transfer!
ERROR: Failed to download PIT file!
Ending session...
ERROR: Failed to send end session packet!
Releasing device interface...
````

amo13, 2021-06-09:
Did you try with sudo?
Maybe also try to compile a version 1.4.2 with fixes from the pull requests. Or use the binary I compiled a while back from [here](https://heimdall.free-droid.com).

paskalito, 2021-06-10:
Heyho, thanks a lot with the 1.4.2 binary you linked i am able to PIT transfer - and also able to flash. !
Sadly it doesn't boot the recovery but i guess this is another problem..
````
Could not do normal boot
````

amo13, 2021-06-11:
glad the flashing worked
maybe you've got the wrong twrp file

paskalito, 2021-06-12:
Exactly this is what happened - i figered out i've gotten a Galaxy s5 **plus**  - which has another twrp (found on its website) and also no Lineage Support - but i found an image from /e - so i have a working phone at least in the end. (Y)

doctordoggo, 2021-06-15:
@amo13 Sorry for another dumb question but how could I get these heimdall 1.4.2 binaries in operation or how can I apply them? Have the same issue as mentioned above and heimdall 1.4.1 installed...
Thanks in advance. :)

amo13, 2021-06-16:
Unpack the archive and, in a terminal, call the binary directly with something like `./heimdall flash ...`. You might want to `cd` into the folder of the binary first and have the twrp image in the same folder.
I hope this helps

Nikod1, 2022-02-11:
I'm having the same issue, I have heimdall 1.4.2 on Linux and I also tried the one from the binary @amo13 suggested also with sudo and still same thing. Can anyone help me? I have been trying for three weeks now. I have SM-920C Galaxy Note 5, right now the device is stuck at two screens: 1.[ download mode page](https://imgur.com/a/cPNS0l7) 2. [Landing screen ](https://imgur.com/a/RuqaUXW),when I try to do soft reset the icon of download mode stays on the screen with text beneath it not above and device never go off. Before this happens, this is the situation and I tried [those ](https://android.stackexchange.com/questions/244468/galaxy-note-5-suddenly-stopped-booting-up-and-in-recovery-mode-has-dm-verity-and)and nothing is working, given that I never touched the device before all this happen.

amo13, 2022-02-11:
Are you on Linux? If so, can you also try [this build](https://github.com/Benjamin-Dobell/Heimdall/issues/503#issuecomment-998289267) and report here if it works?

Nikod1, 2022-02-22:
Hi @amo13 so sorry for the late reply, I tried the version you proposed, unfortunately I got the same error. Could you please tell me what are other options?

amo13, 2022-02-22:
Then sorry, I don't have any other idea for you, except trying a different cable or computer.

Nikod1, 2022-02-22:
I will try these as well, but do you know if Odin can fix the device without loosing data?