Benjamin-Dobell/Heimdall #474: Bad system.img flash on A520F

loadncode, 2018-06-04:
As the title says I have an A520F (Galaxy A5 2017). I am using Debian Testing's version of Heimdall (1.4.1-2+b1) but have also tried version 1.4.2 with the same results I'm about to describe. Basically Heimdall seems to be able to flash all partitions fine except system.img. Flashing system.img does appear to work (Heimdall reports success) but the device doesn't boot. Upon further investigation (via flashing TWRP and accessing an adb shell) I noticed that after flashing the output of "du -h /system" is only 395.8MB. Upon restoring a TWRP backup of /system the result changes to 4.3GB. So in other words either system.img only gets partially flashed or it gets corrupted somehow during the flash. Flashing the same firmware in Odin results in the device booting without issue.

The command I used for flashing is this:
"heimdall flash --no-reboot --BOOTLOADER sboot.bin --PARAM param.bin --CM cm.bin --BOOT boot.img --RECOVERY  recovery.img --SYSTEM system.img --RADIO modem.bin --CACHE cache.img --HIDDEN hidden.img"
I've tried changing the order the files get flashed in, tried doing it with multiple commands (which causes other errors) all without success.

I've attached a [debug log](https://github.com/Benjamin-Dobell/Heimdall/files/2069601/systemimgflash.zip) of a system.img flash. Unfortunately it's massive and I'm not sure it contains anything useful. Let me know if there's anything else I can provide to help. By the way system.img is far bigger than any of the other firmware files and is 4,669,945,200 bytes in size. None of the other files are bigger than about 40MB.

Blefish, 2018-06-05:
Is the system.img partition to be flashed larger than 3.5GB?

loadncode, 2018-06-05:
Yes system.img is 4.3GB (4,669,945,200 bytes to be exact).

loadncode, 2018-06-05:
It's official firmware, version A520FXXU4BRC4, available here:
https://www.sammobile.com/firmwares/galaxy-a5/SM-A520F/BTU/download/A520FXXU4BRC4/215512

Blefish, 2018-06-05:
There is a change regarding that, and we are successfully using it, however again, its based on older Heimdall version. I can probably push it as well.

Benjamin-Dobell, 2018-06-05:
@Blefish Sure, if it's not going to be a clean PR and you don't have the time to make one, then feel free to push the changes to a fork and I'll take a look at what's different.

Blefish, 2018-06-05:
https://github.com/Blefish/Heimdall/commit/d8a14b556d9a16d1dc4501c1a1db4bc473efd1c1

I am not sure if -D_FILE_OFFSET_BITS=64 is needed with the correct compilation options.

loadncode, 2018-06-05:
If it's any help I can confirm that Blefish's fork resolves this issue for me, system.img now flashes properly. There is one minor aesthetic issue that still happens which is that Heimdall's upload progress indicator returns to zero when it's almost at 100% then keeps going, so I end up with it finiahing on 8%. Not that I'm bothered by it but I presume this is also caused by the large file size.

Blefish, 2018-06-06:
I can fixup the commit as needed, but do you have any idea if the -FILE_OFFSET_BITS definition is really needed? IIRC this was needed on older Heimdall version, but the new versions all use use_largefile.

bzar, 2021-06-29:
@Blefish your fork saved me from a borked S10. I've been banging my head against failed flashing of system.img for two days.

One note, your fork fixes the `heimdall` tool but `heimdall-frontend` still has the bug.

phlash, 2022-11-22:
@Blefish thank you for this, I was struggling with current Debian stable version (1.4.2-dfsg-1) which does not include your patches, same problem as other folks: `system.img` from stock ROM fails to flash correctly. Good news is that your patches are in Debian's source repo (as 1.4.2-dfsg-2) but that's not in any release stream yet.