Benjamin-Dobell/Heimdall #479: is it possible to flash the tar file without extracting

Xellerate005, 2021-03-15:
just want to know if heimdall itself is capable of extracting it then flashing it, because magisk root requires to patch entire ap file

shaun-ba, 2021-08-24:
Also stuck on this point, how to flash Magisk using Heimdall, if anyone has a guide :)

amo13, 2021-08-24:
This is a misconception about what heimdall does. Heimdall only flashes a partition image to a partition with a given name. I have flashed a lot of samsung devices with heimdall and only ever needed to flash twrp to recovery.
Everything else is done from within twrp. If you want to install magisk, do it with twrp, not with heimdall.

shaun-ba, 2021-08-25:
@amo13 Ok thank you for that explanation.

I am following this guide, whereby you can see the Odin step: https://topjohnwu.github.io/Magisk/install.html#samsung-system-as-root

I cannot even download .pit from Heimdall at the moment, so would much prefer another method using TWRP is that's possible?

shaun-ba, 2021-08-25:
Currently I just get this:

Initialising protocol...
ERROR: Failed to send handshake!ERROR: Protocol initialisation failed!

amo13, 2021-08-25:
OK, don't know... Sorry.

tuxayo, 2025-04-26:
Heimdall seem more low level.

It seems this tool might be a more direct alternative to Odin: https://github.com/Samsung-Loki/Thor