Benjamin-Dobell/Heimdall #482: LineageOS developers have compiled own version

frodeseverin, 2021-04-10:
Perhaps it would be worth a try to incorporate their work back in your repo.

See [here](https://wiki.lineageos.org/devices/a5y17lte/install)

Regards
;)Frode

blaueente, 2021-04-10:
I can only find binaries, did they post any source?

eliotb, 2021-05-02:
Apparently only one additional commit relating to OSX : https://gitlab.e.foundation/e/tools/heimdall/-/commit/af5c83a45c8580c3c264db008cd0582c0cf1b757

frodeseverin, 2021-05-05:
I am quite unsure of the details here.
I only saw the claim on LineageOS site that they have made modified version.

;)Frode

Dyras, 2021-05-14:
The LineageOS version indeed works flawlessly, while vanilla Heimdall doesn't really work at all for me.

frodeseverin, 2021-05-14:
I, too had a flawless experience with LinageOS version o Debian GNU/Linux.

I did not try your original version.

;) Frode

Erudition, 2022-10-12:
Need the source code for the modified version, then we can update the builds

Yaksinikos, 2023-08-14:
actually the lineage version does not work for linux as well