Benjamin-Dobell/Heimdall #483: Possible to install on OS X Catalina?

antarr, 2021-04-26:
I'm trying to install on OSX Catalina but the installer says my system is not supported. Is there a way to get around this?

BeastOfBodmin, 2021-05-22:
If it did install, what does it install that requires a reboot?

The 1.4.0 installer seemed to install on Catalina.

I rebooted.

But I can't find anything that resembles the Heimdall app with this

find / | grep -i heimdall

If it did install, where do I find it?

amo13, 2021-05-22:
I'd recommend to simply get a binary compiled with the necessary fixes and to use it directly. 
[Here](https://heimdall.free-droid.com/) is the one I compiled a while back. It should work on OSX Catalina.
[Here](https://wiki.lineageos.org/devices/a5y17lte/install#preparing-for-installation) you can find the heimdall stuff compiled by the lineage team.
Also, I encourage you to check out my custom rom [installer application](https://free-droid.com/).

mrdegger, 2021-05-23:
> I'd recommend to simply get a binary compiled with the necessary fixes and to use it directly.
> [Here](https://heimdall.free-droid.com/) is the one I compiled a while back. It should work on OSX Catalina.

First link worked for me on OSX El Capitan (_heimdall-mac-old.zip_). Thanks a lot!

slhck, 2021-06-13:
Specific error is:

> The package is trying to install content to the system volume. Contact the software manufacturer for assistance.

It tries to install to the System volume, which is not supported.

amo13, 2021-06-13:
What is trying to install what to which system's volume? What did you try and what command did you issue?

slhck, 2021-06-13:
I am referring to installing Heimdall Suite 1.4.0.pkg on macOS Catalina (as the OP wrote). As you can see in https://github.com/Benjamin-Dobell/Heimdall/blob/master/OSX/install-kext.sh — assuming that is what is being used for the `.pkg` code — it is attempting to copy a kernel extension to `/System`, which is not possible starting with 10.15.

amo13, 2021-06-13:
I see. Well since heimdall pretty much is an abandoned project and the file you are referring to is from 2015, I don't think there is much hope in getting an installer to work on current systems.
I guess we are better off just using compiled binaries as long as they are still working...

slhck, 2021-06-13:
Oh, alright. I didn't see that it was abandoned. Perhaps the authors could at least explain the macOS limitations in the README.

I see the binrary you linked to just runs. Thank you.