Benjamin-Dobell/Heimdall #484: For anyone having problems flashing an image on Galaxy Tab E

The-EDev, 2021-05-14:
I was trying to flash TWRP on my SM-T561, first I was getting libusb error -7, which I fixed by running only 1 command to flash my image (only 1st command runs because Heimdall doesn't reset the USB connection).

The second issue was that the image wasn't actually being flashed, this was because I was opening download mode using the buttons. using `adb reboot download` allowed me to flash the new recovery.

Hope this helps anyone trying to restore their old tablet.

amo13, 2021-05-22:
Yes, this is an issue I have encountered on numerous devices. Often times, you can execute only one single heimdall command and then you have to reboot to download mode if you need to do another one.

teh404guy, 2021-06-11:
yup happened to me while flashing my samsung s10+ with lineage
you have to either flash everything at one
or flash and reboot one by one