Benjamin-Dobell/Heimdall #487: not installable on Ubuntu 21.04

mahdi-cll, 2021-07-06:
```$ sudo dpkg -i '/tmp/mozilla_m0/ubuntu13.04-heimdall-frontend_1.4.0-0_amd64.deb'
(Reading database ... 251885 files and directories currently installed.)
Preparing to unpack .../ubuntu13.04-heimdall-frontend_1.4.0-0_amd64.deb ...
Unpacking heimdall-frontend (1.4.0-0) over (1.4.0-0) ...
dpkg: dependency problems prevent configuration of heimdall-frontend:
 heimdall-frontend depends on libqtcore4 (>= 4:4.7.0~beta1); however:
  Package libqtcore4 is not installed.
 heimdall-frontend depends on libqtgui4 (>= 4:4.5.3); however:
  Package libqtgui4 is not installed.
 heimdall-frontend depends on libqt4-xml (>= 4.6); however:
  Package libqt4-xml is not installed.

dpkg: error processing package heimdall-frontend (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 heimdall-frontend
```

amo13, 2021-07-06:
This project is long abandoned. Just use the binary I compiled a while back from [here](https://heimdall.free-droid.com). Call it directly in a terminal, no gui.

mahdi-cll, 2021-07-08:
Thanks. I used your binary now `heimdall print-pit` works properly but [this problem](https://github.com/Benjamin-Dobell/Heimdall/issues/454#issuecomment-875064850) still in place

ProjectT110, 2021-10-16:
`$ sudo apt install heimdall-flash`

bennylangston, 2022-11-05:
sudo dpkg -i net.downloadhelper.coapp-1.6.3-1_amd64.deb
this is the way