Benjamin-Dobell/Heimdall #488: ERROR: Failed to send file part packet!

mahdi-cll, 2021-07-09:
It stuck a little at 0% and then `ERROR: Failed to send file part packet!`
```
$ sudo ./heimdall  flash --RECOVERY '/TWRP-3.3.1-trhpltexx-20190610.img'
[sudo] password for m: 
Heimdall v1.4.2

.
.
.

Uploading RECOVERY
0%
ERROR: Failed to send file part packet!
[pit.txt](https://github.com/Benjamin-Dobell/Heimdall/files/6788208/pit.txt)

ERROR: RECOVERY upload failed!

Ending session...
ERROR: Failed to receive session end confirmation!
Releasing device interface...
```
img download: https://github.com/ananjaser1211/TWRP_device_samsung_treltexx/releases

amo13, 2021-07-12:
You can see in the pit file that RECOVERY is definitely the right partition name. I guess if it doesn't work, then it doesn't work. Sorry about that.
Since there is no custom rom out there for your tablet, what did you plan to do with twrp anyway? And are you even sure this twrp image really is for your tablet? On the release page you linked, it says it is for the "N910U", but you appear to have the "SPH-P100".

mahdi-cll, 2021-07-15:
I was thinking to flash this ROM: https://forum.xda-developers.com/t/exynos-unofficial-lineageos-17-1-06-30-2021.3937956/
I have a N910U. If you say SPH-P100 because of title of previous issue, it wasn't opened by me

amo13, 2021-07-15:
Don't forget to try a different usb port and especially a different usb cable! Don't underestimate bad usb cables, I had the problem a few times already and marked the cable that I found to be working...

mahdi-cll, 2021-07-21:
I had to use a usb cable shorter than 1.5 meter

closed 2021-07-21 by mahdi-cll