Benjamin-Dobell/Heimdall #490: v1.4 incompatible with MacOs BigSur

clamattu, 2021-08-25:
I've just tried to install it but it is considered too old to be installed by the system (v.11.5.2)

amo13, 2021-08-25:
Did you try using the binary directly?

clamattu, 2021-08-25:
Yes from [Downloads](https://glassechidna.com.au/heimdall/#downloads)

amo13, 2021-08-25:
Then what do you mean by installing? Try putting the binary heimdall file (no .app or .pkg or similar) into a folder and calling it from a terminal (e.g. `./heimdall print-pit`).
Also, the version available from the downloads link you mentioned is somewhat outdated and known to cause trouble on newer macos versions and newer devices (unfortunately, this project has been abandoned for a while). Try the binary that I compiled with a bug fix from a pull request, available [here](https://heimdall.free-droid.com/).

clamattu, 2021-08-25:
Ok, i have downladed, rebooted and tried it.
This is the result:
`heimdall print-pit
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
ERROR: Failed to detect compatible download-mode device.`

amo13, 2021-08-25:
The last line states that no device in download mode was detected. Have you rebootet your device to download mode?

clamattu, 2021-08-25:
Yes, it is with 
fap lock:off
Screen "Downloading... do not turn off target"

I have already changed few port of my MacMini

amo13, 2021-09-05:
@clamattu did you resolve the issue in the meantime?

Grimler91, 2021-09-05:
This fork: https://gitlab.e.foundation/e/tools/heimdall/-/tree/new_master supposedly works with more recent mac versions, AFAIK that is the one that the LineageOS project built their mac binaries from

amo13, 2021-09-05:
I'd rather have expected the other way around since /e/ is entirely relying on lineageos. How do you know? Did I miss a hint somewhere on lineage's site?

Anyway, the only additional commit to heimdall from the/e/ team seems to be only a fix for compilation issues on recent macos. It is pretty much the same as suggested in #468.

Grimler91, 2021-09-06:
> How do you know? Did I miss a hint somewhere on lineage's site?

Hm, no, I might be completely wrong, I think I based it solely on the discussion in [one of the other threads](https://github.com/Benjamin-Dobell/Heimdall/issues/482#issuecomment-830796009).

> Anyway, the only additional commit to heimdall from the/e/ team seems to be only a fix for compilation issues on recent macos. It is pretty much the same as suggested in #468.

The /e/ changes look a bit cleaner than the ones in #468 (`if(APPLE)` seems better than `if(NOT MINGW)` for example), but you are right, pretty equivalent.

For reference, I have merged some of the PRs and made some minor changes over at [sourcehut](https://git.sr.ht/~grimler/Heimdall), but nothing to fix macos compilation (I don't own a mac)

amo13, 2021-09-07:
It is way cleaner, you are right. And it also solves the compilation issues I had in [my fork](https://github.com/amo13/Heimdall). Thank you for the hint, @Grimler91 !

danielboston38, 2023-05-15:
sorry for opening a closed issue but how do I get the binary? the option of opening a packages contents isn't in the menu anymore and I don't know how to do it.