Benjamin-Dobell/Heimdall #491: Samsung SM-G998B -

shaun-ba, 2021-08-25:
So I cannot even print-pit currently on this device, is it even supported?

Currently output is:

`sudo heimdall print-pit
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
Claiming interface...
Setting up interface...

Initialising protocol...
ERROR: Failed to send handshake!ERROR: Protocol initialisation failed!

Releasing device interface...
`

amo13, 2021-09-05:
I merged a few overdue pull requests from this repo the last day. Can you please try with a freshly compiled version from [here](https://github.com/amo13/Heimdall/suites/3594666517/artifacts/86623046)?

amo13, 2021-09-05:
Sorry, that link was only for macOS... [here](https://github.com/amo13/Heimdall/suites/3594457358/artifacts/86614346) is for windows and [this](https://github.com/amo13/Heimdall/suites/3594457358/artifacts/86614345) is for linux.

testman42, 2021-09-09:
@amo13 thank you very much for this
I have been banging my head against the wall for quite a few days now, tried many heimdall builds, and the one you linked above seems to be the first one that can communicate with my SM-T500

amo13, 2021-09-09:
Amazing! Glad to read that, because I merged the pull requests and made the build all totally blindly, since I have no current device to actually test it with.
Thank you very much for reporting, @testman42 .

rdeforest, 2021-10-30:
> I merged a few overdue pull requests from this repo the other day. Can you please try with a freshly compiled version from [here](https://github.com/amo13/Heimdall/suites/3594666517/artifacts/86623046)?

This worked for me with a Samsung S9 and MacOS 11.8. Thank you for your continued work on this.

dylhack, 2021-11-03:
This build got passed the handshake fail, but I ran into `Failed to receive session end confirmation` and for my device it displays `SECURE CHECK FAIL : recovery`

dylhack, 2021-11-03:
It started working all of the sudden when I turned on `--usb-log-level debug`, probably coincidence but I didn't think this would be quantum physics.

Erudition, 2022-10-20:
Same problem, even with a manually compiled build of @amo13 's fork with patches...
On latest Ubuntu LTS

pcolamar, 2022-10-25:
Same problem with LinuxMint 21-Cinnamon. I guess because it is based on the Ubuntu LTS 22

julianfoad, 2024-05-25:
Duplicate of #413? I just added some help there.