Benjamin-Dobell/Heimdall #494: Heimdall can flash files bigger than partitions

GNUtoo, 2021-09-21:
Apparently I can flash a file that is bigger than the RECOVERY partition:

Here I've a recovery.img that is a bit over 8MiB (8388608 bytes)
> $ du --bytes recovery.img
> 9691136	recovery.img`

And here's print-pit on Galaxy SIII (GT-I9300 variant) I managed to flash this image:
> $ heimdall print-pit
> Heimdall v1.4.2
> 
> Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
> http://www.glassechidna.com.au/
> 
> This software is provided free of charge. Copying and redistribution is
> encouraged.
> 
> If you appreciate this software and you would like to support future
> development please consider donating:
> http://www.glassechidna.com.au/donate/
> 
> Initialising connection...
> Detecting device...
> Claiming interface...
> Setting up interface...
> 
> Initialising protocol...
> Protocol initialisation successful.
> 
> Beginning session...
> 
> Some devices may take up to 2 minutes to respond.
> Please be patient!
> 
> Session begun.
> 
> Downloading device's PIT file...
> PIT file download successful.
> 
> Entry Count: 16
> Unknown 1: 1598902083
> Unknown 2: 844251476
> Unknown 3: 30797
> Unknown 4: 0
> Unknown 5: 0
> Unknown 6: 0
> Unknown 7: 0
> Unknown 8: 0
> 
> 
> --- Entry #0 ---
> Binary Type: 0 (AP)
> Device Type: 2 (MMC)
> Identifier: 80
> Attributes: 2 (STL Read-Only)
> Update Attributes: 1 (FOTA)
> Partition Block Size/Offset: 0
> Partition Block Count: 1734
> File Offset (Obsolete): 0
> File Size (Obsolete): 0
> Partition Name: BOOTLOADER
> Flash Filename: sboot.bin
> FOTA Filename: 
> 
> 
> --- Entry #1 ---
> Binary Type: 0 (AP)
> Device Type: 2 (MMC)
> Identifier: 81
> Attributes: 5 (Read/Write)
> Update Attributes: 1 (FOTA)
[... 324 lines cut ...]
> Setting up interface...
> 
> Initialising protocol...
> Protocol initialisation successful.
> 
> Beginning session...
> 
> Some devices may take up to 2 minutes to respond.
> Please be patient!
> 
> Session begun.
> 
> Downloading device's PIT file...
> PIT file download successful.
> 
> Uploading RECOVERY
> 100%
> RECOVERY upload successful
> 
> Ending session...
> Rebooting device...
> Releasing device interface...
So we may (or may not) need to add a new --with-pit or similar argument in --help along with warnings depending on what's already there in heimdall.

Denis.

Newbytee, 2021-09-22:
> I didn't check yet if it wrote part of the image in the RADIO partition or not.

While I have not checked this either, it used to be in postmarketOS that flashing the boot image (with main OS kernel) to i9300 would result in the recovery (on a different partition) becoming unbootable. This was fixed by making sure the boot image was smaller than the boot partition.

Considering this, I think it's likely that Heimdall just continues writing onto the next partition.

Grimler91, 2021-09-28:
On devices with newer bootloaders an error (without any information about what the issue is) is returned instead. On at least galaxy tab s from 2013/2014 this happens.

I agree that an error from heimdall would be better in any case though, should be possible to implement from the information in the PIT

TheAirBlow, 2021-11-30:
It can be implemented very easily.
Just multiply block size by block count, and checks if it's bigger than expected.

Grimler91, 2021-12-10:
Here's a WIP: [12d80ef196de](https://git.sr.ht/~grimler/Heimdall/commit/12d80ef196def8fc56bb44fae4b32df6852314b1) (from this branch: https://git.sr.ht/~grimler/Heimdall/log/size_check)

There are two issues that I have not figured out yet:

* The size of a block differs between devices, older have 512 and newer 4096. I hardcoded 512 for now, but we need to get this from the pit, or device, somehow instead.

* I tried flashing stock android to three devices, and it failed on one due to the BOOTLOADER (sboot) partition being too small for the file. Maybe this means that the BOOTLODER partition works a bit differently, and that we should not check the size in this case.

TimoSairiala, 2021-12-10:
I have to say I am not quite sure about this, but can you check block size from the image file instead of the phone?
```
$ fdisk -lu boot.img
Disk boot.img: 96 MiB, 100663296 bytes, 196608 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
```

Grimler91, 2021-12-10:
@TimoSairiala fdisk, and file, and other tools seem to report/use the same block size no matter which image I look at unfortunately

Looking closer at the pit I think the information is there in the header though, in a part that is not parsed at all right now. For i9300 the PIT starts with:

```
00000000: 7698 3412 1000 0000 434f 4d5f 5441 5232  v.4.....COM_TAR2
00000010: 4d78 0000 0000 0000 0000 0000 0000 0000  Mx..............
00000020: 0200 0000 5000 0000 0200 0000 0100 0000  ....P...........
00000030: 0000 0000 c606 0000 0000 0000 0000 0000  ................
```
and my guess is that 0200 at 00000020 is the block size (in multipliers of 256). For herolte we instead have something like:
```
00000000: 7698 3412 1800 0000 434f 4d5f 5441 5232  v.4.....COM_TAR2
00000010: 4c53 4937 3432 3000 0400 0000 0000 0000  LSI7420.........
00000020: 0800 0000 5000 0000 0200 0000 0100 0000  ....P...........
00000030: 0000 0000 0004 0000 0100 0000 0000 0000  ................
```
0800 instead of 0200, i.e. 512*4=4096. Will have to check with a couple of more devices to verify if this hypothesis seem to hold.

@TheAirBlow you seem to have documented the pit format for https://samsung-loki.github.io/samsung-docs/docs/PIT/, do you have any guess what the other numbers in second half of row 2, and on row 3 and 4 might be?

TheAirBlow, 2021-12-10:
> @TimoSairiala fdisk, and file, and other tools seem to report/use the same block size no matter which image I look at unfortunately
>
> Looking closer at the pit I think the information is there in the [...]

Hello, I am more focused to get Hreidmar GUI working. I'll look at the PIT reader in the `aboot.mbn` itself when I'll be able to.

Grimler91, 2021-12-10:
I confused the offsets, the first pitentry starts at the end of row two, so 0800 0000 / 0200 0000 is not part of the header.

I think the field might still say something about the blocksize though, for i9300 we have 0200 0000 and
```
--- Entry #0 ---
Binary Type: 0 (AP)
Device Type: 2 (MMC)
Identifier: 80
Attributes: 2 (STL Read-Only)
Update Attributes: 1 (FOTA)
Partition Block Size/Offset: 0
Partition Block Count: 1734
Partition Name: BOOTLOADER
Flash Filename: sboot.bin
```
while for herolte with 0800 0000 we have
```
--- Entry #0 ---
Binary Type: 0 (AP)
Device Type: 8 (Unknown)
Identifier: 80
Attributes: 2 (STL Read-Only)
Update Attributes: 1 (FOTA)
Partition Block Size/Offset: 0
Partition Block Count: 1024
File Offset: 1
File Size: 0
Partition Name: BOOTLOADER
Flash Filename: sboot.bin
```

`Device Type: 8 (Unknown)` could be mmc, but with blocksize 4096

TheAirBlow, 2021-12-10:
> I confused the offsets, the first pitentry starts at the end of row two, so 0800 0000 / 0200 0000 is not part of the header.
>
> I think the field might still say something about the blocksize though, [...]

Please create an issue [here](https://github.com/Samsung-Loki/Hreidmar)

Grimler91, 2021-12-16:
I've [tagged a release](https://git.sr.ht/~grimler/Heimdall/refs) in the repo I am maintaining which contains a commit to compare the file and partition sizes before flashing. 

The check is (for now) only based on the actual file size. For sparse images (i.e. {system,cache,userdata,hidden}.img and maybe others) the file size doesn't equal the expanded size, so we still risk flashing something that is too large to these partitions. A better check would instead get the image size from the [header of the sparse image](https://android.googlesource.com/platform/system/core/+/7b444f08c17ed1b82ea1a1560e109c0a173e700f/libsparse/sparse_format.h#32).

As discussed above, I've also changed how the [PIT header is printed](https://git.sr.ht/~grimler/Heimdall/commit/e51c9119f1ff89a56efd4bb673e1870cef59f7be). We seem to have a string "COM_TAR2", followed by a string related to the CPU or bootloader, and then a number that might be a version of some sort (0000, 0003, 0004 and 0005 have been observed, if you have a device with another version please let me know).

Some of the fixes from the various PRs here on github and gitlab are also merged, in particular:

* The libusb reset patch needed on ubuntu https://github.com/Benjamin-Dobell/Heimdall/pull/478
* Support for flashing images larger than 3.5 GB https://github.com/Benjamin-Dobell/Heimdall/pull/459
* Avoid libusb_set_interface_alt_setting when it anyways doesn't do anything https://github.com/Benjamin-Dobell/Heimdall/issues/497, fixed in [2ecc98020c](https://git.sr.ht/~grimler/Heimdall/commit/2ecc98020c607631d1371c8596d20cce84586047)
* Some compilation issues on [freeBSD](https://git.sr.ht/~grimler/Heimdall/commit/64bd479db680ff7c00967f15a1c8ce8225e65efb), [Android](https://git.sr.ht/~grimler/Heimdall/commit/719edaff9ebe6921cde28aab6a1a5088fe57b5bf) and [macOS](https://git.sr.ht/~grimler/Heimdall/commit/6ede1f1f337f72c5b8b51bd16e351f412c351342) have been fixed.

Pre-compiled binaries can be downloaded for [ubuntu 20.04](https://patchouli.sr.ht/builds.sr.ht/artifacts/~grimler/650736/bf10d326fa210f7b/heimdall), [archlinux](https://patchouli.sr.ht/builds.sr.ht/artifacts/~grimler/650735/07ba1155b5754e4a/heimdall) and [alpine linux](https://patchouli.sr.ht/builds.sr.ht/artifacts/~grimler/650734/8a998383f5f18428/heimdall). The source can be found and browsed at https://git.sr.ht/~grimler/Heimdall.

TheAirBlow, 2021-12-16:
> I've [tagged a release](https://git.sr.ht/~grimler/Heimdall/refs) in the repo I am maintaining which contains a commit to compare the file and partition sizes before flashing.
>
> The check is (for [...]

Thanks for researching the PIT innerworkings! Gotta check it with some random devices firmware's PIT to confirm this. Looking for adding this into Hreidmar, and displaying PIT in GUI manner.
I will look more into the header. Never looked into the PIT much, more focused on other things.

Grimler91, 2021-12-16:
@TheAirBlow Thanks, please let me know if you come in contact with any devices where the last bytes of the header is something other than 0000, 0003, 0004 or 0005 (seems weird that 0001 and 0002 are skipped, if it is a version number), and I'll see if I can write something up for your wiki pages!

TheAirBlow, 2021-12-17:
> @TheAirBlow Thanks, please let me know if you come in contact with any devices where the last bytes of the header is something other than 0000, 0003, 0004 or 0005 (seems weird that 0001 and 0002 are skipped, if it is a version number), and I'll see if I can write something up for your wiki pages!

You can download random firmwares by using [this](https://github.com/samsung-loki/syndical), unzip, and untar the CSC. Here is the PIT that you need. Use Download&Decrypt mode for this.

Also, Samsung Galaxy A20s (SER) seems to have different numbers in the header.

TheAirBlow, 2021-12-17:
Also, I think we need to look into the Odin mode PIT flashing (located in `aboot,mbn`) and check how it read the PIT

steve8x8, 2022-05-25:
"Let me reboot into Windows real quick." - that looks like "famous last words" :'(

TheAirBlow, 2022-05-26:
> "Let me reboot into Windows real quick." - that looks like "famous last words" :'(

By using the ~~completely legal sources~~ I updated the docs. Also, currently I have no motivation to develop Hreidmar, sadly.

TheAirBlow, 2022-05-26:
But, ~~Thor~~ and the ~~Bootloader leak~~ said different things, so idk which one to actually reference. A lot of confusion.

TheAirBlow, 2022-05-26:
Also, no people have "sacrifices their phone for science", and my phone is unflasheable, I cannot even test anything.

TheAirBlow, 2022-05-27:
A new discovery!
```cpp
    //Descritption for old Pit version
    m_oldPit.m_mapBinaryType[0] = "AP";
    m_oldPit.m_mapBinaryType[1] = "CP";

    m_oldPit.m_mapDeviceType[0] = "ONENAND";
    m_oldPit.m_mapDeviceType[1] = "NAND";
    m_oldPit.m_mapDeviceType[2] = "MOVINAND";

    m_oldPit.m_mapAttribute[0] = "RO";
    m_oldPit.m_mapAttribute[1] = "RW";
    m_oldPit.m_mapAttribute[2] = "STL";

    m_oldPit.m_mapUpdateAttribute[0] = "NONE";
    m_oldPit.m_mapUpdateAttribute[1] = "FOTA";
    m_oldPit.m_mapUpdateAttribute[2] = "SECURE";
    m_oldPit.m_mapUpdateAttribute[3] = "FOTA+SECURE";

    m_oldPit.m_horizontalHeader << QString("Partition Name") << QString("Binary Type") << QString("Device Type") << QString("Id")
            << QString("Attribute") << QString("Update Attribute") << QString("Unit Size") << QString("Units Count")
            << QString("File Offset") << QString("File Size")  << QString("File Name") << QString("Delta Name");

    //Descritption for new Pit version
    m_newPit.m_mapBinaryType[0] = "AP";
    m_newPit.m_mapBinaryType[1] = "CP";

    m_newPit.m_mapDeviceType[0] = "ONENAND";
    m_newPit.m_mapDeviceType[1] = "NAND";
    m_newPit.m_mapDeviceType[2] = "EMMC";
    m_newPit.m_mapDeviceType[3] = "SPI";
    m_newPit.m_mapDeviceType[4] = "IDE";
    m_newPit.m_mapDeviceType[5] = "NAND X16";

    m_newPit.m_mapAttribute[0] = "NONE";
    m_newPit.m_mapAttribute[1] = "BCT";
    m_newPit.m_mapAttribute[2] = "BOOTLOADER";
    m_newPit.m_mapAttribute[3] = "PARTITION TABLE";
    m_newPit.m_mapAttribute[4] = "NVDATA";
    m_newPit.m_mapAttribute[5] = "DATA";
    m_newPit.m_mapAttribute[6] = "MBR";
    m_newPit.m_mapAttribute[7] = "EBR";
    m_newPit.m_mapAttribute[8] = "GP1";
    m_newPit.m_mapAttribute[9] = "GP1";

    m_newPit.m_mapUpdateAttribute[0] = "NONE";
    m_newPit.m_mapUpdateAttribute[1] = "BASIC";
    m_newPit.m_mapUpdateAttribute[2] = "ENHANCED";
    m_newPit.m_mapUpdateAttribute[3] = "EXT2";
    m_newPit.m_mapUpdateAttribute[4] = "YAFFS2";
    m_newPit.m_mapUpdateAttribute[5] = "EXT4";

    m_newPit.m_horizontalHeader << QString("Partition Name") << QString("Binary Type") << QString("Device Type") << QString("Id")
            << QString("Partition Type") << QString("File System") << QString("Start Block") << QString("Block Number")
            << QString("File Offset") << QString("File Size")  << QString("File Name") << QString("Delta Name");
```

It is new (v2) if all of the partitions have the same block size, old (v1) otherwise.

TheAirBlow, 2022-05-27:
[image]
:thinking:

TheAirBlow, 2022-05-27:
```cpp
enum Type {
    TYPE_UNKNOWN,
    TYPE_LOKE,  // Galaxy S+
    TYPE_ODIN,  // Galaxy SII
    TYPE_PROTO_MJOLNIR,
    TYPE_MJOLNIR2,
    TYPE_PLATFORM,
    TYPE_ENUM_SIZE
};
```

Mjolnir is not available on usual devices. It seems to be used internally.
Also, Loke and Odin bootloaders seem to be different?

TheAirBlow, 2022-05-27:
@Grimler91

GNUtoo, 2022-05-29:
About the PIT format, some devices don't have COM_TAR2.

For instance here's a GT-I9100 PIT: https://git.replicant.us/replicant/vendor_replicant-data/tree/devices/PIT/GT-I9100G/stock/16G.pit

Denis.